zephyrproject-rtos/zephyr #25156: Unable to use --use-elf option in 'west flash' to correctly flash the .elf file

vanti, 2020-05-08:
**Describe the bug**
When flashing using the --use-elf option, I get the following error:

```
$ west --verbose flash --use-elf
ZEPHYR_BASE=/home/vwan/zephyrproject-work/zephyr (origin: env)
-- west flash: rebuilding
cmake version 3.13.2 is OK; minimum version is 3.13.1
Running CMake: /home/vwan/cmake/cmake-3.13.2-Linux-x86_64/bin/cmake --build /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build
ninja: no work to do.
-- west flash: using runner openocd
runners.openocd: /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support -s /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'load_image /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'resume 0x00000000010012e9' -c shutdown
Open On-Chip Debugger 0.10.0+dev-01340-ga0e8edc4e-dirty (2020-02-14-05:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.16
Info : XDS110: hardware version = 0x0027
Info : XDS110: connected to target via SWD
Info : XDS110: SWCLK set to 2500 kHz
Info : clock speed 2500 kHz
Info : SWD DPIDR 0x2ba01477
Info : cc32xx.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc32xx.cpu         cortex_m   little cc32xx.cpu         running

Warn : Only resetting the Cortex-M core, use a reset-init event handler to reset any peripherals or configure hardware srst support.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000eb30 msp: 0x20004000
Info : SWD DPIDR 0x2ba01477
Error: Failed to write memory at 0x01000014

Info : XDS110: disconnected
FATAL ERROR: command exited with status 1: /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support -s /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'load_image /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'resume 0x00000000010012e9' -c shutdown
```

It seems to be using different command line options to OpenOCD when compared with flashing using 'west flash' (ie. using .hex file), which works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world
2. west build -b cc3220sf_launchxl
3. west --verbose flash --use-elf

**Expected behavior**
I'd have expected it to correctly flash and run zephyr.elf, using the same OpenOCD command line invocation as for .hex files:

```
$ west --verbose flash
ZEPHYR_BASE=/home/vwan/zephyrproject-work/zephyr (origin: env)
-- west flash: rebuilding
cmake version 3.13.2 is OK; minimum version is 3.13.1
Running CMake: /home/vwan/cmake/cmake-3.13.2-Linux-x86_64/bin/cmake --build /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.hex
runners.openocd: /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support -s /home/vwan/bin/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/vwan/zephyrproject-work/zephyr/boards/arm/cc3220sf_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

**Impact**
Unable to use west to flash .elf files, which are natively supported by OpenOCD.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA or Version used: 6ba69f19d4a4faaa75abbc8a75845049e51077db

mbolivar-nordic, 2020-05-08:
@abrodkin can you take a look please?

abrodkin, 2020-05-09:
@vanti I guess you see the difference between flashing binary in the flash-memory (via quite special procedure with block erase, write protection management etc) and later execution of that code after board reset and loading contents of the Elf-file in word-writable RAM (be it on-chip SRAM, DDR etc).

From your log I see this: `Error: Failed to write memory at 0x01000014` which means you're trying to write non-writable memory (I guess Flash in your case). And that's pretty expected.

First of all why are you trying to use Elf? If you really want to use it and have enough on-board RAM you may link your application so that it is loaded in RAM but not flash.

Let me know if you need more help with that.

vanti, 2020-05-11:
@abrodkin We are making some changes to OpenOCD for our platform, and wanted to validate the behavior when loading elf files was when we came across this issue. Afaik, elf format is generic and does not imply that code would reside in RAM vs flash. and when I look at the help text for "--use-elf": 

```
    --use-elf             if given, Elf file will be used for loading instead of
                          HEX image
```

It simply states that it would use the elf file as opposed to the hex file, so I wouldn't have expected it to make any assumption on code location. OpenOCD itself supports loading to flash when an elf file is used; thus there is no limitation on that front either. 

It seems to me that either 1) the option is badly named and should state that it is there to load to RAM or 2) there should be a different command line flag (e.g. --load-to-ram) that when used in conjunction with --use-elf would determine whether RAM or flash is to be used.

abrodkin, 2020-05-13:
@vanti indeed Elf itself has nothing to do with a way its contents end-up in target's memory. But I may guess from OpenOCD stand-point there's a difference between mechanics of pushing data in memory in case of `flash write_image erase` command and `load_image`.

And I do understand your confusion which I guess happens due to the fact that `west`'s command name is `flash` in both cases - I did chose to just extend its functionality in a bit unusual for meaning of "flash" (as a verb) in a more of "load" sense.

@nashif do you think we should implement a separate `load` command for `west` to overcome mentioned confusion? But I guess if we go that far we'd need to change some more things as well like `sanitycheck` script in a way so it determines somehow for which platform we use `west run` and for which `west flash`.

closed 2020-07-27 by github-actions[bot]