zephyrproject-rtos/zephyr #25164: Remove `default:` functionality from devicetree bindings

galak, 2020-05-09:
We currently support specifying a `default: ` option in the DTS binding YAML.  This option in the binding gets used if a property is not set in the .dts file, then if a `default` is specified the generator will act as if the property was set to that default value.

This concept was partially picked up from the dt-schema (https://github.com/devicetree-org/dt-schema) project.

I'd like to remove the functionality for several reasons:
1. having default values in the dts bindings violates the idea the a DTS is portable.  One of the initial reasons for using DTS was that it could be utilized as a means to convey the hardware configuration for a system to other pieces of software in a portable manor.  For example micropython or javascript/jerryscript could utilize the DTS as a means to know how the hardware was configured, what UARTs, etc.  Also, if other RTOSes or other parts of a more complicated SoC/System utilize DTS we would be violating the portability principle as properties that should be specified in the DTS are not.  And thus the DTS file is not standalone.

2. The intent we assumed from dt-schema is not correct:
Since dt-schema is utilized for validation of dts files I chatted with Rob Herring the maintainer of dt-schema about its use, below is the transcript:
```
<galak> @robher what's the purpose of default in dt-schema?
<•robher> galak: nothing ATM
<•robher> galak: Folks like to say 'default is X' and json-schema had a keyword for 'default'...
<galak> any clue to why people are adding them to YAMLs in kernel bindings?
<•robher> galak: 'cause we tell them to rather than the free form description.
<•robher> galak: Could check for unneeded properties is one thing I had in mind.
<galak> robher: How would that work.  Would the assumption be that the driver or HW is set to the default value?
<•robher> yeah
```

3. Potentially error prone:
By setting defaults in the binding the user doesn't know they need to specify a configuration.  We recently discussed the idea of pinctrl defaults in DTS and how we shouldn't set them in the SoC dtsi files because they could potentially be harmful.  I see this similar to that.  We could have subtle bugs that a user or even developer didn't know they needed to set a value because they inherited a default that was completely wrong for them.

I'd propose we treat `default: ` more similar to what the intent from the dt-schema project is.  That it's informational, but that DTS files are still standalone descriptions.  We'd change EDT to first warn about any `default:` that don't have a corresponding DTS property (fixup DTS or drivers), then we'd change that to be an error after 2 releases.

mnkp, 2020-05-09:
The above arguments sound very reasonable. I would like to support the proposal.

pabigot, 2020-05-10:
I understand the argument, only wish [it had been discussed to its conclusion](https://github.com/zephyrproject-rtos/zephyr/issues/17829#issuecomment-517300177) when the feature (which is very useful) was added to Zephyr's infrastructure in #18788.  If we reverse course now there are 14 drivers in Zephyr currently using this feature that have to be changed.

I certainly don't agree that "the intent from the dt-schema project" means all properties must be explicitly set in the DTS and we should error if they're not.  That's not consistent with support for `required: false` in Zephyr's devicetree infrastructure.

The transcript ("we tell them to [add a default]", "yeah [driver or hardware is set to the default value]") supports using this field as documentation of driver behavior when a value it is not present in the devicetree.  That's what Linux does:
```
/**
 * of_property_read_u32_index - Find and read a u32 from a multi-value property.
 *
 * @np:         device node from which the property value is to be read.
 * @propname:   name of the property to be searched.
 * @index:      index of the u32 in the list of values
 * @out_value:  pointer to return value, modified only if no error.
 *
 * Search for a property in a device node and read nth 32-bit value from
 * it. Returns 0 on success, -EINVAL if the property does not exist,
 * -ENODATA if property does not have a value, and -EOVERFLOW if the
 * property data isn't large enough.
 *
 * The out_value is modified only if a valid u32 value can be decoded.
 */
int of_property_read_u32_index(const struct device_node *np,
                                       const char *propname,
                                       u32 index, u32 *out_value);
```

Documentation/devicetree/bindings/i2c/st,stm32-i2c.yaml:
```
        i2c-scl-rising-time-ns:
          default: 25
```
drivers/i2c/busses/i2c-stm32f7.c:
```
#define STM32F7_I2C_RISE_TIME_DEFAULT           25      /* ns */
...
static const struct stm32f7_i2c_setup stm32f7_setup = {
        .rise_time = STM32F7_I2C_RISE_TIME_DEFAULT,
...
        ret = device_property_read_u32(i2c_dev->dev, "i2c-scl-rising-time-ns",
                                       &rise_time);
        if (!ret)
                i2c_dev->setup.rise_time = rise_time;
```

There are two ways to make a default work in Zephyr:

* Have the visible documented value from the binding filled in by the tooling when the devicetree doesn't include the property.  This is something we can do that Linux can't because we have all this tooling doing the translation for us.
* Have the driver implementation check for the property and when it's missing provide the default value, just like Linux.

The latter seems more error-prone to me (how do you keep the documented and actual values in sync), but we can go that way.  It's certainly more friendly and less error prone than going back to the old way of making everybody fill in a value for every possible property in their devicetree source.

galak, 2020-05-14:
Dev-review (May 14):
* discussed the topic, need input or review from others like @mbolivar-nordic.  @galak had the idea if we generate the zephyr/zephyr.dts with defaults injected so at least there's a form of the DTS that is consistent.

Need to also see how we limit abuse of defaults if we maintain the feature.

mbolivar-nordic, 2020-05-18:
@galak @pabigot I read the transcript with robher and looked at the I2C example for STM32F7. I can't really parse some of the details of the transcript, but it looks like the dt-schema default is saying "this is the default set by the driver", which is consistent with the I2C example.

If that's so, I wonder why we can't just use the bindings as a single point of truth for both documenting the driver's default and setting it.

What would you think about something like this?

```
#define DT_PROP_OR_DEFAULT(node_id, prop)				\
	COND_CODE_1(DT_NODE_HAS_PROP(node_id, prop),			\
		DT_PROP(node_id, prop),					\
		DT_CAT(node_id, _P_##prop##_DEFAULT)
```

Where the `DT_<node_path>_P_<property>_DEFAULT` macro is new, and would be generated by gen_defines.py using the default value in the bindings if one is set. We could try to make it cause an understandable build error if one is not set rather than just bailing on undefined symbol.

The value of `DT_PROP_OR_DEFAULT(foo, bar)` would be the `bar` property value if it is explicitly defined, or `xyz` if the binding contains `default: xyz` if it is not defined.

pabigot, 2020-05-18:
@mbolivar-nordic I'd considered that the tooling could forward the binding default to the build (e.g. `DT_DEFAULT_compat_P_prop`) but I saw that as a more complex solution than just having the devicetree tooling set the default directly as the node's property value when it isn't in the devicetree source.  Which we already have and which works as expected (except that the substitution doesn't show up in `zephyr.dts` as proposed above).

I'm not really seeing how `DT_PROP_OR_DEFAULT` improves on the current solution either.

mbolivar-nordic, 2020-05-18:
Like I said, I couldn't parse the IRC transcript very well, but it seems like my proposal is semantically what Linux is doing today, except the default value appears once (in the binding) instead of twice as it seems to in Linux (in the binding and again in the driver).

I also think using a macro with OR_DEFAULT in the name may be helpful during debugging, to @galak's point number 3. in the issue description.

mbolivar-nordic, 2020-05-18:
Another thing I think is potentially in DT_PROP_OR_DEFAULT()'s favor relative to the status quo is that adding it would ultimately allow DT_PROP() to error out if the property is not explicitly set, and is therefore a way to enforce that it is.

But if it's not useful and I've misunderstood the problem, disregard the noise.

pabigot, 2020-05-18:
> Like I said, I couldn't parse the IRC transcript very well, but it seems like my proposal is semantically what Linux is doing today, except the default value appears once (in the binding) instead of twice as it seems to in Linux (in the binding and again in the driver).

I don't think your comments are noise; I agree entirely, and my point is the tooling we have today does exactly what you describe: if a property default is specified in the binding, and the devicetree source doesn't include a property, the generated `devicetree_unfixed.h` provides the property definitions using the default from the binding.

Right now we can use `required: true` to force setting a property explicitly (the absence produces a build error, or at least it used to), and neither approach fixes mis-use of `default` in a binding where it doesn't make sense (as with `ti,bq274xx`).

mbolivar-nordic, 2020-05-18:
I don't really disagree with anything you've said, so I suppose we should leave it up to @galak to decide what to do. FTR:

> 1. having default values in the dts bindings violates the idea the a DTS is portable.

Linux seems to have default values in the drivers, and a problem where the defaults in their dt-schema bindings could get out of sync with the drivers, so I don't really get the portability concern here.

> 2. The intent we assumed from dt-schema is not correct:

The intent here seems to be that it's "just documentation for the default value" in dt-schema. But in Zephyr, we're both documenting the default and enforcing it, which seems better to me.

> 3. Potentially error prone

I'm not sure I follow this either. If we are in a situation where there is no sane default, marking the property `required: True` as @pabigot suggests would seem to be a good way out. Perhaps there is an example (or something about the Atmel pinctrl cited above) where there's something I'm missing?

pabigot, 2020-07-16:
At this time I've reviewed the in-tree bindings and provided a PR for the only one that I think is not appropriate.

pabigot, 2020-09-30:
@galak please lets continue the discussion from  https://github.com/zephyrproject-rtos/zephyr/pull/28814#issuecomment-701443566 here.

I thought we'd agreed that there were cases where defaulting is reasonable, and that if a default is provided it should be exposed  in the binding where people can see what it is.  It does not make sense to put a documentation-only property in the binding, then ignore it and hard-code a value in the implementation, when the tooling provides the default that came from the binding.

You're not removing the fact that there's a default, just making it hard to discover that.  We can't just give a general rule that if a `required: false` property is not provided it defaults to the zero value for its type, because that's not true: the absence may be interpreted differently than presence with a zero value.

galak, 2020-09-30:
> I thought we'd agreed that there were cases where defaulting is reasonable, and that if a default is provided it should be exposed in the binding where people can see what it is. It does not make sense to put a documentation-only property in the binding, then ignore it and hard-code a value in the implementation, when the tooling provides the default that came from the binding.
> 
> You're not removing the fact that there's a default, just making it hard to discover. We can't just give a general rule that if a `required: false` property is not provided it defaults to the zero value for its type, because that's not true: the absence may be interpreted differently than presence with a zero value.

We need to figure out some guidelines about `default` usage as document them.  It has been my concern from the start w/`default` as its very subjective right now.  Right now it feels like syntax short hand and based on personal preference on how a developer feels like they should write their driver.

pabigot, 2020-09-30:
I don't see how it can ever be anything but subjective, but we could provide some general guidelines.  Maybe this?

> Some properties require a value but the majority of devices will be correctly configured with a default.  Rather than force every instance in a devicetree to replicate these values, the value should be provided in a `default: ` key in the binding.  If the devicetree source doesn't explicitly provide a value for a `required: false` property the devicetree tooling will supply the value from the binding.
> 
> Driver developers should use their best judgment as to whether a value can be defaulted or the property should be made `required: true`.  Candidates for default values include delays that would different only under unusual conditions (such intervening hardware), or configuration for devices that have a standard initial configuration (such as a USB audio headset).    Properties where a default value would likely be incorrect (such as the capacity of an external battery) should be made required rather than defaulted.

mbolivar-nordic, 2020-09-30:
> Maybe this?

That seems pretty reasonable IMO.

PR #30095 (2020-11-19): dts: binding-template: document conditions for use of default values
The YAML binding allows for a default which is used to substitute a
value if one is not explicit in the devicetree source.  This feature
is mis-used when defaults are provided that are likely to be
incorrect.

Document concerns with the use of default and provide guidance for
when it can be used, what to do when it shouldn't, and noting the need
to explain the source of the default in the binding documentation.

Closes  #25164 assuming we don't really need to insert the default into the generated `zephyr.dts` binding file.
closed 2020-11-19 by carlescufi