zephyrproject-rtos/zephyr #25169: soc/arm/infineon_xmc/4xxx/soc.h not found

sglass68, 2020-05-09:
**Describe the bug**
I would like all the tests to pass locally but I seem to be msising a file

**To Reproduce**
Steps to reproduce the behavior:
1. ./scripts/sanitycheck 
2. See error

**Expected behavior**
Woulkd like all tests to pass

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**

cat /opt/zephyr-sdk/sdk_version 
0.11.2

cat /home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization/build.log
Including boilerplate (Zephyr base): /home/sglass/dev/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/sglass/dev/zephyrproject/zephyr/samples/synchronization
-- Zephyr version: 2.2.99 (/home/sglass/dev/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Board: xmc45_relax_kit
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found west: /home/sglass/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/sglass/dev/zephyrproject/zephyr/boards/arm/xmc45_relax_kit/xmc45_relax_kit.dts
-- Generated zephyr.dts: /home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/sglass/dev/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/sglass/dev/zephyrproject/zephyr/boards/arm/xmc45_relax_kit/xmc45_relax_kit_defconfig'
Merged configuration '/home/sglass/dev/zephyrproject/zephyr/samples/synchronization/prj.conf'
Configuration saved to '/home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization/zephyr/.config'
Kconfig header saved to '/home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/sglass/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sglass/dev/zephyrproject/zephyr/sanity-out/xmc45_relax_kit/samples/synchronization/sample.kernel.synchronization
Scanning dependencies of target parse_syscalls_target
[  1%] Generating misc/generated/syscalls.json, misc/generated/subsystems.json
[  3%] Built target parse_syscalls_target
Scanning dependencies of target kobj_types_h_target
[  4%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[  4%] Built target kobj_types_h_target
Scanning dependencies of target syscall_list_h_target
[  5%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  8%] Built target syscall_list_h_target
Scanning dependencies of target driver_validation_h_target
[  9%] Generating include/generated/driver-validation.h
[ 12%] Built target driver_validation_h_target
Scanning dependencies of target offsets
[ 13%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
In file included from /home/sglass/dev/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/sglass/dev/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from /home/sglass/dev/zephyrproject/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu.h:13,
                 from /home/sglass/dev/zephyrproject/zephyr/include/arch/arm/aarch32/arch.h:248,
                 from /home/sglass/dev/zephyrproject/zephyr/include/arch/cpu.h:19,
                 from /home/sglass/dev/zephyrproject/zephyr/include/kernel_includes.h:39,
                 from /home/sglass/dev/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/sglass/dev/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:25,
                 from /home/sglass/dev/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:12:
/home/sglass/dev/zephyrproject/zephyr/soc/arm/infineon_xmc/4xxx/soc.h:11:10: fatal error: system_XMC4500.h: No such file or directory
   11 | #include <system_XMC4500.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
zephyr/CMakeFiles/offsets.dir/build.make:82: recipe for target 'zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj' failed
make[2]: *** [zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj] Error 1
CMakeFiles/Makefile2:2083: recipe for target 'zephyr/CMakeFiles/offsets.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/offsets.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA or Version used: 1fed229f6e

**Additional context**
Add any other context about the problem here.

galak, 2020-05-09:
did you do a `west update`?

carlescufi, 2020-05-09:
> did you do a `west update`?

Exactly. @sglass68 support for Infineon Arm ICs was merged earlier today and it requires a new HAL layer which is a separate repository. Running `west update` should fix the issue.

sglass68, 2020-05-09:
Oopts, that did it, thanks! Will make a note.

closed 2020-05-09 by sglass68