zephyrproject-rtos/zephyr #25173: k_sem_give(struct k_sem *sem) should report failure when the semaphore is full

mofo, 2020-05-09:
I am porting some code and I need to check if I can lose a semaphore if I don't check it is full the sem_give  call fails silently.

sem_give should return an error code like sem_take does when the semaphore is empty
int k_sem_give(struct k_sem *sem)

as a workaround I have to check  if the semaphore is full before the give request to make sure I will not lose the action:

currentCount = k_sem_count_get(semphr);

This is bad because it will not be atomic obviously

carlescufi, 2020-05-10:
@andyross and @andrewboie would appreciate feedback as to whether this can be considered a bug.

mofo, 2020-05-10:
I would consider it a bug since k_sem_take does have a return code and conceptually the two functions should be symmetric.

andrewboie, 2020-05-10:
Confirmed by looking at the code, it ends up being just a reschedule point if the limit has already been reached.

Honestly, if I could get away with it, I would remove the concept of semaphore limits entirely, it doesn't fit the textbook definition of a semaphore, although there is a precedent for it in at least a few other OSes.

I'm not entirely sure this counts as a bug, as the documented behavior is "this routine gives a sem, unless the semaphore is already at its maximum permitted count". So in a pedantic sense it is working correctly. I'm leaning more towards 'enhancement' for this.

A patch to implement this seems straightforward if you'd like to contribute one, I can't see how adding this could break anyone (other than a very minor footprint increase).

mofo, 2020-05-11:
Hi @andrewboie, I actually agree 100% with all of your statements.

"sys_sem" does currently return error in case the semaphore reached the max count

/**
 * @brief Give a semaphore.
 *
 * This routine gives @a sem, unless the semaphore is already at its
 * maximum permitted count.
 *
 * @param sem Address of the semaphore.
 *
 * @retval 0 Semaphore given.
 * @retval -EINVAL Parameter address not recognized.
 * @retval -EACCES Caller does not have enough access.
 * @retval -EAGAIN Count reached Maximum permitted count and try again.
 */
int sys_sem_give(struct sys_sem *sem);

andyross, 2020-05-11:
Yeah, limits are weird as an API feature.  The internal code actually requires that some kind of clamping be present (otherwise you could wrap at 32 bits and get wrong behavior), but presenting it to users as a tunable is just asking for trouble.

As for whether we want to add a return value to `k_sem_give()` (it is void right now)... I vote no, honestly.  That's a pretty heavy API change; basically everything that uses semaphores now needs an extra test and branch for an error that will never reasonably occur in practice in "standard" usage.

Can you be more explicit about the app that's having trouble?  Maybe there's another way to do this.

andyross, 2020-05-11:
(whoops, clicked wrong button and closed it by accident, sorry)

andrewboie, 2020-05-11:
> As for whether we want to add a return value to `k_sem_give()` (it is void right now)... I vote no, honestly. That's a pretty heavy API change; basically everything that uses semaphores now needs an extra test and branch for an error that will never reasonably occur in practice in "standard" usage.

@andyross Would all code really need an extra test and branch? Couldn't existing code just ignore the return value? The proposed change doesn't alter the behavior of k_sem_give() other than to provide a return value if the limit is reached for those that care about it.

mofo, 2020-05-11:
@andyross: I am implementing the OSI interface for a block that I do not control (just playing around with the ESP32 BT controller and making a VHCI driver for it) but expects to fail if the semaphore is full:
int32_t (*_semphr_give)(void *semphr);

The OSI also requests a MAX for the created semaphore:
void *(*_semphr_create)(uint32_t max, uint32_t init);

I do agree with @andrewboie that the change would not affect anyone besides adding a few instructions to sem_give: users can still ignore the return value as if it was void. Also the change would make sys_sem_give and k_sem_give symmetric.

andyross, 2020-05-11:
It would by current coding style guidelines, yeah.   That's an unchecked error. :)

I guess I'm just saying I don't see the value.  If someone is actually using the limit feature, they surely **want** the silent saturation behavior we document, because no one else has a semaphore like this so clearly the desire is for our (weird) scheme, right?  If the app doesn't want to saturate, it should leave the limit at the very high (~INT_MAX or whatever) value it has by default, which matches the way semaphore work on other systems.

If the app desire is for a saturating semaphore with atomic notification of saturation, well, OK, we don't have that.  But no one else does either, and I guess I want to know why that's a desired feature.

Personally I've very much prefer to deprecate the limit feature than add another wart to the API.

andyross, 2020-05-11:
@mofo : what exactly is the API you're wrapping?  Are you sure it expects the error detection you're trying to implement?

andrewboie, 2020-05-11:
> Personally I've very much prefer to deprecate the limit feature than add another wart to the API.

No disagreement here. But I don't see a path to doing that without a complete API overhaul like the micro/nanokernel merge.

> no one else has a semaphore like this so clearly the desire is for our (weird) scheme, right?

There's other OSes that do implement a limit, it's not just us.

For example in FreeRTOS: https://www.freertos.org/CreateCounting.html

```
uxMaxCount | The maximum count value that can be reached. When the semaphore reaches this value it can no longer be ‘given’.
```

mofo, 2020-05-11:
> @mofo : what exactly is the API you're wrapping? Are you sure it expects the error detection you're trying to implement?

No, I can just assume since I can see the FreeRTOS implementation

andrewboie, 2020-05-11:
> > @mofo : what exactly is the API you're wrapping? Are you sure it expects the error detection you're trying to implement?
> 
> No, I can just assume since I can see the FreeRTOS implementation

@mofo let me ask you this: does FreeRTOS counting semaphore return an error if the limit is reached? Your argument would be bolstered if you can show precedent.

mofo, 2020-05-11:
> **xSemaphoreGive( SemaphoreHandle_t xSemaphore );**
**Macro** to release a semaphore. The semaphore must have previously been created with a call to xSemaphoreCreateBinary(), xSemaphoreCreateMutex() or xSemaphoreCreateCounting().
**Returns**:
pdTRUE if the semaphore was released. pdFALSE if an error occurred. Semaphores are implemented using queues. An error can occur if there is no space on the queue to post a message – indicating that the semaphore was not first obtained correctly

From the API description: [https://www.freertos.org/a00123.html](url)

andrewboie, 2020-05-11:
> From the API description: [https://www.freertos.org/a00123.html](url)

I saw that too but that doesn't answer my question as there is no specific language about what causes pdFALSE returned other than "no space on the queue".

mofo, 2020-05-11:
Not sure if this is what you are asking but from a quick code inspection:

xSemaphoreGive is implemented as a Queue as specified by the API description
> #define xSemaphoreGive( xSemaphore )		xQueueGenericSend( ( QueueHandle_t ) ( xSemaphore ), NULL, semGIVE_BLOCK_TIME, queueSEND_TO_BACK )

**xQueueGenericSend** ONLY has 3 possible exits:
1 - everything good and an item can be added
**pdPASS**: 

2 - There is no space in the queue and the timeout expires
3 - There is no timeout and there is no space in the queue
**errQUEUE_FULL**: 

**errQUEUE_FULL** can be mapped as **pdFALSE** or **pdFAIL**

> #define pdFALSE			( ( BaseType_t ) 0 )
#define pdTRUE			( ( BaseType_t ) 1 )
#define pdPASS			( pdTRUE )
#define pdFAIL			( pdFALSE )
#define errQUEUE_EMPTY	( ( BaseType_t ) 0 )
#define errQUEUE_FULL	( ( BaseType_t ) 0 )

andrewboie, 2020-05-11:
> Not sure if this is what you are asking but from a quick code inspection:

What I am asking is, specifically, **is pdFALSE returned if the counting semaphore is given when the limit is reached**? Nothing cited so far seems to cover this case, all it discusses is its internal queue implementation

mofo, 2020-05-11:
> > Not sure if this is what you are asking but from a quick code inspection:
> 
> What I am asking is, specifically, **is pdFALSE returned if the counting semaphore is given when the limit is reached**? Nothing cited so far seems to cover this case, all it discusses is its internal queue implementation

I'm not sure I understand:

We agree that **xSemaphore** doesn't really exist and it is a **xQueueGeneric** right:
>#define xSemaphoreCreateCounting( uxMaxCount, uxInitialCount ) xQueueCreateCountingSemaphore( ( uxMaxCount ), ( uxInitialCount ) )

           ......

		configASSERT( uxMaxCount != 0 );
		configASSERT( uxInitialCount <= uxMaxCount );

		xHandle = xQueueGenericCreate( uxMaxCount, queueSEMAPHORE_QUEUE_ITEM_LENGTH, queueQUEUE_TYPE_COUNTING_SEMAPHORE );

		if( xHandle != NULL )
		{
			( ( Queue_t * ) xHandle )->uxMessagesWaiting = uxInitialCount;

			traceCREATE_COUNTING_SEMAPHORE();
           ......
           ......
           ......

		return xHandle;
	}

and the return from **xSemaphoreGive** is directly the return from **xQueueGenericSend**:
>#define xSemaphoreGive( xSemaphore ) xQueueGenericSend( ( QueueHandle_t ) ( xSemaphore ), NULL, semGIVE_BLOCK_TIME, queueSEND_TO_BACK )

So to be specific: I'm not using freeRTOS right now but as far as I can tell the **xSemaphoreGive** will return **pdFAIL** (**errQUEUE_FULL**) when the queue (or the "semaphore") contains **uxMaxCount** (which would be the max parameter set at creation) already and we try to add ("xSemaphoreGive"). The timeout is always 0 because the macro sets it to 0 all the time.

andrewboie, 2020-05-11:
> I'm not sure I understand:
> 
> We agree that **xSemaphore** doesn't really exist and it is a **xQueueGeneric** right:

I absolutely do not care about the implementation details.

Does it return false if the limit is already reached, or not? Simple question. Sounds like we have no idea because the FreeRTOS documentation is terrible.

mofo, 2020-05-11:
If you are asking me if philosophically they meant to return **pdFALSE** when a semaphore reaches the max count I can't comment :)

In reality, as far as I can tell and just short of running a test, the answer to the question
> Does it return false if the limit is already reached, or not?

is  **YES**

**API reference**:
> Returns:
**pdTRUE** if the semaphore was released. **pdFALSE** if an error occurred. Semaphores are implemented using queues. **An error can occur if there is no space on the queue to post a message**

Clearly states that the semGive will return **pdFALSE** if we reached the MaxCount. I agree that Semantically It does leave open the possibility of other errors that can cause it to return pdFALSE

**Code inspection**:
confirms the reading of the API reference:
>when timeout expires and there is no space in the queue it will returns: ** pdFAIL**

Also seems to confirm that there is basically only **ONE** return **pdFALSE** path and that is when the semaphore reached Max.

So **YES** it seems like xSemaphoreGive() returns **pdFAIL** when and only when it reaches Max.

Besides the FreeRTOS behavior did we want to make make **sys_sem_give** and **k_sem_give** symmetric?

andyross, 2020-05-11:
I think the practical framing was more: "What real world code actually cares about the distinction?"

FWIW, turns out win32 has a similar limit feature, and a return value from ReleaseSemaphore() that can give an error.  Though it has a similar documentation failure to FreeRTOS in that it doesn't actually say whether or not an overflow constitutes an error.

Hilarious, POSIX[1], while it does not define a limit parameter to sem_init(), actually does define a EOVERFLOW erno result from `sem_post()` indicating "The maximum allowable value for a semaphore would be exceeded.".

So I reiterate my position that basically no one does this "right" and that Zephyr ought to be picking the simplest subset people actually need in practice (which IMHO would include ejecting the whole limit stuff from the API).  Though I guess there's enough evidence of this stuff out there that I shouldn't oppose a submission that did this.

[1] POSIX semaphores are actually fairly obscure, probably owing to confusion over the absolutely horrid SysV semaphore mistake.  Condition variables are the more natural synchronization tool there.

andrewboie, 2020-05-11:
I wouldn't stand in the way of a PR for this either, but I concur with @andyross remarks above.

> Besides the FreeRTOS behavior did we want to make make **sys_sem_give** and **k_sem_give** symmetric?

Yes, regardless of what ultimately gets done, these should be synchronized (either removing `-EAGAIN` from `sys_sem`, or adding to `k_sem`). That was my fault, I added `sys_sem` for futex use-cases and didn't notice.

carlescufi, 2020-05-21:
@andyross @andrewboie is this a bug or an enhancement?

nashif, 2020-10-01:
based n the last comment from @andrewboie , I consider this a bug

closed 2020-12-15 by github-actions[bot]