zephyrproject-rtos/zephyr #25174: qemu test failures when running sanitycheck

sglass68, 2020-05-10:
**Describe the bug**
I see test faillures most times when I run this script

**To Reproduce**
Steps to reproduce the behavior:
1. ./scripts/sanitycheck 
2. See error

**Expected behavior**
Expect all tests to pass

**Impact**
annoyance

**Screenshots or console output**
$ ./scripts/sanitycheck 
Renaming output directory to /home/sglass/dev/zephyrproject/zephyr/sanity-out.10
INFO    - JOBS: 64
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 4156 test configurations selected, 199120 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:  830/4156  19%  skipped:   91, failed:    0
ERROR   - qemu_x86_coverage         samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization FAILED: Timeout
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_coverage/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/handler.log
INFO    - Total complete:  831/4156  19%  skipped:   91, failed:    1
ERROR   - qemu_x86                  samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization FAILED: Timeout
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/handler.log
INFO    - Total complete:  832/4156  20%  skipped:   91, failed:    2
ERROR   - qemu_xtensa               samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization FAILED: Timeout
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_xtensa/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/handler.log
INFO    - Total complete: 1217/4156  29%  skipped:  134, failed:    3
ERROR   - mps2_an521_nonsecure      samples/tfm_integration/tfm_ipc/sample.tfm_ipc     FAILED: Build failure
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/mps2_an521_nonsecure/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/build.log
INFO    - Total complete: 1218/4156  29%  skipped:  134, failed:    4
ERROR   - mps2_an521_nonsecure      samples/tfm_integration/psa_level_1/sample.tfm_ipc FAILED: Build failure
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/mps2_an521_nonsecure/samples/tfm_integration/psa_level_1/sample.tfm_ipc/build.log
INFO    - Total complete: 1875/4156  45%  skipped:  176, failed:    5

$ cat /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_coverage/samples/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization/handler.log
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 7
*** Booting Zephyr OS build zephyr-v2.2.0-2692-gff856e01f5d8  ***
Wrote to message queue: 5
Read from message queue: 5

Wrote to message queue: 6
Read from message queue: 6

Wrote to message queue: 7
Read from message queue: 7

Wrote to message queue: 8
Read from message queue: 8

Read from message queue: 9

**Error reading message from message queue**

**Error reading message from message queue**

**Error reading message from message queue**

**Error sending message to message queue**

**Error reading message from message queue**
Error in execution! 

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.2
 - Commit SHA or Version used: cd0659e36f

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-05-10:
@microbuilder can you take a look at the tf-m failures?
@andyross the cmsis failures, are they related to the timeout porting?

andrewboie, 2020-05-10:
If these are intermittent rather than consistent failures I think this is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/14173

Do they pass if you re-run `sanitycheck` with `--only-failed`?

microbuilder, 2020-05-11:
I've run the following just now, with no failures:

```
$ ./zephyr/scripts/sanitycheck --all --enable-slow -p mps2_an521_nonsecure
Install the anytree module to use the --test-tree option
Install tabulate python module with pip to use --device-testing option.
Install tabulate python module with pip to use --device-testing option.
Renaming output directory to /Users/kevin/Linaro/zephyr/upstream/sanity-out.2
INFO    - JOBS: 12
INFO    - Selecting all possible platforms per test case
INFO    - Building initial testcase list...
INFO    - 222 test configurations selected, 598 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:  222/ 222  100%  skipped:   23, failed:    0
INFO    - 199 of 199 tests passed (100.00%), 0 failed, 23 skipped with 0 warnings in 764.69 seconds
INFO    - In total 831 test cases were executed on 1 out of total 249 platforms (0.40%)
```

I wonder if this is failing due to certain requirements around TF-M not being met, specifically: https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/tfm_integration/tfm_integration.rst#tf-m-requirements

The `_nonsecure` board target should generally only be used with TF-M samples as well, since there are some post-build steps, as described here: https://github.com/zephyrproject-rtos/zephyr/issues/22872#issuecomment-626608136

andyross, 2020-05-11:
Not seeing this on my laptop (which usually matches the host timer failure rates in CI).  Loading the CPUs with external shell loops and running sanitycheck in a loop over the cmsis_v2 sample showed no failures after 44 iterations.

sglass68, 2020-05-11:
@andrewboie 
Yes './scripts/sanitycheck --only-failed' makes them pass.

sglass68, 2020-05-11:
@microbuilder I found that a while ago and it definitely is the problem with some of the tests. Could it be moved to the main page? https://docs.zephyrproject.org/latest/getting_started/installation_linux.html

Am running again with master and will report back here

andrewboie, 2020-05-11:
This is just #14173

closed 2020-05-11 by andrewboie

sglass68, 2020-05-11:
Results of my re-run. All seem to be due to the qemu problem.

./scripts/sanitycheck 
Renaming output directory to /home/sglass/dev/zephyrproject/zephyr/sanity-out.13
INFO    - JOBS: 64
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 4160 test configurations selected, 199116 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete: 1954/4160  46%  skipped:  177, failed:    0
ERROR   - qemu_xtensa               tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_xtensa/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
INFO    - Total complete: 2049/4160  49%  skipped:  177, failed:    1
ERROR   - qemu_x86_64               tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy/handler.log
INFO    - Total complete: 2078/4160  49%  skipped:  177, failed:    2
ERROR   - qemu_cortex_m0            tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_cortex_m0/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy/handler.log
INFO    - Total complete: 2192/4160  52%  skipped:  209, failed:    3
ERROR   - qemu_x86_coverage         tests/kernel/context/kernel.common                 FAILED: Exited with 2
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_coverage/tests/kernel/context/kernel.common/handler.log
INFO    - Total complete: 2201/4160  52%  skipped:  209, failed:    4
ERROR   - qemu_x86                  tests/kernel/context/kernel.common                 FAILED: Exited with 2
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86/tests/kernel/context/kernel.common/handler.log
INFO    - Total complete: 2276/4160  54%  skipped:  209, failed:    5
ERROR   - qemu_xtensa               tests/kernel/context/kernel.common                 FAILED: Timeout
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_xtensa/tests/kernel/context/kernel.common/handler.log
INFO    - Total complete: 2325/4160  55%  skipped:  209, failed:    6
ERROR   - qemu_cortex_m3            tests/kernel/common/kernel.common                  FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_cortex_m3/tests/kernel/common/kernel.common/handler.log
INFO    - Total complete: 2457/4160  59%  skipped:  220, failed:    7
ERROR   - qemu_nios2                tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_nios2/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/handler.log
INFO    - Total complete: 2488/4160  59%  skipped:  228, failed:    8
ERROR   - qemu_x86                  tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/handler.log
INFO    - Total complete: 2497/4160  60%  skipped:  229, failed:    9
ERROR   - qemu_x86_coverage         tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_coverage/tests/kernel/sched/schedule_api/kernel.scheduler.multiq/handler.log
INFO    - Total complete: 2507/4160  60%  skipped:  233, failed:   10
ERROR   - qemu_x86_nommu            tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_nommu/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete: 2519/4160  60%  skipped:  233, failed:   11
ERROR   - qemu_xtensa               tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_xtensa/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete: 2533/4160  60%  skipped:  237, failed:   12
ERROR   - qemu_cortex_a53           tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_cortex_a53/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete: 2560/4160  61%  skipped:  251, failed:   13
ERROR   - qemu_x86                  tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete: 2562/4160  61%  skipped:  252, failed:   14
ERROR   - qemu_x86_coverage         tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_coverage/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
INFO    - Total complete: 2563/4160  61%  skipped:  252, failed:   15
ERROR   - qemu_x86_64               tests/kernel/sched/schedule_api/kernel.scheduler   FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_64/tests/kernel/sched/schedule_api/kernel.scheduler/handler.log
[... cut ...]
ERROR   - qemu_x86_nommu            tests/posix/common/portability.posix.common        FAILED: Failed
ERROR   - see: /home/sglass/dev/zephyrproject/zephyr/sanity-out/qemu_x86_nommu/tests/posix/common/portability.posix.common/handler.log
INFO    - Total complete: 4160/4160  100%  skipped:  483, failed:   31
INFO    - 3646 of 3677 tests passed (99.16%), 31 failed, 483 skipped with 0 warnings in 2196.86 seconds
INFO    - In total 830 test cases were executed on 243 out of total 249 platforms (97.59%)
INFO    - 2166 tests executed on platforms, 1994 tests were only built.

andrewboie, 2020-05-11:
I'd like to know more details about your build environment, these intermittent failures are known but they seem to fail way more often for you than the rest of us.

```
INFO - JOBS: 64
```

How many CPUs/threads does your workstation actually have?

sglass68, 2020-05-11:
It has 32 CPUs with 64 threads.

$ head -28  /proc/cpuinfo 
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 49
model name	: AMD Ryzen Threadripper 3970X 32-Core Processor
stepping	: 0
microcode	: 0x8301025
cpu MHz		: 4045.911
cache size	: 512 KB
physical id	: 0
siblings	: 64
core id		: 0
cpu cores	: 32
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 16
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate sme ssbd ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif umip rdpid overflow_recov succor smca
bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 7400.53
TLB size	: 3072 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]

andrewboie, 2020-05-11:
Nice CPU :)

I routinely pass -j48 to my 3950X (16c/32t) but don't experience the magnitude of errors you are seeing. Weird. Wonder if it's I/O bottleneck or something? The QEMU issues tend to happen more if the system is heavily loaded with large jumps in emulated time.

At any rate, I expect a lot of this would go away (except for x86_64 which has special problems with icount due to being SMP) once #14173 is complete. 

You could try cherry-picking https://github.com/zephyrproject-rtos/zephyr/pull/22904 see if it helps.