zephyrproject-rtos/zephyr #25176: tests/kernel/context failed on multiple platforms.

chen-png, 2020-05-10:
**Describe the bug**
This test  tests/kernel/context failed on reel_board,  frdm_k64f,  sam_e70_xplained, mec15xxevb_assy6853 and up_squared.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto  tests/kernel/context
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-2667-g57671d97165f  ***
Running test suite context

starting test - test_kernel_interrupts
PASS - test_kernel_interrupts

starting test - test_kernel_timer_interrupts
SKIP - test_kernel_timer_interrupts

starting test - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread

starting test - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)
**Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/context/src/main.c:786: test_busy_wait: (rv is true)
*** thread timed out waiting for k_busy_wait()
FAIL - test_busy_wait**

starting test - test_k_sleep
**Testing k_thread_create() without cancellation
thread (q order: 0, t/o: 1000) is running
E: ***** BUS FAULT *****
E:   Precise data bus error**
E:   BFAR Address: 0xfffffff5
E: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x20000c18
E: r3/a4:  0x20000558 r12/ip:  0x00000000 r14/lr:  0x000074fb
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x000074fa
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000558 (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.2
 - Commit ID: 57671d971

erwango, 2020-05-10:
Seen on stm32 boards as well (disco_l475_iot1, nucleo_f429zi, nucleo_f746zg, nucleo_l152re, nucleo_wb55rg, stm32f3_disco, ...)

stephanosio, 2020-05-11:
This seems to be happening on all Cortex-M platforms (also happens on Cortex-M4 `atsame54_xpro` and Cortex-M0 `atsamd21_xpro`).

stephanosio, 2020-05-11:
`atsame54_xpro` (Cortex-M4):
```
*** Booting Zephyr OS build v2.3.0-rc1  ***
Running test suite context
===================================================================
starting test - test_kernel_interrupts
PASS - test_kernel_interrupts
===================================================================
starting test - test_kernel_timer_interrupts
SKIP - test_kernel_timer_interrupts
===================================================================
starting test - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread
===================================================================
starting test - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)

    Assertion failed at ../src/main.c:786: test_busy_wait: (rv is true)
 *** thread timed out waiting for k_busy_wait()
FAIL - test_busy_wait
===================================================================
starting test - test_k_sleep
Testing k_thread_create() without cancellation
 thread (q order: 0, t/o: 1000) is running
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0xfffffff5
E: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x20000b10
E: r3/a4:  0x20000498 r12/ip:  0x00000000 r14/lr:  0x0000597b
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0000597a
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000498 (unknown)
E: Halting system
```

`atsamd21_xpro` (Cortex-M0):
```
*** Booting Zephyr OS build v2.3.0-rc1  ***
Running test suite context
===================================================================
starting test - test_kernel_interrupts
PASS - test_kernel_interrupts
===================================================================
starting test - test_kernel_timer_interrupts
SKIP - test_kernel_timer_interrupts
===================================================================
starting test - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread
===================================================================
starting test - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)

    Assertion failed at ../src/main.c:786: test_busy_wait: (rv is true)
 *** thread timed out waiting for k_busy_wait()
FAIL - test_busy_wait
===================================================================
starting test - test_k_sleep
Testing k_thread_create() without cancellation
 thread (q order: 0, t/o: 1000) is running
E: ***** HARD FAULT *****
E: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x20000ad0
E: r3/a4:  0x20000540 r12/ip:  0x00000000 r14/lr:  0x000060f5
E:  xpsr:  0x01000000
E: Faulting instruction address (r15/pc): 0x000057e4
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200004b8 (unknown)
E: Halting system
```

stephanosio, 2020-05-11:
This is the culprit: https://github.com/zephyrproject-rtos/zephyr/commit/b7e363661d3daf4475e67acf1a4d96d39df18da1

stephanosio, 2020-05-11:
Fixed by #25203

andrewboie, 2020-05-11:
> This is the culprit: [b7e3636](https://github.com/zephyrproject-rtos/zephyr/commit/b7e363661d3daf4475e67acf1a4d96d39df18da1)
> 
> cc @andrewboie

argh, this is another really old PR that got merged

PR #25203 (2020-05-11): tests: kernel: context: Fix test_busy_wait timeout
The commit b7e363661d3daf4475e67acf1a4d96d39df18da1 added an additional
busy wait call in the `busy_wait_thread` function -- effectively making
the minimum time required for the thread to exit twice that of the
original implementation.

This commit updates the busy wait thread completion timeout to reflect
that change.

Fixes #25176
Fixes #25194
closed 2020-05-11 by carlescufi