zephyrproject-rtos/zephyr #25180: tests: drivers/i2s/i2s_api: Build failed on 96b_argonkey

erwango, 2020-05-10:
**Describe the bug**
Not  much more info for now

**To Reproduce**
sanitycheck --ninja --warnings-as-errors --runtime-artifact-cleanup -b -p 96b_argonkey -T tests/drivers/i2s

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

henrikbrixandersen, 2020-05-10:
Potentially fixed by #25163.

erwango, 2020-05-12:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/25163

closed 2020-05-12 by erwango