zephyrproject-rtos/zephyr #25182: Raspberry Pi 4B Support

Nukersson, 2020-05-10:
**Is your feature request related to a problem? Please describe.**
[Raspberry Pi4B](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/) support feature.

**Describe the solution you'd like**
Introduce [BCM2711](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) SoC, Cortex-A72 and BSP for [RPI4](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf).

Debugging can be done as it described [here](https://iosoft.blog/2019/01/28/raspberry-pi-openocd/)

**Describe alternatives you've considered**
Maybe [this](https://www.reddit.com/r/raspberry_pi/comments/egwo6z/was_the_broadcom_bcm2711_designed_specifically/) info could be helpful.

**Additional context**
Any suggestions?

stephanosio, 2020-05-11:
Other than "because we can," what would be a practical application for this?

As far as I can see, the main reason for one to choose Raspberry Pi over, for instance, Arduino is to run Linux. I wonder if there is any reason why one would want to run Zephyr on it.

bwasim, 2020-05-11:
> Other than "because we can," what would be a practical application for this?

@stephanosio , I would ask the same question for the Armv8-A port on Zephyr altogether (which is available / usable in the mainline). 

Given that Armv8-A is a high end application processor targeted for products like phones / servers, supports Multi-OS scenarios like Hypervisor etc, does it make sense for Zephyr to even support it ? If it does, then we should have no problem adding support for Pi 4B. 

The way I see it - Zephyr should run on the low-end cores (the Armv7-Ms) or real-time cores (the Armv7-R / Armv8-Rs) and leave the "Cortex-A" series to Linux completely. Thoughts ?

carlescufi, 2020-05-11:
CC @Abhishek-brcm. Any thoughts on this since @bwasim mentions a Broadcom IC?

stephanosio, 2020-05-12:
> I would ask the same question for the Armv8-A port on Zephyr altogether (which is available / usable in the mainline).

I suppose Broadcom is working on their new SoC with specific applications in mind (e.g. IoT applications with high processing power requirements, like computer vision and AI -- all without the overhead of Linux).

In that sense, Raspberry Pi 4 support would not be pointless, but it is not as low footprint/end/price as one would want such systems to be, so I am not really sure of its practical application (with the resources available on Raspberry Pi 4, you might as well just run Linux).

Nonetheless, it would still be nice to have a proof-of-concept Raspberry Pi 4 support so that we have a generally available platform for the ARMv8-A AArch64 arch port validation.

p.s. As for the ARMv7-A arch port, what I described above is exactly the kind of applications I have in mind.

Abhishek-brcm, 2020-05-12:
> CC @Abhishek-brcm. Any thoughts on this since @bwasim mentions a Broadcom IC?

@carlescufi sorry, wouldn't be able to comment here from Broadcom IC point of view, the group designing SoCs and boards for RPi is different, I don't have much knowledge on it. However, I will surely forward this thread to someone- I know works on RPi open source.

carlescufi, 2020-05-27:
> @carlescufi sorry, wouldn't be able to comment here from Broadcom IC point of view, the group designing SoCs and boards for RPi is different, I don't have much knowledge on it. However, I will [...]

Thanks @Abhishek-brcm.

ExtremeGTX, 2020-06-14:
@stephanosio I asked your question before on #general, and i got the following ideas (just for sharing and discussion):
- Zephyr as bootloader (UEFI Boot, OS loading)
- Running zephyr on one A53 core and Linux on the other
- Use a hypervisor that can schedule Zephyr for real-time tasks and Linux for HMI tasks

stephanosio, 2020-06-15:
> * Zephyr as bootloader (UEFI Boot, OS loading)

Why would one want to use Zephyr as a bootloader when there are already other purpose-made open source boot loaders that do much better job (e.g. u-boot, TianoCore)?

> * Running zephyr on one A53 core and Linux on the other

I am not really convinced that this is something desirable. You will be wasting a lot of processing power, which is scarce enough on Raspberry Pi, by dedicating one powerful Cortex-A core to Zephyr.

> * Use a hypervisor that can schedule Zephyr for real-time tasks and Linux for HMI tasks

This is something completely different; it has more to do with developing an ARM hypervisor that can do that than implementing something in Zephyr -- if there is a hypervisor suitable for this purpose, it would certainly be interesting to take a look at.

p.s. If you really want something like this, you should probably look into more purpose-made solutions: e.g. Cortex-M+A (NXP has plenty) or Cortex-R+A (TI just released AM65xx) combo SoC.

ExtremeGTX, 2020-06-15:
I agree with you, these kind of use cases have a lot of alternatives.

For me, I see the Cortex-A support is the first step to turn Zephyr into Linux, which is I don't think it is the goal of Zephyr.

Nukersson, 2020-10-16:
Closed since staled.

closed 2020-10-16 by Nukersson

fkromer, 2021-08-02:
#37134

fkromer, 2021-11-02:
A few months ago the guys from RT-Thread announced RT-Thread Smart. As far as I know it is the first microkernel RTOS which runs on Raspberry Pi. Read more about why they created RT-Thread Smart here: https://opensource.com/article/21/7/rt-thread-smart That‘s exactly why many users would be interested in Zephyr OS on RPi as well.

stephanosio, 2021-11-03:
> That‘s exactly why many users would be interested in Zephyr OS on RPi as well.

Not really. "Supporting Cortex-A targets" and "supporting Raspberry Pi target" are two different things.

"Supporting Cortex-A targets" is certainly both interesting and useful to many users because the Cortex-A family cores tend to be much more powerful than the Cortex-M family cores, and that can be useful for edge computing applications.

In fact, Zephyr RTOS already supports the Cortex-A targets (both ARMv7-A/AArch32 and ARMv8-A/AArch64) for that reason.

As for "supporting Raspberry Pi target," as I have already pointed out in https://github.com/zephyrproject-rtos/zephyr/issues/25182#issuecomment-627042143, you might as well just run Linux with the resources available on Raspberry Pi 4.

The RT-Thread Smart article you linked above also mentions "The compressed RT-Thread Smart kernel is just 217KB, with a root filesystem of 127KB. Typical memory usage is about 2MB," implying its main purpose is to run on the Cortex-A (and other high performance) targets with limited memory resources, which Raspberry Pi 4 certainly is not one.

The only reason I can see why they have decided to support Raspberry Pi 4 is for publicity -- Raspberry Pi 4 is much more accessible than the vendor-provided development boards, so newcomers may find it useful for just "trying out" the OS; but, that is as far as its usefulness goes.

If you want to "try out" the Zephyr RTOS on a Cortex-A target, you can run `qemu_cortex_a9` or `qemu_cortex_a53` on the QEMU included in the Zephyr SDK.

arvid-r, 2021-11-03:
I agree in one sense. This is of course only for trying stuff out, to see how it works and performs. But that's not super easy today with the list of supported Arm64 boards for example. As far as I can tell, NXP's LS1046A board is the only thing that is not an emulation or a FPGA something, and that isn't cheap or available everywhere.

fkromer, 2021-11-03:
> The only reason I can see why they have decided to support Raspberry Pi 4 is for publicity -- Raspberry Pi 4 is much more accessible than the vendor-provided development boards, so newcomers may find it useful for just "trying out" the OS; but, that is as far as its usefulness goes.

Right.

> "Supporting Cortex-A targets" is certainly both interesting and useful to many users because the Cortex-A family cores tend to be much more powerful than the Cortex-M family cores, and that can be useful for edge computing applications.
> 
> In fact, Zephyr RTOS already supports the Cortex-A targets (both ARMv7-A/AArch32 and ARMv8-A/AArch64) for that reason.
>
> ...
> 
> If you want to "try out" the Zephyr RTOS on a Cortex-A target, you can run qemu_cortex_a9 or qemu_cortex_a53 on the QEMU included in the Zephyr SDK.

I see the point. Sticking to already supported Cortex-A boards is probably the better design decision.