zephyrproject-rtos/zephyr #25183: west build error after while "getting started"

lenhart, 2020-05-10:
**Describe the bug**
When following the getting started guide I encounter the error:
```
No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /home/.../zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:69 (assert_exists)
```
which looks similar to #8977 and #23418.
However the toolchain seems to be right and the build tools changed or differ from the above mentioned issues (see below for my west output). I tried the suggested comments there, but most did not yield more information because the tool paths/setup has changed as it seems.

I hope I did not miss anything obvious here, am greatful for help, and will happily provide more information.

**To Reproduce**
follow the [getting started guide](https://docs.zephyrproject.org/latest/getting_started/index.html)

**Expected behavior**
successful build

**Impact**
does not work

**Screenshots or console output**
```
Including boilerplate (Zephyr base (cached)): /home/.../zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/.../zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.2.99 (/home/.../zephyrproject/zephyr)
-- Board: esp32
-- Found west: /home/.../.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/.../zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr (/home/.../zephyr-sdk-0.11.2)
-- Found BOARD.dts: /home/.../zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: /home/.../zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/.../zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/.../zephyrproject/zephyr/Kconfig
Loaded configuration '/home/.../zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/home/.../zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/home/.../zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
CMake Error at /home/.../zephyrproject/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /home/.../zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:65 (assert_exists)
  /home/.../zephyrproject/zephyr/cmake/target_toolchain.cmake:49 (include)
  /home/.../zephyrproject/zephyr/cmake/app/boilerplate.cmake:516 (include)
  /home/.../zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/.../zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)
```

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 20.04
 - Toolchain: Zephyr SDK (esp-idf also installed previously on system, just in case this might influence something)
 - Commit SHA 61c6e2b17247d4bfb7e24b5cf2d6c2c292ea2d5c

**Additional context**
Before that I also had the error message 
`ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored. `
which I fixed w/ the additon of `export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0` to .zephyrrc  as suggested in this [stackoverflow question](https://stackoverflow.com/questions/53825857/error-ld-so-object-libgtk3-nocsd-so-0-from-ld-preload-cannot-be-preloaded)
(is this something common or did I do something wrong? Maybe it could be added to the getting started if it is the first case)

Thanks for the great work :-)

carlescufi, 2020-05-10:
@lenhart it would really help if you wrote what sample you were trying to build for which board, and the exact command-line command you issued.

lenhart, 2020-05-10:
@carlescufi oh, you are absolutely right, I am sorry for that.

I am building for the esp32 w/
```west build -p auto -b esp32 samples/hello_world```

Now I re-read the board page and saw that I missed the board specific setup path (I was just looking for the board name before). Alright, did that now but am greeted with a new error. It finds the zephyr and esp toolchains but still complains that ```Zephyr was unable to find the toolchain.  Is the environment misconfigured?```

Is there a better place to ask questions than here? This is probably not a bug but a problem at my side (although the ```export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0``` was still needed to get so far)

Thanks and best regards!

```
Including boilerplate (Zephyr base (cached)): /home/.../zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/.../zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.2.99 (/home/.../zephyrproject/zephyr)
-- Board: esp32
-- Found toolchain: zephyr (/home/.../zephyr-sdk-0.11.2)
-- Found west: /home/.../.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /home/.../zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: espressif (/home/.../.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf)
CMake Error at /home/.../zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: espressif

  Internal variables:

  CROSS_COMPILE:
  /home/.../.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/bin/xtensa-esp32-elf-

  TOOLCHAIN_HOME:
  /home/.../.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf

Call Stack (most recent call first):
  /home/.../zephyrproject/zephyr/cmake/generic_toolchain.cmake:70 (include)
  /home/.../zephyrproject/zephyr/cmake/app/boilerplate.cmake:502 (include)
  /home/.../zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/.../zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/home/.../zephyrproject/zephyr/build -S/home/.../zephyrproject/zephyr/samples/hello_world -GNinja

```

carlescufi, 2020-05-11:
Not quite sure what is going on, but I see that the xtensa toolchain you are trying to use is not listed [here](https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html#third-party-x-compilers)

Have you tried using the [Zephyr SDK](https://docs.zephyrproject.org/latest/getting_started/installation_linux.html#zephyr-sdk) instead?

>Is there a better place to ask questions than here? 
Sure, join our [Slack](https://tinyurl.com/y5glwylp)

dcpleung, 2020-05-12:
Are you using the Espressif ESP32 toolchain? If so, is the environment `ESPRESSIF_TOOLCHAIN_PATH` set according to https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html?highlight=espressif_toolchain_path ?

lenhart, 2020-05-12:
Thank you both!

@carlescufi : with using the zephyr you mean w/o the esp32 specific environment variables or by replacing `esp32` with `zephyr` again in the .zephyrrc? Then I get the error from the first post.
I shouldn't have to use Crosstool-NG because from the esp32 page it looks as if there is a flavour for the espressif toolchain? Could it be that that is just missing in the 3rd party toolchain page?

@dcpleung yes, I set it according to the documentation. Although it is a bit confusing because the instruction is ```export ESPRESSIF_TOOLCHAIN_PATH="/path/to/xtensa-esp32-elf/"``` and the path is ```/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/``` so you could be inclined to stop after the first occurentce of xtensa-esp32-elf (in which case the error from the first post occurs). Testet w/ both variants though.

I noticed an other issue here: ```-p auto``` seems not to clear everything, does it? I have to manually remove the build folder in order for changes to ~/.zephyrrc to take effect (and source it of course)

dcpleung, 2020-05-12:
The installation path has changed quite a bit since the documentation is first written, so that needs to be updated.

IIRC, `-p always` will clean out the build directory every time.

nashif, 2020-05-18:
This was broken by:

```
commit 299a154fddfee098232fef804a5041e82344d527
Author: Torsten Rasmussen <Torsten.Rasmussen@nordicsemi.no>
Date:   Mon Apr 6 23:15:39 2020 +0200

    cmake: Zephyr sdk package handling

```

@tejlmand ^^

pabigot, 2020-05-18:
I confirm the problem and the introduction of it in 299a154fddfee098232fef804a5041e82344d527.  I do set `ZEPHYR_BASE` in my environment.  The build trace I get is:

```
tirzah[9]$ rm -rf build && west build -b esp32
-- west build: generating a build system
Including boilerplate (Zephyr base): /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake
-- Application: /mnt/nordic/zp/zephyr/samples/hello_world
-- Zephyr version: 2.2.99 (/mnt/nordic/zp/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: esp32
-- Found toolchain: zephyr (/usr/local/zephyr-sdk-0.11.3-beta-1-pr-210)
-- Found west: /home/pab/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /usr/local/zephyr-sdk-0.11.3-beta-1-pr-210/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /mnt/nordic/zp/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /mnt/nordic/zp/zephyr/Kconfig
Loaded configuration '/mnt/nordic/zp/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/mnt/nordic/zp/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/mnt/nordic/zp/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
CMake Error at /mnt/nordic/zp/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /mnt/nordic/zp/zephyr/cmake/compiler/gcc/target.cmake:65 (assert_exists)
  /mnt/nordic/zp/zephyr/cmake/target_toolchain.cmake:49 (include)
  /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake:516 (include)
  /mnt/nordic/zp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /mnt/nordic/zp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/mnt/nordic/zp/zephyr/samples/hello_world/build -S/mnt/nordic/zp/zephyr/samples/hello_world -GNinja -DBOARD=esp32
```

carlescufi, 2020-05-18:
@tejlmand could you please propose a fix?

tejlmand, 2020-05-18:
Looking at it.

tejlmand, 2020-05-19:
Still investigating, but just tried to take the commit BEFORE https://github.com/zephyrproject-rtos/zephyr/commit/299a154fddfee098232fef804a5041e82344d527 which is https://github.com/zephyrproject-rtos/zephyr/commit/6789ecbd5e13cc19e69f25314cf486493df9dde3 . 

And I still get: 
```
> git checkout 299a154~1
HEAD is now at 6789ecbd5e usb: samples: Add USB Audio Headset sample
> west update > /dev/null 
HEAD is now at 542b229 DSP: Integrate CMSIS-DSP 1.8.0 (CMSIS 5.7.0)
HEAD is now at 1fe96f0 samv71: patch: Fix GMAC priority queue register offsets
...
> rm -rf build; west build -b esp32
-- west build: generating a build system
Including boilerplate (Zephyr base): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /projects/github/ncs/zephyr/samples/hello_world
-- Zephyr version: 2.2.99 (/projects/github/ncs/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Board: esp32
-- Found west: /home/tora/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found BOARD.dts: /projects/github/ncs/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: /projects/github/ncs/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /projects/github/ncs/zephyr/Kconfig
Loaded configuration '/projects/github/ncs/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/projects/github/ncs/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
CMake Error at /projects/github/ncs/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /projects/github/ncs/zephyr/cmake/compiler/gcc/target.cmake:65 (assert_exists)
  /projects/github/ncs/zephyr/cmake/target_toolchain.cmake:49 (include)
  /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake:516 (include)
  /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/projects/github/ncs/zephyr/samples/hello_world/build -S/projects/github/ncs/zephyr/samples/hello_world -GNinja -DBOARD=Parsing /projects/github/ncs/zephyr/Kconfig
Loaded configuration '/projects/github/ncs/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/projects/github/ncs/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
CMake Error at /projects/github/ncs/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /projects/github/ncs/zephyr/cmake/compiler/gcc/target.cmake:65 (assert_exists)
  /projects/github/ncs/zephyr/cmake/target_toolchain.cmake:49 (include)
  /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake:516 (include)
  /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /projects/github/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/projects/github/ncs/zephyr/samples/hello_world/build -S/projects/github/ncs/zephyr/samples/hello_world -GNinja -DBOARD=esp32
```
@pabigot can you confirm that this is actually still happening on earlier commits ?

tejlmand, 2020-05-19:
@pabigot  @nashif  To me, it appears that this happened with Zephyr SDK 0.11.1 in this commit:
https://github.com/zephyrproject-rtos/zephyr/commit/5267a5e80976086be0b2188d0f0e3b49b53ae6e1

```
$ git checkout 5267a5e80976086be0b2188d0f0e3b49b53ae6e1
Previous HEAD position was 47ec94a662 ci: Bump docker image and SDK support to 0.11.1
HEAD is now at 5267a5e809 toolchain: Bump min SDK version to 0.11.1
$ rm -rf build; west build -b esp32

 -- west build: build configuration:
       source directory: /projects/github/ncs/zephyr/samples/hello_world
       build directory: /projects/github/ncs/zephyr/samples/hello_world/build (created)
       BOARD: esp32 (origin: command line)
-- west build: generating a build system
-- Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6") 
-- Selected BOARD esp32
-- Found west: /home/tora/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Loading /projects/github/ncs/zephyr/boards/xtensa/esp32/esp32.dts as base
Devicetree header saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing /projects/github/ncs/zephyr/Kconfig
Loaded configuration '/projects/github/ncs/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/projects/github/ncs/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'

warning: ENTROPY_ESP32_RNG (defined at drivers/entropy/Kconfig.esp32:6) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: ENTROPY_GENERATOR (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ENTROPY_ESP32_RNG.html and/or look up
ENTROPY_ESP32_RNG in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

CMake Error at /projects/github/ncs/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /projects/github/ncs/zephyr/cmake/compiler/gcc/target.cmake:69 (assert_exists)
  /projects/github/ncs/zephyr/cmake/target_toolchain.cmake:49 (include)
  /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake:473 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -B/projects/github/ncs/zephyr/samples/hello_world/build -S/projects/github/ncs/zephyr/samples/hello_world -GNinja -DBOARD=esp32
$ 
```

tejlmand, 2020-05-19:
@galak ^^^

tejlmand, 2020-05-19:
@lenhart @pabigot Seems like it doesn't find the C compiler.
Could you try to check the value in the CMakeCache.txt in build folder.
```
$ cat build/CMakeCache.txt |grep CMAKE_C_COMPILER
CMAKE_C_COMPILER:FILEPATH=CMAKE_C_COMPILER-NOTFOUND
```

tejlmand, 2020-05-19:
Zephyr SDK does not contain `xtensa-esp32-elf-gcc`, and hence `ZEPHYR_TOOLCHAIN_VARIANT=zephyr` will not work.

If setting up the `espressif` toolchain according to: https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html?highlight=espressif_toolchain_path#set-up-build-environment and sets `ESP_IDF_PATH` when building, then everything works.

```
$ git rev-parse HEAD
e550302a61a088abe549e7bf4d3a2fd51d8971f3
$ echo $ZEPHYR_TOOLCHAIN_VARIANT
espressif
$ echo $ESPRESSIF_TOOLCHAIN_PATH 
~/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/
$ rm -rf build; west build -b esp32 -- -DESP_IDF_PATH=~/.espressif
-- west build: generating a build system
Including boilerplate (Zephyr base): /projects/github/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: /projects/github/ncs/zephyr/samples/hello_world
-- Zephyr version: 2.3.0-rc1 (/projects/github/ncs/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Board: esp32
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found west: /home/tora/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: espressif (/home/tora/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf)
-- Found BOARD.dts: /projects/github/ncs/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: /projects/github/ncs/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /projects/github/ncs/zephyr/Kconfig
Loaded configuration '/projects/github/ncs/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/projects/github/ncs/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/projects/github/ncs/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/tora/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Cache files will be written to: /home/tora/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /projects/github/ncs/zephyr/samples/hello_world/build
-- west build: building application
[1/104] Preparing syscall dependency handling

[99/104] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
     iram0_0_seg:       12408 B       128 KB      9.47%
     iram0_2_seg:          0 GB      3264 KB      0.00%
     dram0_0_seg:        9728 B       320 KB      2.97%
     drom0_0_seg:          0 GB         8 MB      0.00%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          56 B         8 KB      0.68%
[104/104] Linking C executable zephyr/zephyr.elf
$
```

Closing, as this is working.
I believe the docs could be improved, especially we should consider to add the `espressif` toolchain here:
https://docs.zephyrproject.org/latest/getting_started/toolchain_3rd_party_x_compilers.html
imho.

@lenhart please open a doc issue if you agree.

pabigot, 2020-05-19:
Thanks, I've looked into it further, and I think this is still a bug.

#25441 produces the failure in this PR when set up in this environment:
```
cd /mnt/nordic/zp
. ${HOME}/esp/esp-idf/export.sh
export ZEPHYR_TOOLCHAIN_VARIANT=espressif
export ESPRESSIF_TOOLCHAIN_PATH=${HOME}/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/
export ESP_IDF_PATH=${IDF_PATH}
export ZEPHYR_SDK_INSTALL_DIR=/usr/local/zephyr-sdk
export ZEPHYR_BOARD_ALIASES=${HOME}/.zephyr_board_aliases.cmake
export SIGNING_KEY=${HOME}/pab-mcuboot.pem
cd /mnt/nordic/zp/zephyr
. zephyr-env.sh
export UNCRUSTIFY_CONFIG=${ZEPHYR_BASE}/.uncrustify.cfg
```
The reason seems to be that `ZEPHYR_SDK_INSTALL_DIR` take precedence over `ZEPHYR_TOOLCHAIN_VARIANT`, which appears to have been introduced by the commit identified.

We used to be able to change `ZEPHYR_TOOLCHAIN_VARIANT` and have it take effect.  I believe the proper fix is to only make use of `ZEPHYR_SDK_INSTALL_DIR` if `ZEPHYR_TOOLCHAIN_VARIANT=zephyr`.

tejlmand, 2020-05-19:
@pabigot you're correct. It is a bug.
Thanks for the details regarding this as a mix of Zephyr SDK Tools and 3rd party toolchain.
Fixed here: #25446

PR #25446 (2020-05-19): cmake: fix usage of Zephyr SDK tools with 3rd party toolchains
Fixes: #25183

This commit fixes the issue where combining a 3rd party toolchain with
Zephyr SDK Tools would result in the Zephyr SDK overruling the toolchain
in use.

This is fixed by keeping track of current toolchain variant.
commit c519f6ee (2020-05-19): cmake: fix usage of Zephyr SDK tools with 3rd party toolchains

Fixes: #25183

This commit fixes the issue where combining a 3rd party toolchain with
Zephyr SDK Tools would result in the Zephyr SDK overruling the toolchain
in use.

This is fixed by keeping track of current toolchain variant.
closed 2020-05-19 by carlescufi