zephyrproject-rtos/zephyr #25185: Adding CONFIG_BT_SETTINGS creates errors on bt_hci_core & bt_gatt

eshaulx, 2020-05-10:
I am trying to override BT DIS configs on runtime to update the DIS->SN, DIS->FW version etc...
Therefore, I added the following configs:
`CONFIG_SETTINGS_RUNTIME=y
CONFIG_SETTINGS=y
CONFIG_SETTINGS_NONE=y
CONFIG_BT_SETTINGS=y

CONFIG_BT_GATT_DIS_SETTINGS=y
CONFIG_BT_GATT_DIS_STR_MAX=21`

In addition to that, I am calling settings_load() and override the relevant configs after bt_enable() and before advertising.

the problem is that the above changes only, adding 1 error and a warning:
1. err: bt_gatt: Failed to save Database Hash (err -2)
this cause by this failure on bt_gatt: 
`err = settings_save_one("bt/hash", &db_hash, sizeof(db_hash));`

2. wrn: bt_hci_core: Read Static Addresses command not available
this is also problematic as I sure need the BLE Static address.

I am trying to understand what else need to be added to make it clean & correct.

In another zephyr example I saw (peripheral_dis), the CONFIG_BT_SETTINGS was not configured and settings_load() wasn't called, only the use of settings_runtime_set was there to override these configs of the DIS, I tired that also but the configs were not updated without the CONFIG_BT_SETTINGS, so it's not clear to me how the example works... it was taken from

[image]

**Environment Zephyr NCS fork 2.1.0

fnde-ot, 2020-05-11:
To be able to store settings you need to choose a setting backend or register a custom one.

The example seems to disable backends (CONFIG_SETTINGS_NONE), it was probably done to make it run on posix. It will cause -harmless- errors when gatt uses the settings subsystem to try to persist something.

The warning you mention seems unrelated to settings, looks like the controller does not support the read_static_addrs vendor command.

fnde-ot, 2020-05-11:
Maybe the calls to settings_save_one could check for the -ENOENT return code (meaning there is no setting backend) and change the error message to a warning indicating the setting was not persisted?

eshaulx, 2020-05-11:
thanks @fnde-ot 

- If i'll create a custom BE the warnings about persisting will be removed?

- read_static_addrs is related somehow because I am able to do that without the settings module as I mentioned in my original message, and I need that feature.

eshaulx, 2020-05-11:
> Maybe the calls to settings_save_one could check for the -ENOENT return code (meaning there is no setting backend) and change the error message to a warning indicating the setting was not persisted?

I agree, it's good idea

eshaulx, 2020-05-11:
> To be able to store settings you need to choose a setting backend or register a custom one.
>
> The example seems to disable backends (CONFIG_SETTINGS_NONE), it was probably done to make it run on [...]

thanks @fnde-ot

If i'll create a custom BE the warnings about persisting will be removed?

read_static_addrs is related somehow because I am able to do that without the settings module as I mentioned in my original message, and I need that feature.

joerchan, 2020-05-11:
The warning about "read static address command not available" can be ignored. The host will generate a random one. It will however not be persistently stored unless you have a settings backend.
Alternatively: use `settings_runtime_set` to set `bt/id` and `bt/irk`. Or call `bt_id_create`.

The peripheral_dis sample is using CONFIG_SETTINGS_NONE because it is demonstrating how to set  settings values using `settings_runtime_set`. You will have to enable a proper backend in order to be able to store values persistently
The error from gatt is because the feature CONFIG_BT_GATT_CACHING is enabled and it needs to store the database hash, you should either enable a settings backend or disable that feature.

eshaulx, 2020-05-11:
> The warning about "read static address command not available" can be ignored. The host will generate a random one. It will however not be persistently stored unless you have a settings backend. [...]

Actually I don't want a custom BT ID, I want to use the static one, that can be read without a problem when **_CONFIG_BT_SETTINGS_** is not set.
I don't really understand the need for backend in case I want to make a runtime change only, it doesn't have to be stored persistently but it has to update the original configuration strings to updated ones.

In case it can't be done without a costume BE implementation, I also tried, to implement costume BE, with only **_csi_load_**, on the API, with **_settings_runtime_set_** for the required configs, under that costume load function, but I think I need to implement also the **_csi_save_** interface of the API, should I use **_settings_runtime_get_** under the save function? I am really confused here and the Settings documentation is not really helping my confusion.

Thanx

joerchan, 2020-05-11:
> Actually I don't want a custom BT ID, I want to use the static one, that can be read without a problem when CONFIG_BT_SETTINGS is not set.

CONFIG_BT_SETTINGS shouldn't affect this though. The controller you are using does not have a Vendor Specific (VS) HCI command to read static addresses. Probably disabling CONFIG_BT_HCI_VS_EXT would fix this.

> I don't really understand the need for backend in case I want to make a runtime change only, it doesn't have to be stored persistently but it has to update the original configuration strings to updated ones.

You need to be able to save persistently when using certain Bluetooth features (unless you never power-cycle your board?). You have enabled GATT Caching (CONFIG_BT_GATT_CACHING), which needs to store a database hash persistently. If you don't need this feature then disable this option, otherwise you will need persistent storage.

> In case it can't be done without a costume BE implementation, I also tried, to implement costume BE, with only csi_load, on the API, with settings_runtime_set for the required configs, under that [...]

There are existing backends available. So making a custom backend shouldn't be needed.
See the `samples/bluetooth/peripheral` which uses the NVS backend.

eshaulx, 2020-05-16:
The issue was solved by:
CONFIG_SETTINGS_NONE=y (no BE)
CONFIG_BT_SETTINGS=y
CONFIG_BT_GATT_CACHING=n
disabling CONFIG_BT_HCI_VS_EXT

closed 2020-05-16 by eshaulx