zephyrproject-rtos/zephyr #25187: Generic ethernet packet filtering based on link (MAC) address

lmajewski, 2020-05-10:
**Is your feature request related to a problem? Please describe.**
The problem is to have a generic way to filter (or pass to another interface) packets with a special MAC address (for example link-local as https://elixir.bootlin.com/linux/v5.6-rc7/source/include/linux/etherdevice.h#L66).

Up till now all the filtering tasks are executed in ethernet_recv() (https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/ethernet/ethernet.c#L170) function.

However, there the filtering is based on the packet type (like it is already done for GPTP or LLDP).
Moreover, when the packet address is not equal to the one for the interface it is dropped (as https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/l2/ethernet/ethernet.c#L260)

**Describe the solution you'd like**
I would like to ask about possible solutions to have a "callback" which would in user program
compare the MAC with allowed ones and possibly pass it to other interface (like eth0 to eth1).

**Describe alternatives you've considered**
It seems like possible approaches could be used:
1. Mimic the LLDP code (samples/net/lldp) - but this seems to be one-off and not flexible as this particular example handles the whole packet processing.

2. Use the "Socket NET_MGMT library" - https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/socket_net_mgmt.h - similar to Linux netlink - on the ethernet port, which receives the packet (for example eth0) to setup the mac address for which we want to take the action.
Then add special NET_EVENT to https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/ethernet_mgmt.h , which would call registered callback for each packet with the matching MAC address. 
This option seems like an overkill.

3. Provide a __weak function call to ethernet_recv(), which would be override by user application
defined function.

**Additional context**
Comments/ideas are more than welcome

jukkar, 2020-05-12:
From these the 2. option sounds most generic. User would be able to set some MAC address filters in the system. And then when we have a "match", we would call a user specified callback which could determine what should happen to the packet. @lmajewski is this what you are proposing?

@tbursztyka what do you think?

lmajewski, 2020-05-14:
@jukkar - in my use case (DSA switch):
1. Set the MAC (L2) address for network interface (net_if) - the NET_MGMT library looks like a good mechanism (place) for it. 
2. Then in ethernet_recv() I could trigger proper callback (registered in user's program) for NET_EVENT (NET_FILTER_MAC_ADDR?)

Then are two possible options:
2.1 In the user program's defined callback I would "redirect" this packet to DSA slave network interfaces (net_if lan1..3) and pass it further to the ethernet stack.
Then, the user program would use standard socket API (with SOCK_RAW) to get this "redirected" packet and work on it (in which case the the L2 header shall also be provided).

or

2.2 In the NET_EVENT callback, called when MAC address matches, the user program would handle the packet entirely.

I would prefer approach from point 2.1 as it allows using standard socket API. If not - the approach from 2.2 could be used as a backup plan.

The question is if SOCK_RAW sockets in Zephyr supports such use case ?

jukkar, 2021-03-06:
The DSA support is already merged so it looks like we can close this.

closed 2021-03-06 by jukkar