zephyrproject-rtos/zephyr #25191: tests/drivers/console: drivers.console.semihost can't work

erwango, 2020-05-11:
**Describe the bug**
When running sanitycheck, drivers.console.semihost is run on all cortex-m boards.
Though, it is failed on all boards I could test (various STM32, frdm_k64f_reel_board)
When running with west, test doesn't output anything on console.

**To Reproduce**
Run test with sanitycheck, west 

**Expected behavior**
Test runs successfully

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

stephanosio, 2020-05-11:
While the test can run on real hardware targets, sanitycheck will not be able to validate its output, unless the semihosting console output is somehow redirected to the serial port to which sanitycheck is connected.

I will provide a fix for this.

erwango, 2020-05-11:
@stephanosio Txs for the fix.

About this feature, is that complying will all cortex-m ?
I checked given link: https://developer.arm.com/docs/dui0471/k/what-is-semihosting/what-is-semihosting

And they mention: "ARM processors use the SVC instructions, formerly known as SWI instructions, to make semihosting calls. However, if you are compiling for an ARMv6-M or ARMv7-M, for example a Cortex-M1 or Cortex-M3 processor, semihosting is implemented using the BKPT instruction."

I don't see these cases in `drivers/console/semihost_console.c`

closed 2020-05-11 by carlescufi

stephanosio, 2020-05-11:
> About this feature, is that complying will all cortex-m ?
> I checked given link: https://developer.arm.com/docs/dui0471/k/what-is-semihosting/what-is-semihosting

@erwango Yes, it works on all Cortex-M cores.

https://github.com/zephyrproject-rtos/zephyr/blob/c2a77db6d1a466ce54e05d4027390f7d914c55a2/drivers/console/semihost_console.c#L22

[image]

erwango, 2020-05-11:
@stephanosio Txs for the confirmation!

PR #25195 (2020-05-11): tests: drivers: console: Skip semihosting console device testing
The semihosting console test (`drivers.console.semihost`) is currently
only supported on the ARM Cortex-M QEMU targets.

While running this test on real hardware targets is possible, there is
no standardised way for sanitycheck to validate its output, so we
filter by `CONFIG_QEMU_TARGET` to skip running it on non-QEMU targets
(including physical hardware targets with `--device-testing`).

Fixes #25191