zephyrproject-rtos/zephyr #25194: tests: kernel: context: seems to be failing on Nordic platforms

ioannisg, 2020-05-11:
**Describe the bug**
tests/kernel/context seems to be failing on several Nordic nRF Platforms

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\nrf52dk_nrf52832
3. make
4. See error

**Expected behavior**
Should pass

**Screenshots or console output**
```
*** Booting Zephyr OS build v2.3.0-rc1-9-gb88d79846641  ***
Running test suite context
===================================================================
starting test - test_kernel_interrupts
PASS - test_kernel_interrupts
===================================================================
starting test - test_kernel_timer_interrupts
SKIP - test_kernel_timer_interrupts
===================================================================
starting test - test_kernel_ctx_thread
Testing k_current_get() from an ISR and thread
Testing k_is_in_isr() from an ISR
Testing k_is_in_isr() from a preemptible thread
PASS - test_kernel_ctx_thread
===================================================================
starting test - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)

    Assertion failed at ../src/main.c:786: test_busy_wait: (rv is true)
 *** thread timed out waiting for k_busy_wait()
FAIL - test_busy_wait
===================================================================
starting test - test_k_sleep
Testing k_thread_create() without cancellation
 thread (q order: 0, t/o: 1000) is running
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0xfffffff5
E: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x20000b00
E: r3/a4:  0x20000438 r12/ip:  0x00000000 r14/lr:  0x00005897
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00005896
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000438 (unknown)
E: Halting system

```
**Environment (please complete the following information):**
 - Commit SHA or Version used. v2.3.0 rc1

ioannisg, 2020-05-11:
On nRF5340 the error appears to differ:

```
starting test - test_busy_wait
Thread busy waiting for 20000 usecs
Thread busy waiting completed
Thread busy waiting for 20000 usecs (irqs locked)
Thread busy waiting completed (irqs locked)

    Assertion failed at ../src/main.c:786: test_busy_wait: (rv is true)
 *** thread timed out waiting for k_busy_wait()
FAIL - test_busy_wait
===================================================================
starting test - test_k_sleep
Testing k_thread_create() without cancellation
 thread (q order: 0, t/o: 1000) is running
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0xfffffff5  r1/a2:  0xe000ed00  r2/a3:  0x20000a90
E: r3/a4:  0x20000440 r12/ip:  0x00000000 r14/lr:  0x00005787
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00005786
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000440 (unknown)
E: Halting system

```

Nukersson, 2020-05-11:
Seems to be already reported [here](https://github.com/zephyrproject-rtos/zephyr/issues/25176#issue-615313752)

ioannisg, 2020-05-11:
Yeap, just saw it, and triaged that one, too.

ioannisg, 2020-05-11:
duplicate of #25176

stephanosio, 2020-05-11:
Fixed by #25203

PR #25203 (2020-05-11): tests: kernel: context: Fix test_busy_wait timeout
The commit b7e363661d3daf4475e67acf1a4d96d39df18da1 added an additional
busy wait call in the `busy_wait_thread` function -- effectively making
the minimum time required for the thread to exit twice that of the
original implementation.

This commit updates the busy wait thread completion timeout to reflect
that change.

Fixes #25176
Fixes #25194
closed 2020-05-11 by carlescufi