zephyrproject-rtos/zephyr #25196: tests: portability: cmsis_rtos_v2: fails on several nRF platforms

ioannisg, 2020-05-11:
**Describe the bug**
tests: portability: cmsis_rtos_v2 is hanging when executed on nRF targets

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\nrf52dk_nrf52832
3. make
4. See error

Keeps printing: 

```
periodic_callback (Timer 2) = 75699
periodic_callback (Timer 2) = 75700
periodic_callback (Timer 2) = 75701
periodic_callback (Timer 2) = 75702
periodic_callback (Timer 2) = 75703
periodic_callback (Timer 2) = 75704
periodic_callback (Timer 2) = 75705
periodic_callback (Timer 2) = 75706
periodic_callback (Timer 2) = 75707

```

**Expected behavior**
Should pass

**Environment (please complete the following information):**
 - Commit SHA or Version used: Zephyr v2.3.0-rc1

**Additional context**
Add any other context about the problem here.

vanti, 2020-05-12:
Same output seen printed out indefinitely on cc1352r1_launchxl.

carlescufi, 2020-05-18:
@ioannisg and @pabigot could you please spend some time on this one?
cc @andyross

pabigot, 2020-05-18:
Broken by 10d15d185b63f6616bbb7c9975dff15d52464b11, not  the same problem as #24660.  Here the issue is that the periodic timer handler takes longer to run than the interval, and starves the osDelay() that would turn off the timer.  This is because the default `PERIOD_TICKS=50` on most boards corresponds to 5 ms, but on Nordic it corresponds to 1.5 ms, which is too short to complete the `TC_PRINT` call in the handler.  Fix in #25415

closed 2020-05-19 by carlescufi

chen-png, 2020-05-20:
for nrf platforms, the tests/portability/cmsis_rtos_v2 is passed, **but it seems that causing it failed on up_squared board.**

*** Booting Zephyr OS build v2.3.0-rc1-175-gec9a4139834a  ***
Running test suite test_cmsis_v2_apis

starting test - test_kernel_apis
PASS - test_kernel_apis

starting test - test_delay
PASS - test_delay

starting test - test_thread_apis
PASS - test_thread_apis

starting test - test_thread_prio_dynamic
PASS - test_thread_prio_dynamic

**starting test - test_messageq
E: Page fault at address 0xd25ff8 (error code 0x3)
E: Access violation: supervisor thread not allowed to write**
E: PML4E: 0x0000000000d15827 Writable, User, Execute Enabled
E: PDPTE: 0x0000000000d14827 Writable, User, Execute Enabled
E:   PDE: 0x0000000000d0d023 Writable, Supervisor, Execute Enabled
E:   PTE: 0x8000000000d25001 Read-only, Supervisor, Execute Disable
E: RAX: 0x00000000001a4c06 RBX: 0x0000000000d1cd60 RCX: 0x000000000000000c RDX: 0x0000000000d26030
E: RSI: 0x0000000000d1cd60 RDI: 0x0000000000d1dbc8 RBP: 0x0000000000d1cd60 RSP: 0x0000000000d26000
E:  R8: 0x0000000000d26030  R9: 0x0000000000000000 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000001 R13: 0x0000000000d1dbc8 R14: 0x0000000000d1dbc8 R15: 0x0000000000000000
E: RSP: 0x0000000000d26000 RFLAGS: 0x0000000000010007 CS: 0x0018 CR3: 0x0000000000115000
E: RIP: 0x0000000000103f93
**E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x0000000000d18f90 (send_thread)
E: Halting system**

pabigot, 2020-05-20:
@chen-png please open another issue for that failure, which could instead be due to unrelated recent changes to that platform.

chen-png, 2020-05-21:
yes, i reported another issue #25507

PR #25415 (2020-05-19): tests: cmsis_rtos_v2: adjust timer period to avoid starvation
Most boards run with 10 kHz ticks producing a period of 5 ms for 50
tick interval used in the timer periodic test.  On Nordic 50 ticks
corresponds to 1.5 ms which is too short to complete the TC_PRINT()
call within the handler, causing the periodic timer to starve the
osDelay that would turn off the timer.

Adjust the period to be at least 5 ms or 50 ticks to avoid this
problem while not breaking other platform with slower tick rates.

Fixes #25196