zephyrproject-rtos/zephyr #25200: Build error in Sample App for OpenThread NCP

vijayrao1988, 2020-05-11:
**Describe the bug**
When attempting to build the OpenThread NCP sample app for Silicon Labs Thunderboard Sense board.
Building and flashing other sample apps like blinky and hello-world work fine. The build error refers to CONFIG_NET_CONFIG_IEEE802154_DEV_NAME being undeclared. Is this supposed to supplied as a build argument?

**To Reproduce**
Steps to reproduce the behavior:
Command:
west build -p -b efr32mg_sltb004a samples/net/openthread/ncp

Exits with the following error:
/home/blitz/zephyrproject/zephyr/subsys/net/lib/openthread/platform/radio.c:185:33: error: 'CONFIG_NET_CONFIG_IEEE802154_DEV_NAME' undeclared (first use in this function)
  185 |  radio_dev = device_get_binding(CONFIG_NET_CONFIG_IEEE802154_DEV_NAME);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/blitz/zephyrproject/zephyr/subsys/net/lib/openthread/platform/radio.c:185:33: note: each undeclared identifier is reported only once for each function it appears in
[562/617] Building C object modules/mb..._crypto__mbedtls.dir/library/rsa.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/blitz/zephyrproject/zephyr/build

**Expected behavior**
A sample build of the OpenThread NCP would demonstrate capability of Zephyr running simple communications over the Thread stack.

**Impact**
As I am unable to run this quick check before we move out from Silabs stack, it is stopping me from convincing my team to move to Zephyr as our long term dev platform for our OpenThread based IoT devices.

**Screenshots or console output**
blitz@workdesk:~/zephyrproject/zephyr$ west build -p -b efr32mg_sltb004a samples/net/openthread/ncp
-- west build: making build dir /home/blitz/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/blitz/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/blitz/zephyrproject/zephyr/samples/net/openthread/ncp
-- Zephyr version: 2.2.99 (/home/blitz/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter 
-- Board: efr32mg_sltb004a
-- Found west: /home/blitz/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
-- Found dtc: /home/blitz/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: zephyr (/home/blitz/zephyr-sdk-0.11.2)
-- Found BOARD.dts: /home/blitz/zephyrproject/zephyr/boards/arm/efr32mg_sltb004a/efr32mg_sltb004a.dts
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
efr32mg_sltb004a.dts.pre.tmp:77.26-85.5: Warning (spi_bus_bridge): /soc/usart@40010800: node name for SPI buses should be 'spi'
  also defined at efr32mg_sltb004a.dts.pre.tmp:302.9-320.3
<stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'
-- Generated zephyr.dts: /home/blitz/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/blitz/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/blitz/zephyrproject/zephyr/samples/net/openthread/ncp/Kconfig
Loaded configuration '/home/blitz/zephyrproject/zephyr/boards/arm/efr32mg_sltb004a/efr32mg_sltb004a_defconfig'
Merged configuration '/home/blitz/zephyrproject/zephyr/samples/net/openthread/ncp/prj.conf'
Configuration saved to '/home/blitz/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/blitz/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'

warning: TEST_RANDOM_GENERATOR (defined at boards/shields/esp_8266/boards/sam4e_xpro.defconfig:17,
subsys/random/Kconfig:8) was assigned the value 'y' but got the value 'n'. Check these unsatisfied
dependencies: ((BOARD_SAM4E_XPRO && NETWORKING && SHIELD_ESP_8266) || !ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: SHELL_STACK_SIZE (defined at subsys/shell/Kconfig:20) was assigned the value '3072' but got
the value ''. Check these unsatisfied dependencies: SHELL (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SHELL_STACK_SIZE.html and/or look up
SHELL_STACK_SIZE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/blitz/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/blitz/.cache/zephyr
[... 4 lines cut ...]
-- west build: building application
[1/617] Preparing syscall dependency handling

[460/617] Building C object zephyr/sub...p__l2__openthread.dir/openthread.c.obj
/home/blitz/zephyrproject/zephyr/subsys/net/l2/openthread/openthread.c:85:13: warning: 'ipv6_addr_event_handler' defined but not used [-Wunused-function]
   85 | static void ipv6_addr_event_handler(struct net_mgmt_event_callback *cb,
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/blitz/zephyrproject/zephyr/subsys/net/l2/openthread/openthread.c:78:39: warning: 'ip6_addr_cb' defined but not used [-Wunused-variable]
   78 | static struct net_mgmt_event_callback ip6_addr_cb;
      |                                       ^~~~~~~~~~~
[555/617] Building C object zephyr/sub...es/openthread_platform.dir/radio.c.obj
FAILED: zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj 
ccache /home/blitz/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.2.0-2377-gf95c349c36ab -DEFR32MG12P332F1024GL125 -DKERNEL -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/silabs_exx32/efr32mg12p -I../subsys/settings/include -I/home/blitz/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/blitz/zephyrproject/modules/hal/silabs/gecko/Device/SiliconLabs/EFR32MG12P/Include -I/home/blitz/zephyrproject/modules/hal/silabs/gecko/emlib/inc -Imodules/openthread/build/etc/cmake -I/home/blitz/zephyrproject/modules/lib/openthread/etc/cmake -I/home/blitz/zephyrproject/modules/lib/openthread/include -isystem ../lib/libc/minimal/include -isystem /home/blitz/zephyrproject/modules/lib/openthread/zephyr/../include -isystem /home/blitz/zephyrproject/modules/lib/openthread/zephyr/../examples/platforms -isystem /home/blitz/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /home/blitz/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -imacros/home/blitz/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/blitz/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/blitz/zephyrproject/zephyr/samples/net/openthread/ncp=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/blitz/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/blitz/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj -MF zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj.d -o zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj   -c /home/blitz/zephyrproject/zephyr/subsys/net/lib/openthread/platform/radio.c
/home/blitz/zephyrproject/zephyr/subsys/net/lib/openthread/platform/radio.c: In function 'platformRadioInit':
[... cut ...]
/home/blitz/zephyrproject/zephyr/subsys/net/lib/openthread/platform/radio.c:185:33: note: each undeclared identifier is reported only once for each function it appears in
[562/617] Building C object modules/mb..._crypto__mbedtls.dir/library/rsa.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/blitz/zephyrproject/zephyr/build

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.4 LTS
 - Toolchain Zephyr-SDK-0.11.2

carlescufi, 2020-05-11:
@vijayrao1988 I am not sure 802.15.4 is supported on Silicon Labs devices due to the requirement for a binary module. @mnkp can I'm sure expand on that a bit.

carlescufi, 2020-05-11:
>The build error refers to CONFIG_NET_CONFIG_IEEE802154_DEV_NAME being undeclared. Is this supposed to supplied as a build argument?

No, this is a [Kconfig option](https://docs.zephyrproject.org/latest/guides/kconfig/index.html). Looking at the list of [supported drivers](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/ieee802154/Kconfig#L29) I don't see anything related to Silicon Labs. Likely because of [this issue](https://github.com/zephyrproject-rtos/zephyr/issues/17005)

carlescufi, 2020-05-12:
Closing since this is not supported on SiLabs boards. Please follow [this issue](https://github.com/zephyrproject-rtos/zephyr/issues/17005) to get updates on the binary blob discussion.

closed 2020-05-12 by carlescufi