zephyrproject-rtos/zephyr #25215: enable modules to append to $DTS_ROOT

drandreas, 2020-05-11:
**Is your enhancement proposal related to a problem? Please describe.**
Drivers should also be placeable in modules, e.g. `modules/drv/tpm-tis-spi`. This works for source only drivers but breaks as soon as bindings i.e. `dts/bindings/infineon,tpm.yaml` are needed. 
*Note*: If the dts folder is coped to the app folder the build succeeds.

**Describe the solution you'd like**
Add a function e.g. `zephyr_dts_directories` to `extensions.cmake` so a module can register it's. dts files.

**Describe alternatives you've considered**
```
list(APPEND
  DTS_ROOT
  ${APPLICATION_SOURCE_DIR}
  ${BOARD_DIR}
  ${ZEPHYR_BASE}
  )
```
in ` cmake/dts.cmake` could be extended to also search within all modules.

arvid-r, 2020-05-11:
Is this different than out-of-tree drivers in general?
You can build out-of-tree drivers by setting DTS_ROOT=/tmp/zephyr
with the following file structure:
```
/tmp/zephyr/include/drivers
/tmp/zephyr/dts
/tmp/zephyr/dts/bindings
/tmp/zephyr/drivers
```

carlescufi, 2020-05-11:
Indeed, we allow extending `DTS_ROOT` already, be it by a module or application or anything else.

drandreas, 2020-05-11:
With out of tree builds I'm used to entering an absolute path. But having to type 
`west build -b frdm_k64f my-app  -- -DDTS_ROOT=${PWD}/modules/drv/tpm-tis-spi` seems to be cumbersome especially since west handles module detection for all other cases very well.

carlescufi, 2020-05-11:
> With out of tree builds I'm used to entering an absolute path. But having to type
> `west build -b frdm_k64f my-app -- -DDTS_ROOT=${PWD}/modules/drv/tpm-tis-spi` seems to be cumbersome especially since west handles module detection for all other cases very well.

Wouldn't it be possible to set `DTS_ROOT` in your module's `CMakeLists.txt` with something like:
`list(APPEND DTS_ROOT ${MY_DIR})`

drandreas, 2020-05-11:
According to: https://cmake.org/cmake/help/v3.17/command/list.html
"The list subcommands APPEND, INSERT, FILTER, PREPEND, POP_BACK, POP_FRONT, REMOVE_AT, REMOVE_ITEM, REMOVE_DUPLICATES, REVERSE and SORT may create new values for the list within the current CMake variable scope. Similar to the set() command, the LIST command creates new variable values in the current scope, even if the list itself is actually defined in a parent scope. To propagate the results of these operations upwards, use set() with PARENT_SCOPE, set() with CACHE INTERNAL, or some other means of value propagation."

Never the less I've tried:
```
list(APPEND DTS_ROOT 
  ${CMAKE_CURRENT_SOURCE_DIR}
)

if(CONFIG_TPM_TIS_SPI)
  zephyr_library(tpm-tis-spi)

  zephyr_include_directories(
    include
  )

  zephyr_library_sources(
    src/tpm-tis-spi.c
  )
endif()
```

and it works. I'm puzzled...

drandreas, 2020-05-11:
... it must have been a fluke on my machine. build on CI are all failing.

arvid-r, 2020-05-12:
You should be aware of #23825 also. Don't know if that's a problem here though

mbolivar-nordic, 2020-05-12:
@tejlmand what would you think about making the `_ROOT` variables target properties of some convenient zephyr target instead for use cases like this? Is it possible? (I'm assuming we could also support the existing ROOT variables for command-line use and backwards compatibility and just append the property value onto them).

drandreas, 2020-05-12:
I think its an ordering issue. I have added some debug messages:
```
$ west diff
=== diff for manifest (my-app):
=== diff for cmsis (modules/hal/cmsis):
=== diff for hal-nxp (modules/hal/nxp):
=== diff for mcuboot (bootloader/mcuboot):
=== diff for mbedtls (modules/crypto/mbedtls):
=== diff for segger (modules/debug/segger):
=== diff for littlefs (modules/fs/littlefs):
=== diff for zephyr (zephyr):
diff --git zephyr/cmake/dts.cmake zephyr/cmake/dts.cmake
index a2203d5af2..e243344f5a 100644
--- zephyr/cmake/dts.cmake
+++ zephyr/cmake/dts.cmake
@@ -90,6 +90,7 @@ if(SUPPORTS_DTS)
         dts
         )
       set(full_path ${dts_root}/${dts_root_path})
+      message(${full_path})
       if(EXISTS ${full_path})
         list(APPEND
           DTS_ROOT_SYSTEM_INCLUDE_DIRS
=== diff for fdp (modules/lib/fdp):
=== diff for tpm-tis-spi (modules/drv/tpm-tis-spi):
diff --git modules/drv/tpm-tis-spi/CMakeLists.txt modules/drv/tpm-tis-spi/CMakeLists.txt
index 53136dc..fa8fb26 100644
--- modules/drv/tpm-tis-spi/CMakeLists.txt
+++ modules/drv/tpm-tis-spi/CMakeLists.txt
@@ -1,3 +1,9 @@
+message("appending ${CMAKE_CURRENT_SOURCE_DIR} to list")
+list(APPEND DTS_ROOT
+  ${CMAKE_CURRENT_SOURCE_DIR}
+)
+message("appending ${CMAKE_CURRENT_SOURCE_DIR} to list done")
+
 if(CONFIG_TPM_TIS_SPI)
   zephyr_library(tpm-tis-spi)
```

Build Output on my Mac:
```
$ cmake --version
cmake version 3.17.1

$ west build -b frdm_k64f my-app/
-- west build: generating a build system
-- Application: /data/my-app/my-app
-- Zephyr version: 2.2.99 (/data/my-app/zephyr)
-- Found Python3: /opt/local/bin/python3.8 (found suitable exact version "3.8.2") found components: Interpreter 
-- Board: frdm_k64f
-- Found west: /usr/local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found toolchain: gnuarmemb (/opt/local/gcc-arm-none-eabi-8-2019-q3-update)
-- Found BOARD.dts: /data/my-app/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts
-- Found devicetree overlay: /data/my-app/my-app/boards/frdm_k64f.overlay
/data/my-app/my-app/include
/data/my-app/my-app/dts/common
/data/my-app/my-app/dts/arm
/data/my-app/my-app/dts
/data/my-app/zephyr/boards/arm/frdm_k64f/include
/data/my-app/zephyr/boards/arm/frdm_k64f/dts/common
/data/my-app/zephyr/boards/arm/frdm_k64f/dts/arm
/data/my-app/zephyr/boards/arm/frdm_k64f/dts
/data/my-app/zephyr/include
/data/my-app/zephyr/dts/common
/data/my-app/zephyr/dts/arm
/data/my-app/zephyr/dts
-- Generated zephyr.dts: /data/my-app/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /data/my-app/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /data/my-app/zephyr/Kconfig
[... 11 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /data/my-app/build
-- west build: building application
[1/323] Preparing syscall dependency handling

[52/323] Building C object modules/tpm-tis-spi/CMakeFiles/..__modules__drv__tpm-tis-spi.dir/src/tpm-tis-spi.c.obj
FAILED: modules/tpm-tis-spi/CMakeFiles/..__modules__drv__tpm-tis-spi.dir/src/tpm-tis-spi.c.obj 
/opt/local/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc -DBOARD_FLASH_SIZE=CONFIG_FLASH_SIZE -DBUILD_VERSION=zephyr-v2.2.0-2697-gc308f8069f82 -DCPU_MK64FN1M0VLL12 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DNDEBUG -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/data/my-app/zephyr/include -Izephyr/include/generated -I/data/my-app/zephyr/soc/arm/nxp_kinetis/k6x -I/data/my-app/zephyr/lib/libc/newlib/include -I/data/my-app/zephyr/lib/fnmatch/. -I/data/my-app/zephyr/subsys/net/lib/sockets/. -I/data/my-app/zephyr/subsys/net/lib/dns/. -I/data/my-app/zephyr/subsys/net/l2 -I/data/my-app/modules/hal/cmsis/CMSIS/Core/Include -I/data/my-app/modules/hal/nxp/mcux/devices/MK64F12 -I/data/my-app/modules/hal/nxp/mcux/components/phyksz8081 -I/data/my-app/modules/hal/nxp/mcux/drivers/kinetis/. -I/data/my-app/modules/lib/fdp/zephyr/../include -I/data/my-app/modules/drv/tpm-tis-spi/include -I/data/my-app/modules/lib/suef/import/include -Imodules/suef/modules/suef/include -I/data/my-app/modules/crypto/mbedtls/include -I/data/my-app/modules/crypto/mbedtls/configs -Os -imacros/data/my-app/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/data/my-app/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/data/my-app/my-app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/data/my-app/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/data/my-app=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT modules/tpm-tis-spi/CMakeFiles/..__modules__drv__tpm-tis-spi.dir/src/tpm-tis-spi.c.obj -MF modules/tpm-tis-spi/CMakeFiles/..__modules__drv__tpm-tis-spi.dir/src/tpm-tis-spi.c.obj.d -o modules/tpm-tis-spi/CMakeFiles/..__modules__drv__tpm-tis-spi.dir/src/tpm-tis-spi.c.obj   -c /data/my-app/modules/drv/tpm-tis-spi/src/tpm-tis-spi.c
In file included from /data/my-app/zephyr/include/arch/arm/aarch32/arch.h:20,
                 from /data/my-app/zephyr/include/arch/cpu.h:19,
                 from /data/my-app/zephyr/include/kernel_includes.h:39,
                 from /data/my-app/zephyr/include/kernel.h:17,
[... cut ...]
zephyr/include/generated/devicetree_unfixed.h:4518:34: error: 'DT_N_S_soc_S_spi_4002c000_S_tpm_0_P_spi_max_frequency' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_spi_4002c000_S_tpm_0_P_compatible_LEN'?
 #define DT_N_INST_0_infineon_tpm DT_N_S_soc_S_spi_4002c000_S_tpm_0
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/my-app/zephyr/include/devicetree.h:1574:38: note: in definition of macro 'DT_CAT'
 #define DT_CAT(node_id, prop_suffix) node_id##prop_suffix
                                      ^~~~~~~
/data/my-app/zephyr/include/devicetree.h:1188:34: note: in expansion of macro 'DT_PROP'
...
```

mbolivar-nordic, 2020-05-12:
> I think its an ordering issue. I have added some debug messages:

In other words, you're suggesting moving module processing to happen before DT is run?

I think that might introduce problems; DT runs very early on purpose because it's used for a variety of other things, but I'm not sure. Would be curious what @tejlmand thinks.

drandreas, 2020-05-12:
Scrolling back, @carlescufi sed:
> Indeed, we allow extending `DTS_ROOT` already, be it by a module or application or anything else.

If this was true for earlier versions and it its broken now we might need to fix it. 

Else I'm currently leaning towards extending `zephyr/module.yml` with a dts-base-root param and having west generate `-DDTS_ROOT=` .  A third option would be to populate `DTS_ROOT` from `ZEPHYR_MODULES` but that could have a performance impact.

mbolivar-nordic, 2020-05-12:
> A third option would be to populate `DTS_ROOT` from `ZEPHYR_MODULES` but that could have a performance impact.

My guess would be that it will be minimal. I like this solution as it's nice and clean. @tejlmand ?

tejlmand, 2020-05-13:
> In other words, you're suggesting moving module processing to happen before DT is run?
>
> I think that might introduce problems; DT runs very early on purpose because it's used for a variety of other [...]

The reason for the module to be processed where they are, is because they might rely on either DTS or Kconfig settings, so they must be loaded AFTER DT and Kconfig has been processed.

tejlmand, 2020-05-13:
> > A third option would be to populate `DTS_ROOT` from `ZEPHYR_MODULES` but that could have a performance impact.
> 
> My guess would be that it will be minimal. I like this solution as it's nice and clean. @tejlmand ?

I'm currently doing some thoughts regarding a more generic solution for downstream users.
As we now have Zephyr CMake package, I consider a way that downstream users may create a form of Zephyr extension package.
And that package can then populate any variable they like, such as DTS_ROOT, oot board folders, etc.

Would like a generic solution, instead of doing this on case-by-case.

tejlmand, 2020-05-18:
@drandreas I created this: https://github.com/zephyrproject-rtos/zephyr/pull/25399 which allows for creation on an Extension package and setting up / extending any values.

Does this also cover your needs ?

drandreas, 2020-05-18:
It is not quite what I had expected but I think I can make it work. I will try to write a `zephyr_dts_root_sarch_extension`. It will add all modules/drv/* it finds to DTS_ROOT.

At first glance this should be doable in a couple of lines. I will report back asap.

drandreas, 2020-05-20:
@tejlmand Thank for your adding the extension feature. I've tested it with the code below.
```
# The code below extends the DTS_ROOT-List by all folders found in
# modules/drv enabling us to package dts bindings in driver modules.
FILE(GLOB drivers LIST_DIRECTORIES true ${ZEPHYR_BASE}/../modules/drv/*)
FOREACH(driver ${drivers})
  IF(IS_DIRECTORY ${driver})
    LIST(APPEND DTS_ROOT ${driver})
  ENDIF()
ENDFOREACH()
```

tejlmand, 2020-05-25:
> @tejlmand Thank for your adding the extension feature. I've tested it with the code below.
>
> ```
> # The code below extends the DTS_ROOT-List by all folders found in
> # modules/drv enabling us to [...]

@drandreas Great you found a solution that works in your case, however, as a generic / upstream Zephyr approach, then this would be a to much wildcard.

Using same Zephyr workspace for both up/downstream and just switching manifest repo, and run `west update` could end up including modules not supposed to be included.

But thanks for sharing your solution, as others might find it useful.

tejlmand, 2020-05-26:
Re-opnened as not fixed.

drandreas, 2020-05-26:
> Using same Zephyr workspace for both up/downstream and just switching manifest repo, and run west update could end up including modules not supposed to be included.

I might not see the full picture but my assumption is that `DTS_ROOT` is only responsible for providing .dtsi and dts-bindings having to many of those in the environment should not have any effect until they are "enabled" by including them in DTS oder DTS-Overlay

tejlmand, 2020-05-27:
Note: https://github.com/zephyrproject-rtos/zephyr/pull/25575 does not cover this issue completely.

#25575 is a general hook, allowing users downstream to do any kind of build configuration they like, using CMake. (which is also why that PR is not marked as fixing this enhancement suggestion)

The DTS_ROOT and BOARD_ROOT for modules is a specific use-case that is also a valueable enhancement for Zephyr project itself, as seen here: #25153

tejlmand, 2020-05-27:
> I might not see the full picture but my assumption is that DTS_ROOT is only responsible for providing .dtsi and dts-bindings having to many of those in the environment should not have any effect [...]

I see this enhancement as valid in itself, and should be done together with `BOARD_ROOT` support.
But let my try to give a practical example why this could have side effects, if including your proposal directly in to Zephyr.

Today, it is possible to use a single workspace for multiple projects, and because the `west list` is used for module detection, only modules in that list will be included.

As example:
```
/p/g/n/zephyr> west config manifest.path
zephyr
/p/g/n/zephyr> west list nrf
FATAL ERROR: unknown project name/path: nrf
  Hint: use "west list" to list all projects.
p/g/n/zephyr> west config manifest.path nrf
/p/g/n/zephyr> west list nrf
manifest     nrf                          HEAD                                     N/A
/p/g/n/zephyr>
```
in this case, you see that depending on my `west config` setting, I can easily switch between two manifest versions, and when `manifest.path==zephyr`, then `nrf` is not know, but it is when `manifest.path==nrf`.
So imagine we blindly did a dts and board inclusion, then adding `nrf` (or any other repo) as a `DTS_ROOT` or `BOARD_ROOT` could result in nrf-repo boards being available for a pure Zephyr build, which would be wrong in this case.
And if there are collissions between dts files, cause `manifest.path==zephyr` it could use `moduleA` but `manifest.path=nrf` another downstream version moduleB could be in use instead of moduleA, then how to know which version to use ?
And what if there are collisions ?

I know, a lot of users are having simple setups and workflows, but we need to be careful on not introducing hard to trace issues that occurs when sourcing files just because they are present in a folder. We need to make it controlled what is being included, as it is when we rely on `west list` or user provided `-DZEPHYR_MODULES`.

drandreas, 2020-06-20:
Thank you for explaining your thoughts in detail. Now I do understand the intentions in not supporting an auto-include routine. I consider myself a Zepher novice. Therefore I can’t argue in favor or against a multi-project-workspace yet.

For the sake of simplicity I've decided to complete my [Zepher TPM2 PoC](https://github.com/drandreas/zephyr-tpm2-poc) with the interface added with #25575. I will happily switch over to the new interface when it becomes available. For the time being I added a [warning](https://github.com/drandreas/dts-search-extension) about possible side effects if it is used for anything else then dts bindings.

tejlmand, 2020-07-08:
@drandreas you can cheok out this PR: #26715

drandreas, 2020-07-12:
@tagunil Thank you for the notification.

I've tested your PR with the following driver-config:
```
  cmake: zephyr
  kconfig: zephyr/Kconfig
  settings:
    dts_root: .
```

This is exactly what I was hoping for. Nice work.

PR #26715 (2020-08-05): BOARD, SOC, DTS, ARCH ROOT in Zephyr modules.
This PR contains a series of commits that:
- Cleanup usage of SOC dir.
- Add support for a list of SOC_ROOTs, instead of only a single root
- Kconfig handling of multiple SOC_ROOTs
- Add support for a list of ARCH_ROOTs, instead of only a single root
- Minor fixes where code were wrongly using `${ZEPHYR_BASE}/arch` which doesn't work with out-of-tree archs, instead the correct `${ARCH_DIR}` is now used.

And then finally, support for specifying BOARD, DTS, ARCH, and SOC root in Zephyr modules, `zephyr/module.yml` files.

This fixes: #25215
commit 25d57ba5 (2020-08-05): modules: adding support for ROOTs definitions in zephyr/module.yml

Fixes: #25215

This commit introduces the possibility of defining ROOTs in a Zephyr
module and have it automatically appended to list of other ROOTs.
Supported with this commit:
- BOARD_ROOT
- SOC_ROOT
- DTS_ROOT
- ARCH_ROOT

In order to support this in Zephyr module files, the detection of west
has been moved to dedicated west.cmake file and included immediately
after python.cmake.

Also the inclusion of zephyr_modules.cmake has moved before first use
of BOARD_ROOT.
closed 2020-08-05 by nashif