zephyrproject-rtos/zephyr #25231: net.offload test fails on atsame54_xpro

stephanosio, 2020-05-12:
**Describe the bug**
The `tests/net/checksum_offload/net.offload` test fails on the `atsame54_xpro` board.

**To Reproduce**
Build and run `tests/net/checksum_offload/net.offload` on `atsame54_xpro`.

**Expected behavior**
Test passes

**Screenshots or console output**
```
I: MAC: 02:04:25:35:be:29
I: Queue 0 activated
I: Soft Reset of ETH PHY
I: PHYID: 0x221561 at addr: 0
W: You have 2 IPv6 net_if addresses but 3 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.
W: You have 2 IPv4 net_if addresses but 3 network interfaces
W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.
*** Booting Zephyr OS build v2.3.0-rc1-16-ged2d263e0c23  ***
Running test suite net_chksum_offload_test
===================================================================
starting test - eth_setup
Assertion failed at WEST_TOPDIR/zephyr/tests/net/checksum_offload/src/main.c:367: eth_setup: (ud.eth_if_count not equal to sizeof(eth_interfaces) / sizeof(void *))
Invalid number of interfaces (3 vs 2)
FAIL - eth_setup
===================================================================
starting test - address_setup
PASS - address_setup
===================================================================
starting test - tx_chksum_offload_disabled_test_v6
PASS - tx_chksum_offload_disabled_test_v6
===================================================================
starting test - tx_chksum_offload_disabled_test_v4
PASS - tx_chksum_offload_disabled_test_v4
===================================================================
starting test - tx_chksum_offload_enabled_test_v6
PASS - tx_chksum_offload_enabled_test_v6
===================================================================
starting test - tx_chksum_offload_enabled_test_v4
PASS - tx_chksum_offload_enabled_test_v4
===================================================================
starting test - rx_chksum_offload_disabled_test_v6
PASS - rx_chksum_offload_disabled_test_v6
===================================================================
starting test - rx_chksum_offload_disabled_test_v4
PASS - rx_chksum_offload_disabled_test_v4
===================================================================
starting test - rx_chksum_offload_enabled_test_v6
PASS - rx_chksum_offload_enabled_test_v6
===================================================================
starting test - rx_chksum_offload_enabled_test_v4
PASS - rx_chksum_offload_enabled_test_v4
===================================================================
Test suite net_chksum_offload_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: ed2d263e0c233df015448de8d45a0fae359afc8b

jukkar, 2020-05-13:
> Assertion failed at WEST_TOPDIR/zephyr/tests/net/checksum_offload/src/main.c:367: eth_setup: (ud.eth_if_count not equal to sizeof(eth_interfaces) / sizeof(void *))
> Invalid number of interfaces (3 vs 2)

The reason for this is that the test creates two network interfaces itself, and then one extra network interface comes from the board itself. So the test actually was not expecing to be run on a board that has a network interface (Ethernet in this case). I have changed the checks in some earlier commits like aaaf20e61535767dde0a7971e4709264372d009c and b4dae007410d7a6cee9b6ca42686cb967f82e836.
So we could probably change the check in the test to (not tested)
```
	zassert_true(ud.eth_if_count >= sizeof(eth_interfaces) / sizeof(void *),
		      "Invalid number of interfaces (%d vs %d)\n",
		      ud.eth_if_count,
		      sizeof(eth_interfaces) / sizeof(void *));
```

stephanosio, 2020-07-02:
Same issue as https://github.com/zephyrproject-rtos/zephyr/issues/25226#issuecomment-652787043

Fixed by #26593

PR #26593 (2020-07-02): drivers: ethernet: sam_gmac: Add SOC_FAMILY_SAM0 dependency
The Atmel GMAC Ethernet driver may be used by both the SAM series
(e.g. SAM E70) and SAM0 series (e.g. SAM E54) SoCs.

Fixes #25226 (see https://github.com/zephyrproject-rtos/zephyr/issues/25226#issuecomment-652787043)
Fixes #25231
closed 2020-07-02 by nashif