zephyrproject-rtos/zephyr #25234: kernel.timer.tickless test fails on atsamd21_xpro

stephanosio, 2020-05-12:
**Describe the bug**
The `tests/kernel/timer/timer_api/kernel.timer.tickless` test fails on the `atsamd21_xpro` board.

**To Reproduce**
Build and run `tests/kernel/timer/timer_api/kernel.timer.tickless` on `atsamd21_xpro`.

**Expected behavior**
Test passes

**Screenshots or console output**
```
Running test suite timer_api
===================================================================
starting test - test_time_conversions
PASS - test_time_conversions
===================================================================
starting test - test_timer_duration_period
PASS - test_timer_duration_period
===================================================================
starting test - test_timer_period_0
PASS - test_timer_period_0
===================================================================
starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
starting test - test_timer_periodicity
PASS - test_timer_periodicity
===================================================================
starting test - test_timer_status_get
PASS - test_timer_status_get
===================================================================
starting test - test_timer_status_get_anytime
PASS - test_timer_status_get_anytime
===================================================================
starting test - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
starting test - test_timer_k_define

    Assertion failed at ../src/main.c:64: duration_expire: interval >= 100 is false

FAIL - test_timer_k_define
===================================================================
starting test - test_timer_user_data
PASS - test_timer_user_data
===================================================================
starting test - test_timer_remaining
PASS - test_timer_remaining
===================================================================
starting test - test_timeout_abs
PASS - test_timeout_abs
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: ed2d263e0c233df015448de8d45a0fae359afc8b

pabigot, 2020-05-18:
FWIW I'm unable to reproduce this on my board; running with `ZTEST_RETEST_IF_PASSED=y` passed 100 times in a row without failure.

carlescufi, 2020-05-21:
Thanks @pabigot.
@stephanosio can you take another look to see if you can reproduce? @pabigot clearly cannot.

stephanosio, 2020-05-22:
> FWIW I'm unable to reproduce this on my board; running with `ZTEST_RETEST_IF_PASSED=y` passed 100 times in a row without failure.

@pabigot @carlescufi I re-tested this on the latest master today and the failure is gone (it is still reproducible with https://github.com/zephyrproject-rtos/zephyr/commit/ed2d263e0c233df015448de8d45a0fae359afc8b, so something must have recently fixed it),

Closing since this is no longer an issue.

closed 2020-05-22 by stephanosio