zephyrproject-rtos/zephyr #25240: Building usb audio sample hangs the pre-processor

arvid-r, 2020-05-12:
**Describe the bug**
When trying to build the headset usb audio sample, the build process hangs with 100% cpu usage and never seems to finish.

**To Reproduce**
Steps to reproduce the behavior:
1. checkout zephyr and cd into directory
2. west -v build -p auto -b nrf52840dk_nrf52840 samples/subsys/usb/audio/headset 
3. won't finish ...

**Expected behavior**
Build should finish shortly, and the sample should run on the selected board

**Impact**
Showstopper for USB audio subsystem

**Environment (please complete the following information):**
 - OS: MacOS
 - gcc-arm-none-eabi-9-2019-q4-major
 - Commit SHA or Version used: ed2d263e0c233df015448de8d45a0fae359afc8b

**Additional context**
The pre-processor gets stuck with audio.c open, and the problem goes away if you comment out the following lines:

```
UTIL_LISTIFY(HEADPHONES_DEVICE_COUNT, HEADPHONES_DEVICE, HP)
UTIL_LISTIFY(MICROPHONE_DEVICE_COUNT, MICROPHONE_DEVICE, MIC)
UTIL_LISTIFY(HEADSET_DEVICE_COUNT, HEADSET_DEVICE, HS)
```

So this probably has something to do with macro expansion and the devicetree

emob-nordic, 2020-05-12:
I am observing same with my setup. I am using vm linux. Something was merged in between what broke the build. I am investigating the issue.

emob-nordic, 2020-05-12:
Got the fix, will prepare a PR asap.

PR #25250 (2020-05-12): usb: dt: Update USB audio class to recent changes in DT
This commit updates USB audio class implementation with
DT changes introduced in 7e0eed9235. In particular rename
DT_NUM_INST to DT_NUM_INST_STATUS_OKAY.

Fixes: #25240
closed 2020-05-12 by carlescufi