zephyrproject-rtos/zephyr #25260: drivers: uart_ns16550: device config_info content mutated

pabigot, 2020-05-12:
The device config_info structure is intended to be read-only, and so
the generic pointer stored in struct device is a pointer-to-const.
The uart_ns16550 driver removes this qualifier when casting the pointer to
the device-specific type.  This removal is a violation of MISRA 11.8
as it evokes undefined behavior.

When the qualifier is preserved the driver does not build, because it
modifies the object content at runtime.  Consequently the fix for this
driver has been removed from #25248.

https://github.com/pabigot/zephyr/commits/nordic/issue/24745e provides
the changes necessary to preserve const correctness.  When applied it
produces the following diagnostic:

```
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/drivers/serial/uart_ns16550.c: In function 'uart_ns16550_configure':
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/drivers/serial/uart_ns16550.c:333:25: error: assignment of member 'port' in read-only object
  333 |   dev_cfg->devconf.port = pcie_get_mbar(dev_cfg->pcie_bdf, 0);
      |                         ^
```

The driver must be updated to resolve this and incorporate the change
from the referenced branch.

jenmwms, 2020-05-13:
@pabigot can you please help me understand this - I made the change suggested, but did not see the diagnostic you mentioned. How can I reproduce that?

pabigot, 2020-05-13:
@jenmwms provided details in #25274.

jenmwms, 2020-05-13:
@pabigot Thank you for the details and how to reproduce this. I'm looking at this

pabigot, 2020-05-19:
AFAICT this should have been fixed by af43e14bd0dbcdb0595b1e6620f85fd3513a9eaa.

Could you please confirm @jenmwms @dcpleung ?

pabigot, 2020-05-19:
af43e14bd0dbcdb0595b1e6620f85fd3513a9eaa fixed the assignment to a nominally const structure; #25462 preserves the qualifier in the helper macro and casts.  I don't know if there's anything else to do.

dcpleung, 2020-05-19:
https://github.com/zephyrproject-rtos/zephyr/commit/af43e14bd0dbcdb0595b1e6620f85fd3513a9eaa is simply for the PCIE code to not change the config struct. I haven't vetted anywhere else in the code, so can't say for sure.

jenmwms, 2020-05-20:
Thanks for asking! Great questions. I recognize #25462 looks to provide the minimal changes remaining to resolve the issue after af43e14 merged, however please reconsider the existing PR #25274 (which already contains these changes) along with improved consistency related to this topic.

closed 2020-05-20 by carlescufi

pabigot, 2020-05-20:
@jenmwms I thought this might get more trade-off discussion relative to #25274 which you've been working.  I think you should be able to rebase #25274 on top of master to get your additional changes in.

jenmwms, 2020-05-20:
@pabigot Ok, I will do that.

PR #25462 (2020-05-20): device: avoid casting away const from config_info pointer
Re-run the const_config_info Coccinelle patch to fix code.

Fixes #25260