zephyrproject-rtos/zephyr #25263: Can anyone tell me how can i use external qspi flash "mx25r64"(custom board with nrf52840 soc) for mcuboot slot1 and i'm using zephyr 2.3.0

jrananth1032, 2020-05-12:
I have tested external flash on my custom board.Now i want to use that for mcuboot slot1 space.
I have followed this https://github.com/JuulLabs-OSS/mcuboot/wiki/Using-the-nrf52840_pca10056-SPI-flash-with-Zephyr
but didn't get correctly
my mcuboot log is showing:
primary source is none and panic!
can anyone help me with this.
Thanks.

arvid-r, 2020-05-12:
I'm not even sure 2.3.0 works with mcuboot at this point, does it?

jrananth1032, 2020-05-13:
sorry it is 2.2.0

pabigot, 2020-05-13:
The instructions are slightly out of date, but the current state is that mcuboot's `flash_area_read_is_empty()` routine attempts to read a byte value from the flash, and the QSPI flash API requires that all I/O operations be a multiple of 4 bytes.

I've opened #25289 to address this.  Please follow that issue for updates.

(Also the instructions on that page are somewhat out of date, so resolution of #25289 will likely be accompanied by an example that shows exactly how to do this.)

closed 2020-05-13 by pabigot

pabigot, 2020-05-13:
@jrananth1032 please see additional configuration information in #25294, hope this helps.  I don't know when it'll get integrated into an example.

jrananth1032, 2020-05-13:
thank you ,that information is very good.
i am looking forward for that change.