zephyrproject-rtos/zephyr #25272: tests/drivers/gpio/gpio_basic_api failed on mec15xxevb_assy6853 board.

chen-png, 2020-05-13:
**Describe the bug**
running tests/drivers/gpio/gpio_basic_api on mec board,  **i connected Pin11 and Pin 13 on JP27, and removed jumper on JP15, but it still failed with "Check GPIO040_076 output 8 connected to input 9".**

i read the related issue #22931 and git bisect, found the **first bad commit is  e11da0582d8.**

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck --hardware-map ~/mec1.file --device-testing 
    -T tests/drivers/gpio/gpio_basic_api/ -vv

**Screenshots or console output**

DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.2.0-322-ge11da0582d85  ***
DEBUG   - DEVICE: Running test suite gpio_basic_test
DEBUG   - DEVICE: 
DEBUG   - DEVICE: starting test - test_gpio_port
**DEBUG   - DEVICE: Validate device GPIO040_076
DEBUG   - DEVICE: Check GPIO040_076 output 8 connected to input 9
DEBUG   - DEVICE: FATAL output pin not wired to input pin? (out high => in low)**

DEBUG   - Timed out while monitoring serial output on mec15xxevb_assy6853
INFO    - 1/1 mec15xxevb_assy6853       tests/drivers/gpio/gpio_basic_api/drivers.gpio.2pin FAILED Timeout (device 113.219s)

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.2
 - Commit ID: 4980a13fc4

chen-png, 2020-05-13:
@franciscomunoz @dcpleung  Could you take a look of this?

dcpleung, 2020-05-13:
Because GPIO_050 has set to be used for TACH, so this is supposed to fail. A new GPIO line is needed.

carlescufi, 2020-05-13:
@chen-png could you please label test failures with the "Tests" label?

carlescufi, 2020-05-13:
> Because GPIO_050 has set to be used for TACH, so this is supposed to fail. A new GPIO line is needed.
@dcpleung 
Is this something that will be fixed, or should this test be blacklisted for this platform?

dcpleung, 2020-05-13:
I am going to look for another GPIO line today.

chen-png, 2020-05-14:
> @chen-png could you please label test failures with the "Tests" label?

yes,i will.

PR #25296 (2020-05-15): tests: gpio_basic_api: change GPIOs for mec15xxevb_assy6853
GPIO 050/051 are being used for tachometer sensor as
CONFIG_TACH_XEC is enabled by default. So for the gpio_basic_api
test, another set of GPIOs are needed. GPIO 156/157 are chosen
as they are (more or less) dedicated for two LEDs on board and
not being used for other functions (pinmux-wise).

Fixes #25272
commit 01445ef8 (2020-05-15): tests: gpio_basic_api: change GPIOs for mec15xxevb_assy6853

GPIO 050/051 are being used for tachometer sensor as
CONFIG_TACH_XEC is enabled by default. So for the gpio_basic_api
test, another set of GPIOs are needed. GPIO 156/157 are chosen
as they are (more or less) dedicated for two LEDs on board and
not being used for other functions (pinmux-wise).

Fixes #25272
closed 2020-05-15 by carlescufi