zephyrproject-rtos/zephyr #25276: OpenThread not work after upgrade to latest version

ghlualven, 2020-05-13:
**Describe the bug**
Upgrade to the latest code Zephyr 2.3.0-rc1 (Zephyr OS build v2.3.0-rc1-26-g4980a13fc4f7), the firmware initial works. But after execute "ot scan" command, no any output and the board bricked. Re-power is not working, the board must be re-flashed.

The same program is working at: Zephyr OS build zephyr-v2.2.0-881-gecb85a7db1f1.

rlubos, 2020-05-13:
@ghlualven I think we need more information here. I've just tried the `ot scan` command  in the `echo_server` sample, running on nRF52840, and it works just fine. I've used the recent rc tag (Zephyr OS build v2.3.0-rc1).

What board do you use? Can you reporoduce the issue on any of the existing Zephyr samples? Did you run `west update` after switching to the rc1? There were changes on both sides, Zephyr and OpenThread.

ghlualven, 2020-05-14:
@rlubos I tried again today, and find the echo_server project works fine, but lwm2m_client doesn't.
I copied the overlay-ot.conf file from echo_server to lwm2m_client project, the build and flash both successful, but thread network can not work the same:
1. ot scan command will no response, and will not get back "Done" message to timeout;
2. No "Beacon" packet send out from radio.
3. Always show busy after this command executed.

This is the output from echo_server sample:
```
 *** Booting Zephyr OS build v2.3.0-rc1-26-g4980a13fc4f7  ***
[00:00:00.250,793] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.264,160] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.264,160] <inf> fs_nvs: alloc wra: 0, e10
[00:00:00.264,190] <inf> fs_nvs: data wra: 0, 3ac
[00:00:00.269,012] <inf> net_l2_openthread: State changed! Flags: 0x011fd309 Current role: 0
[00:00:00.270,385] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/ga83d18cf; NONE; May 14 2020 08:08:14
[00:00:00.270,416] <inf> net_l2_openthread: Network name: kirale-06B4
[00:00:00.270,416] <inf> net_l2_openthread: State changed! Flags: 0x00000004 Current role: 1
[00:00:00.270,507] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.270,599] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.270,721] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.270,843] <inf> net_l2_openthread: State changed! Flags: 0x00000010 Current role: 1
[00:00:00.270,874] <inf> net_l2_openthread: State changed! Flags: 0x00000020 Current role: 1
[00:00:00.282,531] <inf> net_config: Initializing network
[00:00:00.282,653] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.298,309] <inf> net_l2_openthread: State changed! Flags: 0x000010e4 Current role: 3
[00:00:00.319,793] <inf> net_l2_openthread: State changed! Flags: 0x00000200 Current role: 3
[00:00:00.319,824] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 3
[00:00:03.283,081] <inf> net_echo_server_sample: Run echo server
[00:00:03.283,111] <inf> net_echo_server_sample: Network connected
[00:00:03.283,142] <inf> net_echo_server_sample: Starting...
[00:00:03.283,264] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
uart:~$ ot scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread       | dead00beef00cafe | abcd | 2e6c35c5155ae11a | 11 | -55 | 160 |
| 1 | kirale-06B4      | 0948130241ac0b32 | 06b4 | eb003594d71e6f1a | 11 | -58 | 140 |
Done
uart:~$ log halt
uart:~$ ot scan
| J | Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+---+------------------+------------------+------+------------------+----+-----+-----+
| 0 | OpenThread       | dead00beef00cafe | abcd | 2e6c35c5155ae11a | 11 | -54 | 160 |
| 1 | kirale-06B4      | 0948130241ac0b32 | 06b4 | eb003594d71e6f1a | 11 | -58 | 140 |
Done
uart:~$ 
```

This is the output from lwm2m_client sample:
```
*** Booting Zephyr OS build v2.3.0-rc1-26-g4980a13fc4f7  ***
[00:00:00.247,528] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.260,681] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.260,681] <inf> fs_nvs: alloc wra: 0, f30
[00:00:00.260,681] <inf> fs_nvs: data wra: 0, 174
[00:00:00.262,054] <inf> net_l2_openthread: State changed! Flags: 0x011fd319 Current role: 0
[00:00:00.262,786] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/ga83d18cf; NONE; May 14 2020 08:49:57
[00:00:00.262,786] <inf> net_l2_openthread: Network name: ot_zephyr
[00:00:00.262,817] <inf> net_l2_openthread: State changed! Flags: 0x00000004 Current role: 1
[00:00:00.262,908] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.263,000] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.263,122] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.263,244] <inf> net_l2_openthread: State changed! Flags: 0x00000010 Current role: 1
[00:00:00.263,275] <inf> net_l2_openthread: State changed! Flags: 0x00000020 Current role: 1
[00:00:00.272,216] <dbg> net_lwm2m_engine.lwm2m_engine_init: LWM2M engine socket receive thread started
[00:00:00.272,277] <dbg> net_lwm2m_obj_security.security_create: Create LWM2M security instance: 0
[00:00:00.272,308] <dbg> net_lwm2m_obj_server.server_create: Create LWM2M server instance: 0
[00:00:00.272,399] <dbg> net_lwm2m_obj_device.device_create: Create LWM2M device instance: 0
[00:00:00.272,430] <dbg> net_lwm2m_obj_conn_mon.connmon_create: Create LWM2M connectivity monitoring instance: 0
[00:00:00.272,460] <dbg> net_lwm2m_obj_firmware.firmware_create: Create LWM2M firmware instance: 0
[00:00:00.272,460] <dbg> net_lwm2m_obj_location.location_create: Create Location instance: 0
[... 89 lines cut ...]
HEAD is now at 724f7e2 lib: update open-amp lib to release v2020.04.0
=== updating loramac-node (modules/lib/loramac-node):
HEAD is now at 29e516ec radio: sx1276: allow overriding Buffer reand/write functions
=== updating openthread (modules/lib/openthread):
HEAD is now at a83d18cf cmake: Make sure libc is linked after OT libs
=== updating segger (modules/debug/segger):
HEAD is now at 6fcf616 ext: segger: update to SystemView v2.52h
=== updating tinycbor (modules/lib/tinycbor):
[... cut ...]
=== updating nrf_hw_models (modules/bsim_hw_models/nrf_hw_models):
HEAD is now at fec6970 Convert into a west module
=== updating hal_xtensa (modules/hal/xtensa):
HEAD is now at e7a57d0 hal: Add README for information on updating HAL versions.
=== updating edtt (tools/edtt):
HEAD is now at c39888f Update unsupported commands test to handle command status response
=== updating trusted-firmware-m (modules/tee/tfm):
HEAD is now at 7de2daa zephyr: cmake: TF-M libraries to be used by the ns-app
zephyrproject % 
```

ghlualven, 2020-05-14:
build command:
`west build -b nrf52840dk_nrf52840 -d build_nrf52840dk_nrf52840_ot -- -DCONF_FILE="prj.conf overlay-ot.conf"`

overlay-ot.conf
```
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y

CONFIG_BT=n

# Disable TCP and IPv4 (TCP disabled to avoid heavy traffic)
CONFIG_NET_TCP=n
CONFIG_NET_IPV4=n

CONFIG_NET_IPV6_NBR_CACHE=n
CONFIG_NET_IPV6_MLD=n
CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=n
CONFIG_NET_CONFIG_MY_IPV4_ADDR=""
CONFIG_NET_CONFIG_PEER_IPV4_ADDR=""

CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_NET_TX_STACK_SIZE=2048
CONFIG_NET_RX_STACK_SIZE=5120
CONFIG_SHELL_STACK_SIZE=3072

CONFIG_NET_L2_OPENTHREAD=y
CONFIG_OPENTHREAD_SLAAC=y

CONFIG_OPENTHREAD_DEBUG=y
CONFIG_OPENTHREAD_L2_DEBUG=y
CONFIG_OPENTHREAD_L2_LOG_LEVEL_INF=y

CONFIG_OPENTHREAD_CHANNEL=26

CONFIG_NET_CONFIG_MY_IPV6_ADDR="fdde:ad00:beef::2"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="fdde:ad00:beef::1"

# Thread by default registers quite a lot addresses.
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=8

# Other OpenThread dependencies
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_MAP=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y
CONFIG_NVS=y
CONFIG_SETTINGS=y

CONFIG_REBOOT=y

# mbedTLS tweaks
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=768

# A sample configuration to enable Thread Commissioner, uncomment if needed
#CONFIG_OPENTHREAD_JOINER=y
#CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
#CONFIG_OPENTHREAD_COMMISSIONER=y
#CONFIG_MBEDTLS_HEAP_SIZE=8192
```

prj.conf
```
CONFIG_NETWORKING=y
CONFIG_LOG=y
CONFIG_LWM2M_LOG_LEVEL_DBG=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_NET_IPV6=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=2
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=2
CONFIG_PRINTK=y
CONFIG_NET_PKT_RX_COUNT=10
CONFIG_NET_PKT_TX_COUNT=10
CONFIG_NET_BUF_RX_COUNT=10
CONFIG_NET_BUF_TX_COUNT=10
CONFIG_NET_MAX_CONTEXTS=5

CONFIG_NET_LOG=y
CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_NEED_IPV6=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_SETTINGS=y

CONFIG_LWM2M=y
CONFIG_LWM2M_COAP_BLOCK_SIZE=512
CONFIG_LWM2M_IPSO_SUPPORT=y
CONFIG_LWM2M_IPSO_TEMP_SENSOR=y
CONFIG_LWM2M_IPSO_LIGHT_CONTROL=y
CONFIG_LWM2M_IPSO_TIMER=y
# compile test
CONFIG_LWM2M_CONN_MON_OBJ_SUPPORT=y
CONFIG_LWM2M_LOCATION_OBJ_SUPPORT=y
CONFIG_LWM2M_IPSO_ACCELEROMETER=y
CONFIG_LWM2M_IPSO_BUZZER=y
CONFIG_LWM2M_IPSO_ONOFF_SWITCH=y
CONFIG_LWM2M_IPSO_PUSH_BUTTON=y

CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
```

rlubos, 2020-05-14:
I'd bet it's a stack issue, I've been testing this sample a few days ago, and it turned out to have insufficient stack size for OpenThread in the default configuration:
https://github.com/zephyrproject-rtos/zephyr/issues/24840#issuecomment-624717991

I'll verify it locally and report back.

rlubos, 2020-05-14:
Yes, I confirm and recommend to increase `CONFIG_MAIN_STACK_SIZE` to `2048`, OpenThread works way better.

ghlualven, 2020-05-14:
@rlubos Thank you very much, I also verified this,  the issue was fixed.

closed 2020-05-14 by ghlualven