zephyrproject-rtos/zephyr #25283: sam0: watchdog: Times out twice as fast as expected

KubaFYI, 2020-05-13:
The watchdog times out at twice the speed expected on a samd21-based Adafruit Feather M0. 

In order to reproduce the problem I used the watchdog sample code modified to let the watchdog time out 30 times (so that I could time the behaviour accurately against a wall clock). Setting a timeout window of 1000ms has the whole thing complete in about 16sec, 2000ms times out in about 32sec etc.

carlescufi, 2020-05-19:
@pabigot could you please try to reproduce?

pabigot, 2020-05-20:
I can't duplicate this on `atsamd21_xpro`.  From what I can tell the timer operates correctly within the limits of the SAM D21 WDT.

Allowed intervals are 8 ms * 2^k for k in [0, 12), i.e. 8 ms, 16 ms, 32 ms, ..., 8 s, 16 s.  Ignoring windowed mode, the configured period is rounded up to the smallest duration not less than the requested duration.  The warning interrupt (when enabled) fires at half this duration.

So at 2000 ms max the warning interrupt will fire at 1 s, and in the sample will clear the watchdog, so the board will reset 2 s later at 3 s after boot.  I've confirmed this relationship for several delays.

I don't see anything obvious about the `adafruit_feather_m0_basic_proto` that suggests a misconfiguration of clocks.  I don't have that board to test directly.

**Now:** reading the description of your modification more carefully: If what you've done is set it up so the warning callback is invoked 30 times, then what you see is correct.  At a 1000 ms timeout you're feeding the watchdog 30 times at a 500 ms interval (15 s), then getting a timeout 1000 ms after the last feed (16 s).

Does that explain it?

pabigot, 2020-06-29:
@KubaFYI Could you verify that the analysis above and the observed behavior is correct?

nzmichaelh, 2020-07-01:
The analysis LGTM

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi