zephyrproject-rtos/zephyr #25293: Add USB OTG Support to STM32411E-DISCO

bpbradley, 2020-05-13:
The STM32F411E supports USB OTG, and has a USB OTG port, but as far as I can tell it is not supported in the board configuration in Zephyr. I have begun work towards a solution to this, but haven't been able to completely succeed, yet.

Currently, I am trying to get the board to work with the existing USB samples, namely the samples regarding CDC mode of operation. Trying to build these on the latest Zephyr (2.3-rc0) fails in compilation, as expected.

So far, I have made changes to the boards dts, by simply adding

```dts
&usbotg_fs {
	status = "okay";
};
```

and I have updated the pin config in pinmux.c according to other stm32 USB implementations and the reference datasheet for the STM32F411E MCU.

```c
static const struct pin_config pinconf[] = {
    ...
#ifdef CONFIG_USB_DC_STM32
	{STM32_PIN_PA11, STM32F4_PINMUX_FUNC_PA11_OTG_FS_DM},
	{STM32_PIN_PA12, STM32F4_PINMUX_FUNC_PA12_OTG_FS_DP},
#endif	/* CONFIG_USB_DC_STM32 */
};
```

With just those changes, all USB samples compile perfectly. However, on three boards I have tested, none of them work correctly so I assume there is more work to be done. 

Right now, after successfully flashing the usb_cdc sample, for example, when I connect the STM32F411E-DISCO board to my PC (Windows 10, but I can test Linux if it helps) the OS appears to be aware of the device, but it is failing with "Device Descriptor Request Failed." Since this is happening on 3 different boards, I am reasonably sure it is not hardware related. Note that I have tested connecting the USB OTG port both before and after power on of the device, in case it is related to the processor not being ready to send data when the OS wants it.

I am happy to continue working on this, but may need some advice on where to look next. I can also work towards creating a PR with the above additions, and anything else that may be discovered to resolve the device enumeration issue.

Also note that I tested the sample as-is on a nrf52840 dvk and it worked perfectly, so I do not believe it is anything in the sample itself.

carlescufi, 2020-05-13:
By "OTG" you mean using the hardware as a USB Device, or actually supporting USB Host? Because that is covered in a separate issue:
https://github.com/zephyrproject-rtos/zephyr/issues/12386

bpbradley, 2020-05-13:
> By "OTG" you mean using the hardware as a USB Device, or actually supporting USB Host? Because that is covered in a separate issue:
> #12386

Sorry if I wasn't clear. I meant to simply state that the STM32F411E supports USB OTG, which includes both host and device capabilities. But in this case, I am specifically trying to get it to function as a device, specifically CDC. So I did not intend to broaden the scope of this to host capabilities.

To more succinctly state the issue, I believe the STM32F411E should be compatible with samples/subsys/usb/cdc_acm (among the other usb samples), and out of the box it isn't. Making the changes I showed previously results in being able to compile the project (and other usb samples) successfully, but it still results in a "Device Descriptor Request Failure" when trying to communicate with it, which is not true of other boards (like nrf52840 dvk, for instance, where it works perfectly).

If I am mistaken in my thought that this MCU should be compatible with CDC samples, then I apologize. I tried discussing this on the mailing lists first but raising an issue was recommended.

erwango, 2020-05-15:
@bpbradley I've tested cdc_acm sample and it works fine on nucleo_f429zi.

In this case two usual suspects are board pin configuration that would require jumper or solder bridge fixing (but according to doc ref manual there is nothing specific for this board) and clock configuration.
On this last front, looking to the board _defconfig:
```
# 100MHz system clock (highest value to get a precise USB clock should be 96MHz)
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=100000000
```
Can you tweak the PLL clock params to achieve a 96MHz sys clock, and have a new check ?

bpbradley, 2020-05-15:
> @bpbradley I've tested cdc_acm sample and it works fine on nucleo_f429zi.
>
> In this case two usual suspects are board pin configuration that would require jumper or solder bridge fixing (but [...]

Great suggestion. I looked into the HSE clock previously but saw it was correct and moved on. I was very quickly able to confirm that you are 100% correct that this is PLL related. A quick test following your suggestion on a 96 MHz PLL output, but also trying for a 48 MHz USB clock output, and the example worked perfectly.

However, I am not sure it is the full story. Digging around it is possible that 100MHz is okay, but I am coming to understand the comparison frequency should be set to 1MHz generally (not sure why), which would be M=8 on the F4x DISCO boards, and then N and P should be set accordingly to achieve 100MHz (and Q to achieve 50MHz on USB clock, I suppose?) Doing this, it also works correctly. I am not sure which is the ideal solution, but both work and the latter results in a higher clock frequency.

Interestingly, it would appear the other DISCO-f4xx boards follow the convention of choosing M=8 as well, so I am leaning towards thinking this is correct.

Should I work towards putting these changes into a PR?

Thanks again for the help,
Brian

bpbradley, 2020-05-18:
As a follow up to this, it seems I am not quite there, but close.

I modified my previous preference and now believe we should in fact be using a 96MHz clock, despite some conflicting information around the web. The reference manual seems to agree the USB clock should be set to precisely 48MHz for best results, and the datasheet suggests setting the pll in comparison frequency to 2MHz (M=4) for less jitter, even though 1 MHz (M=8) is typical. So with that, I believe I have the optimal settings. 

However, I notice another interesting issue. Before submitting a PR with these changes, I wanted to test all of the usb samples, or as many as I could on the DISCO board (not just the ones I need for my application). 

Interestingly, most that I could test, (cdc, HID, console) work perfectly except cdc_acm_composite and hid_cdc. 

These samples work perfectly on the nrf52840 dvk as a control, but with the DISCO board, they once again fail to enumerate (Device Descriptor Request Failed). In this case, I checked usb dp/dm with an oscilloscope, and the frequency appears to be good (waveforms appear identical to nrf52840, as far as I can tell).

I would be interested to know if you are able to get these to work with nucleo_f429zi. Perhaps it is an unrelated bug?

hid_cdc also had a separate issue where I needed to add ```CONFIG_TEST_RANDOM_GENERATOR=y``` in order to resolve ```undefined references to sys_rand32_get``` but that is just a footnote.

erwango, 2020-05-19:
@bpbradley, I've tested `hid-cdc` and `cdc_acm_composite`, they're failing also on my side (nucleo_f429zi), can you raise an issue so that we can investigate ?

bpbradley, 2020-05-19:
> @bpbradley, I've tested `hid-cdc` and `cdc_acm_composite`, they're failing also on my side (nucleo_f429zi), can you raise an issue so that we can investigate ?

Yes I will do that. Thanks for looking into it!

bpbradley, 2020-06-23:
These changes have been merged to master. I can close this issue now.

closed 2020-06-23 by bpbradley