zephyrproject-rtos/zephyr #25294: Nordic mcuboot + smp_svr + QSPI smp_shell incompatibility

pabigot, 2020-05-13:
The following configuration can replace [this outdated information](https://github.com/JuulLabs-OSS/mcuboot/wiki/Using-the-nrf52840_pca10056-SPI-flash-with-Zephyr) to enable Nordic QSPI storage of non-primary image data for MCUBOOT, as long as #25292 is also present:

For devicetree:
```
diff --git a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
index 1068670006..02c98faaa3 100644
--- a/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
+++ b/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
@@ -249,10 +249,12 @@ arduino_spi: &spi3 {
                        label = "image-0";
                        reg = <0x0000C000 0x000067000>;
                };
+/*
                slot1_partition: partition@73000 {
                        label = "image-1";
                        reg = <0x00073000 0x000067000>;
                };
+*/
                scratch_partition: partition@da000 {
                        label = "image-scratch";
                        reg = <0x000da000 0x0001e000>;
@@ -274,6 +276,19 @@ arduino_spi: &spi3 {
        };
 };
 
+&mx25r64 {
+       partitions {
+               compatible = "fixed-partitions";
+               #address-cells = <1>;
+               #size-cells = <1>;
+
+               slot1_partition: partition@0 {
+                       label = "image-1";
+                       reg = <0x00000000 0x00080000>;
+               };
+       };
+};
+
 &usbd {
        compatible = "nordic,nrf-usbd";
        status = "okay";
```

For MCUBOOT:
```
diff --git a/boot/zephyr/prj.conf b/boot/zephyr/prj.conf
index b316f0e..691a07d 100644
--- a/boot/zephyr/prj.conf
+++ b/boot/zephyr/prj.conf
@@ -42,11 +42,10 @@ CONFIG_FLASH=y
 
 CONFIG_MULTITHREADING=y
 
-CONFIG_SPI=y
 CONFIG_FLASH=y
-CONFIG_SPI_NOR=y
 CONFIG_BOOT_MAX_IMG_SECTORS=256
-CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
+CONFIG_NORDIC_QSPI_NOR=y
+CONFIG_NORDIC_QSPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
 
 ### Various Zephyr boards enable features that we don't want.
 # CONFIG_BT is not set
```

For `samples/subsys/mgmt/mcumgr/smp_svr/`:
```
diff --git a/samples/subsys/mgmt/mcumgr/smp_svr/boards/nrf52840dk_nrf52840.conf b/samples/subsys/mgmt/mcumgr/smp_svr/boards/nrf52840dk_nrf52840.conf
index 5b3c3cdd91..3838b4f28f 100644
--- a/samples/subsys/mgmt/mcumgr/smp_svr/boards/nrf52840dk_nrf52840.conf
+++ b/samples/subsys/mgmt/mcumgr/smp_svr/boards/nrf52840dk_nrf52840.conf
@@ -5,7 +5,7 @@ CONFIG_BT_RX_BUF_LEN=260
 # Enable the Bluetooth (unauthenticated) and shell mcumgr transports.
 CONFIG_MCUMGR_SMP_BT=y
 CONFIG_MCUMGR_SMP_BT_AUTHEN=n
-CONFIG_MCUMGR_SMP_SHELL=y
+#CONFIG_MCUMGR_SMP_SHELL=y
 
 # Enable the LittleFS file system.
 CONFIG_FILE_SYSTEM=y
@@ -13,3 +13,7 @@ CONFIG_FILE_SYSTEM_LITTLEFS=y
 
 # Enable file system commands
 CONFIG_MCUMGR_CMD_FS_MGMT=y
+
+CONFIG_NORDIC_QSPI_NOR=y
+CONFIG_NORDIC_QSPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096
```

(A) If `CONFIG_MCUMGR_SMP_SHELL=y` is selected the smp_svr application deadlocks.  This should be investigated
(B) Once the issue with the shell is resolved it would be good to integrate this capability into an example to address the question raised in #25263.

rerickson1, 2020-06-03:
I tried this and it's not quite working.  I am able to upload the image and see it in slot1 with the hash.  When I test the image, it looks like it never sets the flags to pending.  Upon reset it just doesn't try switching the image, just runs slot0 image again.  Any thoughts?

pabigot, 2020-06-03:
I believe you have to use `mcumgr image confirm`

rerickson1, 2020-06-04:
@pabigot I tested with internal flash first and everything works fine.
I made the changes above and flashed the new mcuboot and app.  Then I made a small change to smp_svr and compiled/signed the new binary.
I followed these steps:
1. Upload image
2. List images
3. Test image
4. reset
5. confirm

with external flash, the test image step looks different the flags of the slot1 are still blank and not marked as pending.  With internal flash, after the test image, the flags are marked as pending.  See external qspi mcumgr steps and output here: https://pastebin.com/QUA60vA4

With internal flash steps here:https://pastebin.com/YwGgZvhy

pabigot, 2020-06-04:
OK, I can confirm that.  The problem is probably lack of sub-word write support (the other side of #25292).  A solution is in nrfconnect/sdk-zephy#305 but has not been upstreamed.

pabigot, 2020-06-04:
@rerickson1 OK I think this works.  You'll need #25971.  See https://github.com/zephyrproject-rtos/zephyr/pull/25971#issuecomment-639017038 for pointers to mcuboot and zephyr branches that I've confirmed work.

I've opened  #25981 for the issues we've been discussing; please continue any feedback there so this issue remains specific to the apparent inability to use `smp_shell` in this environment (which may or may not be fixed as a result of #25971).

closed 2020-08-18 by github-actions[bot]