zephyrproject-rtos/zephyr #25295: Odd sanitycheck bug

galak, 2020-05-13:
Running the following:

`./scripts/sanitycheck -p native_posix -T samples/subsys/usb/hid`

Gets:

```
ERROR   - native_posix              samples/subsys/usb/hid/sample.usb.hid              FAILED: Failed
ERROR   - see: /home/galak/git/zephyr/sanity-out/native_posix/samples/subsys/usb/hid/sample.usb.hid/handler.log
INFO    - Total complete:    2/   2  100%  skipped:    0, failed:    1
INFO    - 1 of 2 tests passed (50.00%), 1 failed, 0 skipped with 0 warnings in 7.12 seconds
INFO    - In total 2 test cases were executed on 1 out of total 249 platforms (0.40%)
```

Now if you try:

`./scripts/sanitycheck -p native_posix -s samples/subsys/usb/hid/sample.usb.hid_composite` you get:

```
INFO    - JOBS: 32
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 6.46 seconds
INFO    - In total 1 test cases were executed on 1 out of total 249 platforms (0.40%)
INFO    - 1 tests executed on platforms, 0 tests were only built.
```

erwango, 2020-05-14:
Doesn't it depends on sample yal file ?
```
tests:
  sample.usb.hid:
    depends_on: usb_device
    tags: usb
    harness: console
    harness_config:
      type: multi_line
      regex:
        - "composite_pre_init: HID Device: dev"
        - "main: Starting application"
  sample.usb.hid_composite:
    depends_on: usb_device
    extra_configs:
      - CONFIG_USB_COMPOSITE_DEVICE=y
    tags: usb
    harness: console
    harness_config:
      type: multi_line
      regex:
        - "composite_pre_init: HID Device: dev"
        - "main: Starting application"
```

Running `./scripts/sanitycheck -p native_posix -T samples/subsys/usb/hid`, you run both config of the sample (one of hichbeing failed)
Running `./scripts/sanitycheck -p native_posix -s samples/subsys/usb/hid/sample.usb.hid_composite` you run the 'passed' one.

nashif, 2020-05-26:
Reason is that we have two tests running at the same time trying to access the same resource:

```
i9:zephyr(sanitycheck/native_posix_race): sh  /tmp/par.sh
*** Booting Zephyr OS build v2.3.0-rc1-299-g6e216d9b7477  ***
[00:00:00.000,000] <dbg> main.composite_pre_init: HID Device: dev 0x08055474
[00:00:00.000,000] <dbg> main.zephyr_app_main: Starting application
*** Booting Zephyr OS build v2.3.0-rc1-299-g6e216d9b7477  ***
[00:00:00.000,000] <dbg> main.composite_pre_init: HID Device: dev 0x08055474
[00:00:00.000,000] <dbg> main.zephyr_app_main: Starting application
[00:00:00.000,000] <err> native_posix_adapt: bind() failed: Address already in use
```

nashif, 2020-09-11:
open enhancement #28316, closing this as it is not a sanitycheck bug, rather a missing feature.

closed 2020-09-11 by nashif