zephyrproject-rtos/zephyr #25299: Inconsistency between dts status and Kconfig

QSQCaito, 2020-05-13:
**Describe the bug**
In the dts the systick is disabled, however, it's enabled via Kconfig options.
I'm not changing the systick status in my board dts. The dts systick status is taken from armv7-m.dtsi, and it appears as is in my build/zephyr/zephyr.dts file:
```
systick: timer@e000e010 {
			compatible = "arm,armv7m-systick";
			reg = < 0xe000e010 0x10 >;
			status = "disabled";
```
However,  `CORTEX_M_SYSTICK` is enabled by default because `CPU_CORTEX_M_HAS_SYSTICK` is also enabled. So this would mean that the SysTick is indeed enabled.

**To Reproduce**
I'm building a custom board, but any STM32L4 board should produce same results. 

**Expected behavior**
I would imagine that for SysTick to be effectively enabled, it would have to be first enabled in the dts.

**Impact**
None so far.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (GCC, J-link)
 - Commit SHA: 6e8775ff84

mbolivar-nordic, 2020-05-13:
@ioannisg I'm guessing this is just because CONFIG_CORTEX_M_SYSTICK does not depend on the node being enabled, right?

Seems like we could just set it to "okay" and add the appropriate Kconfig dependency. The driver Kconfig is default n so it wouldn't affect nRF as far as I can see.

Am I missing something?

QSQCaito, 2020-05-14:
@mbolivar-nordic , does it make sense to enable it directly at the armv7-m.dtsi level? For STM32 the LPTIM can be used as a Kernel tick, then it would have to be disabled, no?

ioannisg, 2020-05-26:
@mbolivar CONFIG_CORTEX_M_SYSTICK indicates that the system timer is based on the SysTick.

> Seems like we could just set it to "okay" in armv7-m.dtsi and add the appropriate Kconfig dependency. The driver Kconfig is default n so it wouldn't affect nRF as far as I can see.

What does it "mean" the SysTick having status "okay"? Does this mean the timer driver is actually used? If so, setting it as "okay" gives the wrong signal.

mbolivar-nordic, 2020-05-28:
> @mbolivar CONFIG_CORTEX_M_SYSTICK indicates that the system timer is based on the SysTick.

Yep, for sure. 

> What does it "mean" the SysTick having status "okay"? Does this mean the timer driver is actually used? If so, setting it as "okay" gives the wrong signal.

It just means the hardware is available for use. In general we're moving towards drivers being automatically enabled if the subsystem is turned on and the right devicetree nodes have status "okay".

ioannisg, 2020-07-21:
>In general we're moving towards drivers being automatically enabled if the subsystem is turned on and the right devicetree nodes have status "okay".

That's fair.

> Seems like we could just set it to "okay" in armv7-m.dtsi and add the appropriate Kconfig dependency.

@mbolivar-nordic what would be the appropriate Kconfig dependency? You mean something like:

```
config CORTEX_M_SYSTICK
    bool
    depends on "some DTS-macro indicating it is "status okay"
```
??

mbolivar-nordic, 2020-07-21:
>     depends on "some DTS-macro indicating it is "status okay"

Yep. I've got a PR adding some potential ones here if the existing kconfigfunctions aren't sufficient:
https://github.com/zephyrproject-rtos/zephyr/pull/24794

mbolivar-nordic, 2020-07-21:
Aha, I misread this. Actually, it's meant to be `default y if $(dt_<whatever>_enabled,<foo>)`, not `depends on`.

ioannisg, 2020-07-22:
> Aha, I misread this. Actually, it's meant to be `default y if $(dt_<whatever>_enabled,<foo>)`, not `depends on`.

Then this is not so straightforward. That's because the SysTick DTS node is part of the SoC definition only, while the config CORTEX_M_SYSTICK is enabled at board level.

For your proposal to work we need to be setting the SysTick node to status = okay at the board DTS, for all boards (and only those boards) that are supposed to work with the SysTick.

@QSQCaito has a relevant comment about this.

> In general we're moving towards drivers being automatically enabled if the subsystem is turned on and the right devicetree nodes have status "okay".

I'll wait until you make this work for Nordic and the RTC driver; then I 'll revisit this issue.

mbolivar-nordic, 2020-08-04:
> For your proposal to work we need to be setting the SysTick node to status = okay at the board DTS, for all boards (and only those boards) that are supposed to work with the SysTick.

Is that true?

Why can't we set a default to status 'okay' on the SoC level where the SoC expects most users to use systick, and then set it to status 'disabled' on a board specific basis if required?

PR #28199 (2020-09-10): arm: systick: default CORTEX_M_SYSTICK based on dts
Use setting from devicetree to drive the default setting for
CORTEX_M_SYSTICK.  We update the dts files to default systick to be
enabled since the major of cortex-m platforms utilize it by default
(except on Nordic SoCs in which we default to disabled).

Fixes #25299
commit 184a2576 (2020-09-10): arm: systick: default CORTEX_M_SYSTICK based on dts

Use setting from devicetree to drive the default setting for
CORTEX_M_SYSTICK.  We update the dts files to default systick to be
enabled since the major of cortex-m platforms utilize it by default
(except on Nordic SoCs, TI CC13x2/CC26x2 and MEC1501 in which we
default to disabled).

Fixes #25299
closed 2020-09-10 by ioannisg