zephyrproject-rtos/zephyr #25312: samples:mimxrt1010_evk:samples/net/openthread/ncp: build error

hakehuang, 2020-05-14:
**Describe the bug**
samples/net/openthread/ncp can not build in mimxrt1010 
[image]
baord need add limitations to filter upsupport platform out

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1010_evk ..
3. make
4. See error

**Expected behavior**
if it is not support need add filter

**Impact**
NA

**Screenshots or console output**
```
-- Application: /build/src/workspace/mimxrt1010_evk_2_master/zephyr/samples/net/openthread/ncp

-- Zephyr version: 2.3.0-rc1 (/build/src/workspace/mimxrt1010_evk_2_master/zephyr)

-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.10") found components:  Interpreter 

-- git describe failed: fatal: No names found, cannot describe anything.;

   BUILD_VERSION is left undefined

-- Board: mimxrt1010_evk

-- Found toolchain: zephyr (/root/zephyr-sdk)

-- Found west: /root/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")

-- Found dtc: /root/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")

-- Found BOARD.dts: /build/src/workspace/mimxrt1010_evk_2_master/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts

-- Generated zephyr.dts: /build/src/workspace/mimxrt1010_evk_2_master/zephyr/samples/net/openthread/ncp/build_d41d8cd/zephyr/zephyr.dts

-- Generated devicetree_unfixed.h: /build/src/workspace/mimxrt1010_evk_2_master/zephyr/samples/net/openthread/ncp/build_d41d8cd/zephyr/include/generated/devicetree_unfixed.h

warning: MBEDTLS_SSL_MAX_CONTENT_LEN (defined at modules/Kconfig.mbedtls:46) was assigned the value

'768' but got the value ''. Check these unsatisfied dependencies: MBEDTLS_BUILTIN (=n), MBEDTLS

(=n). See

http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN.html

and/or look up MBEDTLS_SSL_MAX_CONTENT_LEN in the menuconfig/guiconfig interface. The Application

Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of

the manual might be helpful too.

warning: MBEDTLS_HEAP_SIZE (defined at modules/Kconfig.mbedtls:112) was assigned the value '8192'

but got the value ''. Check these unsatisfied dependencies: MBEDTLS_ENABLE_HEAP (=n), MBEDTLS (=n).

See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MBEDTLS_HEAP_SIZE.html and/or look

up MBEDTLS_HEAP_SIZE in the menuconfig/guiconfig interface. The Application Development Primer,

Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be

helpful too.

[... 628 lines cut ...]
/root/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 13976 bytes

collect2: error: ld returned 1 exit status

%

make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1

zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:103: recipe for target 'zephyr/zephyr_prebuilt.elf' failed

make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2

CMakeFiles/Makefile2:660: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed

make: *** [all] Error 2

Makefile:83: recipe for target 'all' failed

script returned exit code 2
```

**Environment (please complete the following information):**
 - Linux
 -  Zephyr SDK
 - v2.3.0-rc1

hakehuang, 2020-05-14:
@MaureenHelm

dleach02, 2020-05-20:
@hakehuang, I'm going to close this as invalid. The sample.yaml lists netif as a dependency which is not part of the rt1010 evk board feature set so this should not have been part of the test run. The sanitycheck script skips this because of the netif dependency.

closed 2020-05-20 by dleach02

hakehuang, 2020-05-21:
@dlech  Thanks a lot, this is a false alart.

dlech, 2020-05-21:
I think you mean @dleach02 :wink:

hakehuang, 2020-08-18:
sorry @dlech , and thanks for your kindness.