zephyrproject-rtos/zephyr #25313: samples:

hakehuang, 2020-05-14:
**Describe the bug**
build error report usbd not found, but usbd is supported on this platform

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1010_evk ..
3. make
4. See error

**Expected behavior**
can build and run well

**Impact**
NA

**Screenshots or console output**
```

-- Application: /build/src/workspace/mimxrt1010_evk_usb_master/zephyr/samples/subsys/usb/audio/headset

-- Zephyr version: 2.3.0-rc1 (/build/src/workspace/mimxrt1010_evk_usb_master/zephyr)

-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.10") found components:  Interpreter 

-- git describe failed: fatal: No names found, cannot describe anything.;

   BUILD_VERSION is left undefined

-- Board: mimxrt1010_evk

-- Found toolchain: zephyr (/root/zephyr-sdk)

-- Found west: /root/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")

-- Found dtc: /root/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")

-- Found BOARD.dts: /build/src/workspace/mimxrt1010_evk_usb_master/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts

-- Found devicetree overlay: /build/src/workspace/mimxrt1010_evk_usb_master/zephyr/samples/subsys/usb/audio/headset/app.overlay

Error: mimxrt1010_evk.dts.pre.tmp:671.1-6 Label or path usbd not found

FATAL ERROR: Syntax error parsing input tree

CMake Error at /build/src/workspace/mimxrt1010_evk_usb_master/zephyr/cmake/dts.cmake:195 (message):

  command failed with return code: 1

Call Stack (most recent call first):

  /build/src/workspace/mimxrt1010_evk_usb_master/zephyr/cmake/app/boilerplate.cmake:503 (include)

  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!

make: *** No targets specified and no makefile found.  Stop.

script returned exit code 2
```
**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK
 - v2.3.0-RC1

hakehuang, 2020-05-14:
@MaureenHelm

jfischer-phytec-iot, 2020-05-14:
@galak Can we use common node label for the first usb device?

```diff
diff --git a/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts b/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts
index 1f560ec26f..6443883bd7 100644
--- a/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts
+++ b/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts
@@ -63,6 +63,6 @@ arduino_serial: &lpuart1 {};
        current-speed = <115200>;
 };
 
-&usb1 {
+usbd: &usb1 {
        status = "okay";
 };
```

erwango, 2020-05-19:
> @galak Can we use common node label for the first usb device?

This is in line with what I proposed here:
https://github.com/zephyrproject-rtos/zephyr/issues/24745#issuecomment-630166439
Though, if we decide there's a generic alternate node label that should be used for zephyr samples and tests, should it be called with something like `zephyr_usbd`?

jfischer-phytec-iot, 2020-05-19:
What is wrong with just `usbd` ?

galak, 2020-05-20:
> What is wrong with just `usbd` ?

Since the label is zephyr specific we need to name it as such.

galak, 2020-05-20:
PR https://github.com/zephyrproject-rtos/zephyr/pull/25495 to address a precursor aspect of this issue (ie, sanitycheck doesn't build anything for these samples).

emob-nordic, 2020-08-06:
Either all dts files change or this sample will not work.

dleach02, 2020-08-12:
@galak, are you suggesting that this sample will need a board specific overlay for the rt1010 similar to nrf overlay?

galak, 2020-08-12:
> @galak, are you suggesting that this sample will need a board specific overlay for the rt1010 similar to nrf overlay?

I see 2 ways of handling this:
1. either add board specific overlays similar to what is already there.
2. Add a generic overlay using a common dts anchor (`zephyr_usbd`) that boards can define.

dleach02, 2020-08-13:
I like the second option... seems cleaner (I think). Where does this generic overlay reside? Do you have an example of something like this?

galak, 2020-08-13:
Look at something like `samples/sensor/dps310/app.overlay`

jfischer-phytec-iot, 2020-08-17:
>     2\. Add a generic overlay using a common dts anchor (`zephyr_usbd`) that boards can define.

Sounds good.

galak, 2020-09-10:
We will need to add a section in the docs for these zephyr specific node labels to give the specific intent/usage.

galak, 2020-09-10:
Dev-Review (Sept 10, 2020): No issue w/using a node label for this purpose, just needs a zephyr namespace. (And added to docs)  @jfischer-phytec-iot asked about doing something similar for display.  Don't see any issue, but asked to work up PR to see specific details.

erwango, 2020-09-10:
If we're going wit this solution (which I find elegant), should we add some way to filter on this for sanitycheck?
https://github.com/zephyrproject-rtos/zephyr/pull/25590 was merged recently using alias. I guess it would have to be replaced by a label for coherency. but we'd need a function equivalent to `dt_alias_exists`, so we can do stuff like:
```
tests:
  sample.drivers.adc:
    tags: ADC
    depends_on: adc
    filter: dt_alias_exists("zephyr_adc")
```

jfischer-phytec-iot, 2020-09-17:
Since there are thoughts of supporting more than one device controller, and host support is WiP, would "zephyr_usbd0" node label be acceptable?

carlescufi, 2021-02-09:
@jfischer-no there seems to be an agreement on how to fix this, could you send a PR?

PR #35465 (2021-05-21): boards: add USB node aliases 
USB devicetree nodes in Zephyr have different names,
mostly derived from the designations in data sheets.
We also have an USB audio sample which has a kind of
experimental approach to configure audio devices, that
are not real hardware, via devicetree.
Sample itself is generic and can be run on any board with
USB device controller and isochronous endpoints support.
Add zephyr_usbd0 nodelabel to allow generic USB audio sample
to be build on nRF and i.MX RT boards.

USB audio class samples are generic and can be built for any
board that supports USB device and isoch [... cut]
commit fca49910 (2021-05-21): samples: usb-audio: add app.overlay and remove board specific overlays

USB audio class samples are generic and can be built for any
board that supports USB device and isochronous endpoints.
Add app.overlay that uses aliases and remove board specific overlays.

Note: USB audio is still experimental and even though the sample can
be built for specific platform, it does not mean that it can be run
on it without issues.

Fixes: #25313
closed 2021-05-21 by nashif