zephyrproject-rtos/zephyr #25314: Bluetooth: controller: legacy: Backport conformance test related changes

cvinayak, 2020-05-14:
**Describe the bug**
During conformance testing of v2.2-branch, numerous implementation fixes related to conformance test changes and failures were undertaken. These fixes are missing in legacy controller in v1.14-branch.

**To Reproduce**
Re-run conformance testing using hci_uart application in v1.14-branch

**Expected behavior**
Bluetooth Controller in v1.14-branch shall meet the same level of quality as v2.2-branch for a said Bluetooth v5.1 qualified design listing.

**Impact**
showstopper

**Screenshots or console output**
NA

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: v1.14.2

**Additional context**
backport PR #23091

cvinayak, 2020-08-17:
This issue corresponds to #23091

carlescufi, 2020-09-22:
@cvinayak does #23091 fix this issue?

cvinayak, 2020-09-23:
> 
> 
> @cvinayak does #23091 fix this issue?

@carlescufi Yes.

PR #23091 (2020-11-17): [backport v1.14] Bluetooth: controller: legacy: Backport v2.2 to v1.14-branch
Commits in this PR are manual cherry-pick of commits based on differences in ctrl.c file.

Fixes #25314
closed 2020-12-07 by github-actions[bot]