zephyrproject-rtos/zephyr #25327: Move to C99 integer types and deprecate zephyr specific types

galak, 2020-05-14:
With the introduction of the following commit:

https://github.com/zephyrproject-rtos/zephyr/commit/f32330b22c333012cf5f771b0eae9ef378e446d4

galak, 2020-05-14:
I'd recommend that post 2.3 release we rename all zephyr types (s8_t -> int8_t, u8_t -> uint8_t, etc.) and make the types in include/zephyr/types.h as deprecated.

jhedberg, 2020-05-14:
> post 2.3 release

Do you mean that in the special sense that we defined starting with 2.2, i.e. in the short window between tagging 2.3.0 and opening the merge window again? That might make sense considering how much code this will touch. Just as a reminder: we have a dedicated column for this in https://github.com/zephyrproject-rtos/zephyr/projects/27

galak, 2020-05-14:
> Do you mean that in the special sense that we defined starting with 2.2, i.e. in the short window between tagging 2.3.0 and opening the merge window again? That might make sense considering how much [...]

That is what I meant. :)

pabigot, 2020-05-27:
Should be trivial to do with Coccinelle.

jfischer-phytec-iot, 2020-05-27:
I like Zephyr's short types I do not want to miss it. This will also break coding style in many files, **again**.

stephanosio, 2020-05-28:
As long as there is no technical reason to have these Zephyr-specific types anymore, IMHO, it would be preferable to convert back to/use the C99 standard types (hopefully this will be the last), as doing so will result in overall:

- less ambiguity
- less confusion
- less inconsistencies (e.g. unintentional mixing of `uint32_t` and `u32_t`)
- improved code portability
- improved standard compliance

galak, 2020-05-28:
One question as I worked up PR is how to handle {u}int64_t.  I added a commit in the PR to make {u}int64_t the same across 32 or 64-bit arch's.  This at least matches how zephyr/types.h was defining things.

The one gotcha we run into is native_posix_64.  Since its getting the typedef from system headers.  So not 100% sure what we want to do for that issue yet.

stephanosio, 2020-05-28:
> The one gotcha we run into is native_posix_64. Since its getting the typedef from system headers. So not 100% sure what we want to do for that issue yet.

cc @aescolar

andrewboie, 2020-05-28:
Full support for this change.

> One question as I worked up PR is how to handle {u}int64_t. I added a commit in the PR to make {u}int64_t the same across 32 or 64-bit arch's. This at least matches how zephyr/types.h was defining [...]

What issue are you running into? I would expect {u}int64_t to be an {un}signed long long long everywhere? Or is that not workable?

aescolar, 2020-05-28:
@galak I pressume the problem you refer to is due to the printf format specifiers?
But, as far as I recall, that was not a native_posix_64 issue as such. Wasn't this an issue around the whole  Zephyr, due to zephyr having defined its own types, but not having print specifier macros, so everywhere we have now %llu %lu %whathaveyous (instead of PRIu64, PRIiPTR, etc), which then caused compiler format warnings depending on how those types were defined for one target or another?

If that is the case, and given that we are now moving into stdint types, I'd say we should fix the underlying problem and use the stdint print format specifiers (PRI*), instead of redefining types in the zephyr headers (include/toolchain/zephyr_stdint.h)

jhedberg, 2020-05-28:
> The one gotcha we run into is native_posix_64. Since its getting the typedef from system headers. So not 100% sure what we want to do for that issue yet.

I'd be surprised to see issues when building native_posix and native_posix_64 on normal Linux distributions (or even some more unusual ones). At least I've never come across the need for the PRI macros during my 20+ years Linux user space development experience. The only exception is 64-bit types, for which I think we can't avoid PRI (and that's an acceptable compromise IMO). For anything else, I hope we could stick to the more readable %u, %d, %x, etc, at least until someone comes with an environment where it's a problem. So in that sense I'm kind of of the opposite opinion to @aescolar , i.e. I think we should keep taking advantage of having control of the stdint types where we can have control of them. That's also what I thought was the plan here, i.e. now that Zephyr and stdint types are identical and we don't get warnings from format specifiers, then we won't have to change the format strings even if we drop the Zephyr types completely.

galak, 2020-05-28:
> What issue are you running into? I would expect {u}int64_t to be an {un}signed long long long everywhere? Or is that not workable?

from: 
`/usr/include/bits/types.h`:

```
#if __WORDSIZE == 64
typedef signed long int __int64_t;
typedef unsigned long int __uint64_t;
#else
__extension__ typedef signed long long int __int64_t;
__extension__ typedef unsigned long long int __uint64_t;
#endif
```
(eventually {u}int64_t will get typedef based on __{u}int64_t)

So on x86_64/native_posix_64 we get int64_t as `long int` instead of `long long int`

galak, 2020-05-28:
> I'd be surprised to see issues when building native_posix and native_posix_64 on normal Linux distributions (or even some more unusual ones). At least I've never come across the need for the PRI [...]

I'm in agreement here, the intent is to keep the changes to developers as small as possible here, otherwise we'll just end up leaving things as they are.  So I think having a few PRI.64 users for cases that native_posix_64 builds tests/samples is the easiest solution going forward.  I'll push an update to the PR which such changes.

aescolar, 2020-05-29:
@galak : If the idea is to continue redefining the types in `include/toolchain/zephyr_stdint.h` then please also redefine their PRI macros so users can use them for printing.
Though I really do not understand why you keep singling out native_posix_64 given that that header redefinition are today needed for many platforms (just comment out its content and try to run some test)

galak, 2020-06-01:
> @galak : If the idea is to continue redefining the types in `include/toolchain/zephyr_stdint.h` then please also redefine their PRI macros so users can use them for printing.

What do you mean redefine them?  What is wrong with how they are defined?

> Though I really do not understand why you keep singling out native_posix_64 given that that header redefinition are today needed for many platforms (just comment out its content and try to run some tests)

I single out native_posix_64 as its the only platform that the way sized types differs and we don't have control in Zephyr of how they are defined.

For all non-native posix platforms Zephyr is controlling how {u}int{8,16,32,64}_t are defined.  We are setting them exactly the some across all architectures we support.

So for example we set `uint64_t` to be `unsigned long long` on arm, aarch64, riscv32, riscv64, x86, x86_64, etc..  On native_posix we do not control how the C99 types are defined as we inherit them from the C library's definition.  It happens that on native_posix (32-bit) the Linux C library definitions match what we have in Zephyr.  However on native_posix_64 the `int64_t` and `uint64_t` types get typedef to `long` and `unsigned long` by the C library.  This differs from ALL other Zephyr architectures that define `int64_t` as `long long` and `uint64_t` as `unsigned long long`.  Thus the only outlier is native_posix_64 and why I reference it.

aescolar, 2020-06-01:
> What do you mean redefine them? What is wrong with how they are defined?

For those who use the libC provided by Zephyr nothing as you are matching them.
But I guess users who chose to use another libC can just redefine those on their own if they need (which they may not need anyhow, if their libC is not using those __*INT*_TYPE__ macros)

> For all non-native posix platforms Zephyr is controlling how...

I had assumed you were intending to not do that anymore, as from your comment above ("_One question as I worked up PR is how to handle {u}int64_t. I added a commit in the PR to make {u}int64_t the same across 32 or 64-bit arch's._") I interpreted that commit was just a quick hack that you intended to drop after you found another solution.
Therefore the misunderstanding.

galak, 2020-06-01:
> I had assumed you were intending to not do that anymore, as from your comment above ("_One question as I worked up PR is how to handle {u}int64_t. I added a commit in the PR to make {u}int64_t the [...]

Ah, sorry for the confusion there.  I do think we should keep that commit as the whole reason Nicolas had to introduce `include/toolchain/zephyr_stdint.h` was to get 32-bit integer the same on all platforms.  We should have been doing the 64-bit case as well, but guess the number of 64-bit arch's has been low til recently :)

andrewboie, 2020-06-01:
Is this a situation where we could solve this for most people by defining `__UINT64_TYPE__`  to be `unsigned long` and not `unsigned long long` on 64-bit LLP64 systems in our headers?

galak, 2020-06-01:
> Is this a situation where we could solve this for most people by defining `__UINT64_TYPE__` to be `unsigned long` and not `unsigned long long` on 64-bit LLP64 systems in our headers?
> 
> would still always need PRI_x64 macros for printing though

Unfortunately for all the 32-bit platforms 'unsigned long' isn't going to work ;)

andrewboie, 2020-06-01:
> Unfortunately for all the 32-bit platforms 'unsigned long' isn't going to work ;)

Yes, my suggestion was to conditionally compile it.
There's even a compiler predefined macro `__LP64__` you can use

galak, 2020-06-01:
> Yes, my suggestion was to conditionally compile it.
> There's even a compiler predefined macro `__LP64__` you can use. Or we could hang it off CONFIG_64BIT.

Doing that gets us back into the headache of having to use PRI macros all over the place since the fundamental type would be changing between arch builds.  I thinking its better to keep everything consistent (which was why we introduced the u32_t/u64_t type to start with) is what we should be aiming for.

galak, 2020-06-02:
@sjg20 was curious how the u-boot sandbox deals with mixing use of libc or library from host system while defining its own types.

closed 2020-06-08 by galak

chris-makaio, 2020-08-26:
Uhm, sorry, I’m lost... what’s the current preferred type? U8_t or uint8_t?

andrewboie, 2020-08-26:
> Uhm, sorry, I’m lost... what’s the current preferred type? U8_t or uint8_t?

You want to use `uint8_t`.

sjg20, 2020-08-27:
> @sjg20 was curious how the u-boot sandbox deals with mixing use of libc or library from host system while defining its own types.

Well I didn't see that sorry. 

Do you mean stdint.h ? If so, U-Boot doesn't actually use that. It has its own empty version so that libraries build from U-Boot don't use the libc version.

In fact U-Boot used to support using stdint.h but it was dropped by Masahiro in:

9865543ae65 Remove CONFIG_USE_STDINT

and I have not had the need/energy to get it back in there.