zephyrproject-rtos/zephyr #25328: mesh_demo is failing

krishnabadam, 2020-05-14:
Hi Zephyr Team,

It seems there is a bug in mesh_demo, it looks like the issue is with destination address as a Group Address.

Below is the version on which I am running this demo for BBC microbit.
VERSION_MAJOR = 2
VERSION_MINOR = 2
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

sample used: samples/bluetooth/mesh_demo

I have build a image with NODEADDR as 0x000f(PUBLISHER_ADDR).
After loading on to BBC Microbit I am getting semaphore acquisition failure in bt_hci_cmd_send_sync of hci_core.c (k_sem_take) at the time of sending heartbeat to group address 0xc000.

I have just added a limited logs, follows as below:
============= LOGS START ================
*** Booting Zephyr OS build zephyr-v2.2.0-2292-g0bac2f35cc77  ***
Initializing...
Unicast address: 0x000f
Bluetooth initialized
Mesh initialized
Loading stored settings
Provisioning completed
Configuring...
net_idx 0x0000 app_idx 0xfffd dst 0x000f
len 20: 000000000123456789abcdef0123456789abcdef
SeqZero: 0x0000 Attempts: 4
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 80000001c08d3329220da4fa3bcda4f7
Seq 0x000000
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 80000021d462aaff8bdaf77eff4037bf
Seq 0x000001
net_idx 0x0000 app_idx 0xfffd dst 0x000f
len 10: 803d0f000000f1050000
src 0x000f dst 0x000f len 15 headroom 9 tailroom 5
Payload len 15: 0018d1fd5fef9ac80bc5321ab9cdcd
Seq 0x000002
net_idx 0x0000 app_idx 0xfffd dst 0x000f
len 8: 803d0f0000000200
src 0x000f dst 0x000f len 13 headroom 9 tailroom 7
Payload len 13: 00e309eab124a60883bd68496c
Seq 0x000003
net_idx 0x0000 app_idx 0xfffd dst 0x000f
len 10: 801b0f0000c0f1050000
src 0x000f dst 0x000f len 15 headroom 9 tailroom 5
Payload len 15: 00a8c617ce72b73293745cd785e667
Seq 0x000004
net_idx 0x0000 app_idx 0xfffd dst 0x000f
len 11: 803900c0ff050700000000
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 0035655d318f9a3de982582290beb396
Seq 0x000005
Publishing heartbeat messages
Configuration complete
src 0x000f dst 0x000f len 7 headroom 9 tailroom 13
Payload len 7: 00000000000003
Seq 0x000006
net_idx 0x0000 app_idx 0xfffe dst 0x000f
len 6: 800300000000
src 0x000f dst 0x000f len 11 headroom 9 tailroom 9
Payload len 11: 006b42845b021ac3ad0b98
Seq 0x000007
net_idx 0x0000 app_idx 0xfffe dst 0x000f
len 11: 803e000f000000f1050000
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 004064952afc2313d1363fd2db5df432
Seq 0x000008
net_idx 0x0000 app_idx 0xfffe dst 0x000f
len 9: 803e000f0000000200
src 0x000f dst 0x000f len 14 headroom 9 tailroom 6
Payload len 14: 000b56f40d9c9e5843edd218a826
Seq 0x000009
net_idx 0x0000 app_idx 0xfffe dst 0x000f
len 11: 801f000f0000c0f1050000
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 0057b982e47d7485ed407d4e7c9ef387
Seq 0x00000a
net_idx 0x0000 app_idx 0xfffe dst 0x000f
len 11: 060000c0ff050700000000
src 0x000f dst 0x000f len 16 headroom 9 tailroom 4
Payload len 16: 004e10c0b161e1c0e1f815bf419fe436
Seq 0x00000b
src 0x000f dst 0xc000 len 4 headroom 9 tailroom 16
Payload len 4: 0a070001
Seq 0x00000c
type 0x01 len 21: 37fa67d589ea9962cba9b83dee205a47b5a4336e93
type 1 len 21: 37fa67d589ea9962cba9b83dee205a47b5a4336e93
count 3 interval 20ms duration 120ms
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:370
        k_sem_take failed with err -11
============= LOGS END ================

Please do let me know if you need any further details on this.

Thanks,
Krishna B

carlescufi, 2020-05-14:
@jhedberg any chance you could take the mesh demo on micro:bit for a spin to see if you can reproduce?

krishnabadam, 2020-05-15:
Hi,

Just to add on some more info, Its not limited to GROUP Address instead its applicable to a non local destination address. So the issue appears to be in  'bt_mesh_adv_send' call path.

Thanks,
Krishna B

carlescufi, 2020-05-19:
@trond-snekvik if you have a micro:bit and can take a look at this it would be great.

closed 2020-08-02 by github-actions[bot]