zephyrproject-rtos/zephyr #25334: SPI won't build on microbit with I2C

saewoonam, 2020-05-15:
**Describe the bug**
I am trying to add additional flash memory to a BBC microbit board via its edge connector.   The flash memory is an MX25R64 chip.   After I modify the devicetree file to have both i2c0 and spi1, it doesn't recognize the flash memory when I try to build the spi_flash example.   However, if I disable the i2c0 bus and rename the spi1 bus to spi0 it does recognize the flash and compiles.  I need the i2c0 bus to get readings from the sensors, and I need the SPI bus to store data on the flash memory. 

**To Reproduce**
1.  Modify the microbit dtsi file to include SPI flash memory on the spi1 bus

`&spi1 {
 >......status = "ok";
 >......sck-pin = <23>;
 >......mosi-pin = <21>;
 >......miso-pin = <22>;
 >......cs-gpios = <&gpio0 20 0>;
 >......mx25r6435f@0 {
 >......>.......compatible = "jedec,spi-nor";
 >......>.......reg = <0>;
 >......>.......spi-max-frequency = <80000000>;
 >......>.......label = "MX25R64";
 >......>.......jedec-id = [c2 28 17];
 >......>.......size = <67108864>;
 >......>.......wp-gpios = <&gpio0 16 0>;
 >......>.......hold-gpios = <&gpio0 18 0>;
 >......};
 };`
2.   Try to build spi_flash example.   Will not build, here is the example of the error
/Users/nams/zephyrproject/zephyr/saewoo/ubit/spi_flash/src/main.c:24:2: error: #error Unsupported flash driver
   24 | #error Unsupported flash driver
      |  ^~~~~
/Users/nams/zephyrproject/zephyr/saewoo/ubit/spi_flash/src/main.c: In function 'main':
/Users/nams/zephyrproject/zephyr/saewoo/ubit/spi_flash/src/main.c:42:13: error: expected ')' before 'FLASH_NAME'
   42 |  printf("\n" FLASH_NAME " SPI flash testing\n");
      |             ^~~~~~~~~~~
      |             )
/Users/nams/zephyrproject/zephyr/saewoo/ubit/spi_flash/src/main.c:45:33: error: 'FLASH_DEVICE' undeclared (first use in this function)
   45 |  flash_dev = device_get_binding(FLASH_DEVICE);
      |                                 ^~~~~~~~~~~~
/Users/nams/zephyrproject/zephyr/saewoo/ubit/spi_flash/src/main.c:45:33: note: each undeclared identifier is reported only once for each function it appears in
[56/120] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/mempool.c.obj
ninja: build stopped: subcommand failed.

3.   When I disable i2c0, and rename spi1 to spi0.   It will build.
4.  If I disable i2c0 and try to use the spi as spi1, it also fails

**Impact**
This is a showstopper for my project to add memory to my microbit to log data.  I have been trying to build low-cost bluetooth dongles that implement the exposure tracing protocols proposed by Apple/Google, PACT, TCN, etc..  Here is my github repo:
https://github.com/saewoonam/ble_adv_covid

**Environment (please complete the following information):**
 - OS: OSX Catalina
 - Toolchain Zephyr SDK
 - Version 2.2.99

pabigot, 2020-05-19:
You're missing one piece:
```
&spi1 {
	status = "okay";
	compatible = "nordic,nrf-spi";  /* <<< this */
	sck-pin = <23>;
	mosi-pin = <21>;
	...
````

The nrf51822 dtsi doesn't provide a compatible for spi1, leaving it to the user to select between `nordic,nrf-spi` and `nordic,nrf-spis`.  In other nRF51 boards this is defaulted in the board dts file; that wasn't done on microbit.

If you add that to your devicetree source it should build correctly.  Please confirm.

pabigot, 2020-05-19:
You may want to update to the latest Zephyr master (and to 2.3.0 when it's released later this month).  I can confirm failures against master, but the way it fails is different due to recent changes to the devicetree tooling.  I tested against master.

saewoonam, 2020-05-20:
Great!   Thanks for taking the time to respond to my issue.   I am sorry that I am a newbie and didn't realize that I needed to add that line.  I will try to update against the new master after I see if the SPI bus actually works with a device...

closed 2020-05-20 by saewoonam