zephyrproject-rtos/zephyr #25337: LED pins always configured as PWM outputs

henrikbrixandersen, 2020-05-15:
**Describe the bug**
I have come across an issue with the recent devicetree changes. For several boards, the `pinmux.c` logic previously decided whether to configure LED pins as GPIOs or PWMs based on Kconfig. Now they decide based on `DT_NODE_HAS_COMPAT_STATUS(…, okay)` on the PWM controller, which - at least for some boards - leads to always configuring the LED pins as PWMs, thus breaking samples using the LEDs as GPIOs.

This is definitely an issue with several NXP boards, but similar issues may exist on other boards.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/basic/blinky; mkdir build; cd build
2. cmake -DBOARD=twr\_ke18f
3. make flash
4. Observe no blinking LED

**Expected behavior**
Amber LED (led0, D9) should blink.

**Impact**
Unable to control LEDs using GPIOs.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Commit SHA: v2.3.0-rc1-99-ga0ae53410e

henrikbrixandersen, 2020-05-15:
See e.g. https://github.com/zephyrproject-rtos/zephyr/blob/a0ae53410eef31d1773dbfc5a3a6591e45dee11c/boards/arm/twr_ke18f/pinmux.c#L37 for an example of this. Node `ftm3` is always enabled in the devicetree leading to the LED pins always being configured as PWM outputs.

henrikbrixandersen, 2020-05-15:
Setting priority as high, since this breaks existing board definitions.

henrikbrixandersen, 2020-05-15:
Potentially covered by #24745.

erwango, 2020-05-15:
relates to #25251

galak, 2020-05-15:
In look at this on at least twr_ke18f we need to decide which case should be the default from the board.dts point of view (GPIO or PWM), and then add overlays to any samples that want to behave differently.   The reason for this is we desire to move pinmux configuration into DTS going forward and so when that happens we'll run into this again at that point.  So lets just address this once right now.

galak, 2020-05-15:
Boards impacted:

- [ ] frdm_k82f
- [ ] frdm_kw41z
- [ ] hexiwear_k64
- [ ] twr_ke18f
- [ ] rv32m1_vega

boards/arm/frdm_k82f/pinmux.c:#else
boards/arm/frdm_kw41z/pinmux.c:#else
boards/arm/hexiwear_k64/pinmux.c:#else
boards/arm/mec1501modular_assy6885/pinmux.c:#else
boards/arm/mec1501modular_assy6885/pinmux.c:#else
boards/arm/mec15xxevb_assy6853/pinmux.c:#else
boards/arm/mec15xxevb_assy6853/pinmux.c:#else
boards/arm/twr_ke18f/pinmux.c:#else
boards/arm/twr_ke18f/pinmux.c:#else
boards/arm/twr_ke18f/pinmux.c:#else
boards/arm/twr_ke18f/pinmux.c:#else
boards/arm/twr_ke18f/pinmux.c:#else
boards/arm/v2m_musca/pinmux.c:#else
boards/arm/v2m_musca_b1/pinmux.c:#else
boards/riscv/rv32m1_vega/pinmux.c:#else

galak, 2020-05-15:
@henrikbrixandersen , @MaureenHelm would appreciate if you can comment on the default setting for the PINs.  I would think GPIO and than have blinky_pwm, fade_led, and rgb_led have overlays to switch to PWM instead of GPIO.

henrikbrixandersen, 2020-05-19:
> @henrikbrixandersen , @MaureenHelm would appreciate if you can comment on the default setting for the PINs. I would think GPIO and then have blinky_pwm, fade_led, and rgb_led have overlays to switch to PWM instead of GPIO.

@galak I would say that LEDs as GPIOs would be a sane default. Board specific overlays in the PWM samples could very well be the best way of handling this.

henrikbrixandersen, 2020-05-19:
Another approach to the same issue for STM32 boards was taken by @erwango in #25288.

galak, 2020-05-19:
> Another approach to the same issue for STM32 boards was taken by @erwango in #25288.

It would work, just means punting the issue to 2.4.  The problem will come up again when we move pinmux data to DTS.  So I'd say we try and make a decision about it now if we can.

carlescufi, 2020-05-19:
Release meeting: Decided to follow the approach in #25288 for the 5 boards listed here. @galak will follow-up with a PR.

PR #25456 (2020-05-20): boards: Make GPIO pin config default for LEDs instead of PWM
Add an additional check for CONFIG_PWM to decide if pins associated with
LED are configured for GPIO or PWM.

Fixes #25337
commit 33936000 (2020-05-20): boards: Make GPIO pin config default for LEDs instead of PWM

Add an additional check for CONFIG_PWM to decide if pins associated with
LED are configured for GPIO or PWM.

Fixes #25337
closed 2020-05-20 by carlescufi