zephyrproject-rtos/zephyr #25346: Timestamp in LOG jumps 00:08:32

alexsven, 2020-05-15:
**Describe the bug**
The timestamp in the logging module jumps by 00:08:32 from time to time. This bug occurs if CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000. 
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Build hci_rpmsg on nrf5340pdk_nrf5340_cpunet and peripheral_hr on nrf5340pdk_nrf5340_cpuapp on one PDK and hci_rpmsg on nrf5340pdk_nrf5340_cpunet and central_hr on nrf5340pdk_nrf5340_cpuapp on the other PDK. Add CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000 to prj.conf in hci_rpmsg

**Impact**
May cause other issues in kernel

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
Was built using 14c911de63266f614e1f619f5e6cff59cf49fb92 of Zephyr

pabigot, 2020-05-19:
@alexsven Do you require setting `CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000`?  Using ticks-per-second that is not a power of 2 on Nordic platforms causes problems because the duration of a tick becomes 327 ticks of a 32768 Hz clock.  If your application works leaving this at the default 32768 it would be better to keep it there.

FWIW: 08:32 is 512 s which is the point where the 24-bit 32 KiHz counter wraps.  There may be bugs related to reconstructing a log timestamp, in which case this is something @nordic-krch  should look at.

alexsven, 2020-05-20:
@pabigot Not necessarily no. We used to have this set, and it worked fine, but then when we moved to 2.3.0-rc1 we experienced a lot of weird issues. Since the HW has not changed, there has to be a  bug either in the new code or in the old that didn't have a problem with CONFIG_SYS_CLOCK_TICKS_PER_SEC=1000. 

Also, in the help section of the config we find this: "The optimal values from a computational point-of-view are 1000, 500, 250 and 125, since in these cases there is either no computation required, or it is all done via bit-shifting. These also give a granularity from 1ms to 8ms." So there should maybe be a disclaimer there for Nordic platforms

pabigot, 2020-05-20:
What version did you move from?  The default has been 32768 for a couple releases.

I wouldn't be terribly surprised if there's a bug in the code.  The cumulative error that results from a non-integral divisor could be an issue too.

The config help is outdated as it was written in a world where people assumed the system clock ran at 1 MHz.  It needs to be rewritten.  I'll open an issue for that.

alexsven, 2020-05-25:
Moved from version 2.2.0

nordic-krch, 2020-05-26:
any chance that you try with https://github.com/zephyrproject-rtos/zephyr/pull/25517/commits/b6deff76b5ea876c88ef00ca98a7839c14879587 commit. It may have fixed it.

nordic-krch, 2020-06-03:
@alexsven have you tried that maybe?

alexsven, 2020-06-03:
Sorry, no I haven't tested it out. Will try it out and report back

alexsven, 2020-06-03:
@nordic-krch It seems to be working with b6deff7

closed 2020-06-03 by alexsven