zephyrproject-rtos/zephyr #25351: test:mimxrt1050_evk:tests/subsys/usb/bos/: run failure

hakehuang, 2020-05-15:
**Describe the bug**
usbobs case failure on this platform, other RT platfroms are ok.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. make
4. make flash

**Expected behavior**
test failure

**Impact**
usb descriptor

**Screenshots or console output**
```
*** Booting Zephyr OS version 2.3.0-rc1  ***
Running test suite test_osdesc
===================================================================
starting test - test_usb_bos_macros
length 57
D: Header
D: 05 0f 39 00 02 18 10 05 |..9.....
D: 00 38 b6 08 34 a9 09 a0 |.8..4...
D: 47 8b fd a0 76 88 15 b6 |G...v...
D: 65 00 01 01 01 1c 10 05 |e.......
D: 00 df 60 dd d8 89 45 c7 |..`...E.
D: 4c 9c d2 65 9d 9e 64 8a |L..e..d.
D: 9f 00 00 03 06 03 00 02 |........
D: 02                      |.       
D: webusb cap
D: 05 0f 39 00 02 18 10 05 |..9.....
D: 00 38 b6 08 34 a9 09 a0 |.8..4...
D: 47 8b fd a0 76 88 15 b6 |G...v...
D: webusb cap msos v2
D: 05 0f 39 00 02 1c 10 05 |..9.....
D: 00 df 60 dd d8 89 45 c7 |..`...E.
D: 4c 9c d2 65 9d 9e 64 8a |L..e..d.
D: 9f 00 00 03             |....    

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/usb/bos/src/test_bos.c:192: test_usb_bos_macros: (!memcmp(hdr, &webusb_bos_descriptor, len) || !memcmp(hdr, &webusb_bos_descriptor_2, len) is false)
Wrong data
FAIL - test_usb_bos_macros
===================================================================
starting test - test_usb_bos
test_usb_bos: ret 0 len 57 data 0x800011b1

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/usb/bos/src/test_bos.c:214: test_usb_bos: (!memcmp(data, &webusb_bos_descriptor, len) || !memcmp(data, &webusb_bos_descriptor_2, len) is false)
Wrong data
FAIL - test_usb_bos
===================================================================
Test suite test_osdesc failed.
===================================================================
PROJECT EXECUTION FAILED
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - v2.3.0-rc1

jfischer-phytec-iot, 2020-05-19:
I can not reproduce, tested on mimxrt1050_evk.

hakehuang, 2020-05-20:
@jfischer-phytec-iot, I find it can only reproduce on one of my mimxrt1050_evk board, so looks like a board related issue. So close this issue. Thanks

closed 2020-05-20 by hakehuang