zephyrproject-rtos/zephyr #25369: tests/drivers/gpio/gpio_basic_api: test_gpio_deprecated step fails on STM32 boards

ABOSTM, 2020-05-15:
**Describe the bug**
While running tests/drivers/gpio/gpio_basic_api on STM32 boards (nucleo_f746zg, nucleo_f207zg ...),
the step 'falling edge' of test_gpio_deprecated step fails.

**To Reproduce**
Steps to reproduce the behavior:
1. west -v build -p auto  -b nucleo_f746zg tests/drivers/gpio/gpio_basic_api/
2. See error

**Expected behavior**
Test Passed

**Screenshots or console output**

> starting test - test_gpio_deprecated
> callback triggered: 1
> INT cfg 50000, cnt 1
> FAIL - test_callback@86. config PIN_IN fail: -16
> 
>     Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_basic_api/src/test_deprecated.c:159: test_gpio_deprecated: (test_callback(GPIO_INT_EDGE | GPIO_INT_ACTIVE_LOW) not equal to TC_PASS)
> falling edge failed
> FAIL - test_gpio_deprecated

carlescufi, 2020-05-15:
@ABOSTM could you please label test failures with the "Tests" label? Thanks!

ABOSTM, 2020-05-18:
@carlescufi it looks like I don't have access right to do so.
Can you grantme access right ?

closed 2020-05-18 by carlescufi

carlescufi, 2020-05-27:
> @carlescufi it looks like I don't have access right to do so.
> Can you grantme access right ?

Invited you now to become a collaborator

PR #25370 (2020-05-18): test/drivers/gpio/gpio_basic_api: disable interrupt at end of test
When switching from rising edge to falling edge of test **test_gpio_deprecated()**,
because EXTIcallback is already configured (from rising edge test),
the pin configuration abort for EBUSY reason.
It is necessary to disable interrupt at end of test, so that next test will start with clean configuration.

Fixes #25369