zephyrproject-rtos/zephyr #25378: Installation problems

Terstegge, 2020-05-15:
**Describe the bug**
Zephyr can not be installed following the installation instructions

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the installation procedure for Linux 
2. When typing `west init ~/zephyrproject`
I get the error
```
Traceback (most recent call last):
  File "/home/andreas/.local/bin/west", line 7, in <module>
    from west.app.main import main
  File "/home/andreas/.local/lib/python3.5/site-packages/west/app/main.py", line 180
    log.die(f"file not found: {self.mle.filename}")
                                                 ^
SyntaxError: invalid syntax
```

**Expected behavior**
I would expect that the installation instructions work out -of-the box. There
are so many build systems that work reliabley...

**Impact**
Coming from Mbed OS I want to give zephyr a try. Having errors in such a 
early stage is a show-stopper...

**Screenshots or console output**
see above

**Environment (please complete the following information):**
 - OS: (e.g. Linux 64bit)

**Additional context**
Add any other context about the problem here.

Nukersson, 2020-05-18:
According to [this](https://docs.zephyrproject.org/latest/getting_started/installation_linux.html#installation-linux) Python 3.6 or later is required. As I can see you use Python 3.5. Please follow all istructions for your Linux distribution carefully, before you begin with `west init`.

Good Luck!

Terstegge, 2020-05-19:
Sorry, my fault. After installing python 3.8 everything is working now!

closed 2020-05-19 by Terstegge