zephyrproject-rtos/zephyr #25379: Bluetooth mesh example not working

NilsMinor, 2020-05-15:
**Describe the bug**
Hi, 
I tried to gest started with the zephyr mesh examples on my custom board (https://github.com/NilsMinor/inuk-base) that is based on an E73-2G4M04S1B module (nrf52832 qfaa). 
I tested my custom bluetooth board before using fruitymesh, an other nrf52 mesh software  that is based on nordics soft device that worked fine.

I used the nrf52dk_nrf5832 as zephyr board, I did not change anything on the nrf52dk_nrf5832 its for testing. After testing the Blinky example that worked fine I wanted to try the mesh examples.

I used the examples samples/boards/nrf/mesh/onoff_level_lighting_vnd_app and samples/bluetooth/mesh_demo to get started.

Building and flashing worked fine with the lates zephyr 2.30 on ubuntu with nrfjprog and jlink.
**To Reproduce**
I followed the instructions from https://docs.zephyrproject.org/latest/samples/boards/nrf/mesh/onoff_level_lighting_vnd_app/README.html and https://docs.zephyrproject.org/latest/samples/bluetooth/mesh_demo/README.html#overview

**Expected behavior**
I was not able to get any sign of live from the device. I was not able to get communication on uart (not sure if this is enabled in this examples) but I also could not detect the device over bluetooth.

**Environment (please complete the following information):**
 - OS Ubuntu 18.02
 - Toolchain latest sdk and lates Zephyr 
I have no Idea what step I am missing. I appreciate any tip. I am new to zephyr and really want to get the mesh demo running before I make further investigations on how to use zephyr for my application. Thanks for any help.

Best regards,
Nils

carlescufi, 2020-05-16:
@trond-snekvik @jhedberg does the example need provisioning perhaps?

NilsMinor, 2020-05-16:
Hey @carlescufi I tried different apps to scan BLE and tried to use the nRF mesh app for provisioning. It feels like I am missing something. Did I understood it correctly, I do not need to flash an bootloader or other software? It should just be building the app and flashing it all using west right?

Thanks for your help,
Nils

carlescufi, 2020-05-19:
@NilsMinor you don't need a bootloader, just build and flash the app.
Can you try `samples/bluetooth/beacon` first and see if you can scan that using your mobile phone first?

NilsMinor, 2020-05-19:
@carlescufi ok I found the bug and it was my fault! I designed the hardware my self and I missed something! I was not able to build the beacon example with the **nrf52dk_nrf52832** setup but with the **nrf52_vbluno52** board config. The different was that the nrf52dk enables the DCDC converter that I did not connected on my hardware. So the dcdc startup went wrong and stopped the zephyr. I switched off the dcdc converter in Kconfig and was able to build also the mesh examples! So thanks for your help, I learned a lot :D

closed 2020-05-19 by NilsMinor

carlescufi, 2020-05-27:
> @carlescufi ok I found the bug and it was my fault! I designed the hardware my self and I missed something! I was not able to build the beacon example with the **nrf52dk_nrf52832** setup but with [...]

Glad to hear that, thank you for letting us know.

tcpipchip, 2020-11-23:
wow, 
Do you mean that replace the NRF52DK_NRF52832 by NRF52_VBLUNO52 and and finally run on E73-2G4M04S1B the ZEPHYR ?

I am having the same problem, my E73-2G4M04S1B does not run ZEPHYR!

trond-snekvik, 2020-11-24:
@tcpipchip Please create a separate issue if you're having trouble with this. Sounds like you have a board config or XYAL accuracy  issue.

tcpipchip, 2020-11-24:
yes, make sense! it´s a CRYSTAL problem!

tcpipchip, 2020-11-24:
24Mhz