zephyrproject-rtos/zephyr #25382: devicetree: Add ranges property support for PCIe node

Abhishek-brcm, 2020-05-15:
This requirement originates from : https://github.com/zephyrproject-rtos/zephyr/pull/25166#discussion_r423409878
For PCIe, supported outbound memory windows should be parsed by `ranges` property in DT node.
Below are more details:

Linux supports parsing of `ranges` and `dma-ranges` property under pcie node.
pcie node example:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/arch/arm64/boot/dts/qcom/msm8996.dtsi#n750
Parsing of ranges(for OB resources) and dma-ranges(for IB resources) is supported via below function:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/pci/of.c#n250
This ranges supports passing type of resource(IO/MEM/PREFETCH, etc) that can be mapped to given OB window resource and PCI OB->AXI translation addresses.
However, all this is w.r.t. PCIe RC, we don't have such requirement for PCIe EP.

For PCIe EP, we would ideally want to just parse the multiple OB region addresses, DT node would look like:
```
pcie@4e100000 {
    .... 
    ranges = <0x0 0x50000000 0x0 0x8000000>,
             <0x4 0x00000000 0x0 0x8000000>;
    ....
};
```
and change driver like:

       .map_low_base = DT_INST_PROP_BY_IDX(0, ranges, 0), /* should hold 0x50000000  */
       .map_low_size = DT_INST_PROP_BY_IDX(0, ranges, 1), /* should hold 0x8000000 */
       .map_high_base = DT_INST_PROP_BY_IDX(0, ranges,2), /* should hold 0x400000000*/
       .map_high_size = DT_INST_PROP_BY_IDX(0, ranges, 3), /* should hold 0x8000000 */
Currently, I am parsing specific properties(namely `map_lowmem_*`/`map_highmem_*`) from DT, and would like to replace like above. This enhancement is useful if we want to support multiple outbound windows.

mbolivar-nordic, 2020-05-18:
@galak do you think this should be discussed at dev-review? I don't know PCIe.

galak, 2020-05-19:
> @galak do you think this should be discussed at dev-review? I don't know PCIe.

Let's discuss it a little in this issue first.

@Abhishek-brcm can you explain the use case a bit more for PCIe EP.  I'd expect that the host system would be configuring the EP BARs to convey the PCIe bus address space to the EP.  So what exactly is the ranges property in the DTS describing?

Abhishek-brcm, 2020-05-20:
> I'd expect that the host system would be configuring the EP BARs to convey the PCIe bus address space to the EP.

That's right from RC point of view. Host would configure EP BAR to convey its RC's PCIe address space. It is later used by EP to route the inbound transactions.

> So what exactly is the ranges property in the DTS describing?

Here, we are talking about the outbound transactions from EP, wherein EP would map Host address(to be read/write) to its outbound memory to generate outbound transaction.
We want to parse the outbound memory regions supported by EP to be parsed from DT.

In Linux, while RC node has support for parsing outbound memory regions as explained in the issue description, it does not have this support for EP node. For EP nodes, range is being supplied through reg property itself. For example, in below EP node, `addr_space` conveys entire OB range :

```
pcie1_ep: pcie_ep@51000000 {
    reg = <0x51000000 0x28>, <0x51002000 0x14c>, <0x51001000 0x28>, <0x1000 0x10000000>;
    reg-names = "ep_dbics", "ti_conf", "ep_dbics2", "addr_space";
    ...
};

```

However, considering that there could be multiple discrete OB regions supported by EP (We have 2 in our current implementation of driver) OR
RC support could be enhanced to generate Memory read/write transactions through OB memory instead of just Cfg read/write as it is now, ranges property parsing is good addition.

galak, 2020-05-20:
What address space are the ranges mapping between?

Is there a reference to a linux dts that is using ranges as you mention?  I glanced at several EP bindings but didn't seem ranges used in them like (Documentation/devicetree/bindings/pci/nvidia,tegra194-pcie.txt and designware-pcie.txt).  I even seen comments like:

From `designware-pcie.txt`:

```
    (The old way of getting the configuration address space from "ranges"
    is deprecated and should be avoided.)
```

Abhishek-brcm, 2020-05-20:
> What address space are the ranges mapping between?

For EP, we do not give fixed mapping through the DT node. We only have to give the available outbound region. The Host address(PCIe address) to be mapped changes runtime and is passed to the PCIe driver by the requester. That is the core functionality of EP driver.

> Is there a reference to a linux dts that is using ranges as you mention? I glanced at several EP bindings but didn't seem ranges used in them

As I had mentioned in above comment, "In Linux, while RC node has support for parsing outbound memory regions as explained in the issue description, it does not have this support for EP node"..
Only RC has support for parsing OB memory regions, But along with it, it mentions PCIe address range to be mapped as well:
```
pcie0_rc: pcie@48000000 { {
...
                ranges = <0x83000000 0x0 0x42000000 0x0 0x42000000 0 0x2000000>,
                         <0x43000000 0x4 0x00000000 0x4 0x00000000 0 0x80000000>;
...
```
Here:
`ranges = <[PCI resource flags(cell 1)]  [RC Outbound address(cell 2 and 3)] [PCIe address to be mapped(cell 4 and 5)]  [size(cell 6 and 7)]>
`

> From designware-pcie.txt:
>     (The old way of getting the configuration address space from "ranges"
>     is deprecated and should be avoided.)
> 

This comment seems specific to designware RC's configuration address space, unrelated.

However, considering the fact that we don't really need to parse either PCI resource flags or PCIe address to be mapped for EP node, we just need the OB memory ranges, we could use reg property itself to achieve this - like how Linux does (I have given example in above comment). 
@stephanosio do you agree?

galak, 2020-05-20:
Based on slack discussion for the brcm EP use case encoding should be done as part of the `reg` property.

If there's a desire to support PCIe RC in Zephyr please let us know as there's a bit of support we'd need to add to the devicetree scripts to handle a PCIe address space and bus parsing which doesn't exist today.

Abhishek-brcm, 2020-05-27:
@galak @stephanosio  I have updated PCIe EP PR  https://github.com/zephyrproject-rtos/zephyr/pull/25166 to use `reg` property to specify the available outbound memory ranges. Please review the same.

As for ranges support, I am okay to defer this issue until/if more mature PCIe RC support is required in zephyr.

superna9999, 2021-08-13:
For reference, I pushed `ranges` parsing at #37668

superna9999, 2022-06-21:
This has been implemented and merged in #38106

closed 2022-06-21 by superna9999