zephyrproject-rtos/zephyr #25389: driver MMIO address range management

andrewboie, 2020-05-16:
**Is your enhancement proposal related to a problem? Please describe.**
Most of Zephyr assumes that MMIO addresses are fixed at build time and are simply constants specified in DTS and exposed in generated headers, with values stores in immutable device config structures in ROM.

However, two hardware features can cause this to fall over:

1. Some peripherals have their MMIO ranges enumerated at build time, even for peripherals directly on the board. Various PCI-E peripherals on boards like UpSquared fall into this category. The **physical** base address of MMIO regions must be stored in RAM.

2. If an MMU is used, MMIO regions need to be mapped into the kernel's address space. With a virtual memory scheme that does not use identity page tables, the physical address is not the same as the virtual address it gets mapped to. It might be possible to establish physical-to-virtual mapping at build time through the same multi-phase build tricks used to generate other kernel data structures, but if not (such as if the MMIO is only known at runtime) the **virtual** address of MMIO regions must be stored in RAM as well.

A good solution for this will allow MMU virtual memory to be implemented without introducing overhead in builds that do not use it, as well as not imposing additional RAM requirements for devices which may or may not be hot-plugged.

**Describe the solution you'd like**
Don't have specific solutions, but something workable ought to do the following:

1. Devices that don't have an MMU or runtime MMIO ranges continue to maintain all addresses in ROM. More generally, use of RAM is minimized; if we can know something at build time, whether it's the MMIO physical base address, or the virtual mapping, put it in ROM instead or derive it.
2. A registration mechanism inside drivers which requests a virtual address range for physical MMIO ranges, using the virtual addresses for subsequent access. Equivalent to Linux ioremap()...except this should all compile to nothing if an MMU is not in use.

andrewboie, 2020-05-19:
Initial MMU virtual memory enabling will not reserve virtual addresses at build time, this may be a future enhancement however if there is sufficient interest.

dcpleung, 2021-05-26:
`DEVICE_MMIO_*` implements this already. Build time mapping needs to be done in arch-specific way as each arch has different way to generate page tables. So in general, this is done. So closing...

closed 2021-05-26 by dcpleung