zephyrproject-rtos/zephyr #25395: Websockets Server API

Nukersson, 2020-05-17:
***WHAT***
Implementation of Websocket Server API would be helpful for embedded devices, which can be configured/controlled over Ethernet.
***WHY*** To such device may belong Ethernet connected sound boxes, embedded routers or modems, toys, smart watches etc. Websocket Server API would allow configuration of such devices just right in your local network.

***CURRENT SITUATION***
Currently Zephyr has only [Websocket Client API](https://docs.zephyrproject.org/latest/reference/networking/websocket.html) implementation.

***SUGGESTED SOLUTION***
We may adapt [MinnowServer](https://github.com/RealTimeLogic/MinnowServer)  since it is under [GPL2 License](https://github.com/systemd/systemd/blob/master/LICENSE.GPL2) 

Or implement it same way as MinnowServer.

Any suggestions?

jukkar, 2020-05-18:
> We may adapt [MinnowServer](https://github.com/RealTimeLogic/MinnowServer) since it is under [GPL2 License](https://github.com/systemd/systemd/blob/master/LICENSE.GPL2)

The GPL2 license is not compatible with Apache2 license so MinnowServer cannot be used in Zephyr.

Nukersson, 2020-05-18:
> > We may adapt [MinnowServer](https://github.com/RealTimeLogic/MinnowServer) since it is under [GPL2 License](https://github.com/systemd/systemd/blob/master/LICENSE.GPL2)
> 
> The GPL2 license is not compatible with Apache2 license so MinnowServer cannot be used in Zephyr.
> 

Thank you for your message. Can you change then this Issue to feature request?

Nukersson, 2020-05-21:
@jukkar There were already sample implementation of [Websocket Server in v1.13.0](https://github.com/zephyrproject-rtos/zephyr/tree/v1.13-branch/samples/net/ws_echo_server). Why is this sample no more in newer versions of Zephyr?

jukkar, 2020-05-21:
> There were already sample implementation of [Websocket Server in v1.13.0](https://github.com/zephyrproject-rtos/zephyr/tree/v1.13-branch/samples/net/ws_echo_server). Why is this sample no more in newer versions of Zephyr?

The one in v1.13 was using net_app API which was removed in v1.14. This also meant that the applications using that API were also removed. One of those apps was websocket server. We could create a server sample app for websocket using BSD sockets, but no one has so far volunteered to do it.

Nukersson, 2020-05-21:
> The one in v1.13 was using net_app API which was removed in v1.14. This also meant that the applications using that API were also removed. One of those apps was websocket server. We could create a [...]

Thank you for very fast and good answer. I am interested in writing this implementation or rewriting the older one sample. I'll try provide a link to my pull request later.

Nukersson, 2020-08-04:
@jukkar Does latest version ( v2.3.99 ) of zephyr has http server/client API?

jukkar, 2020-08-04:
> Does latest version ( v2.3.99 ) of zephyr has http server/client API?

No, we have only client APIs in upstream Zephyr. There is the civetweb module that can be used when server side APIs are needed. There is a sample in `samples/net/sockets/civetweb` that uses the module.

Nukersson, 2020-08-04:
@jukkar _Please correct, if I am wrong_:
1. Is it right then, that CivetWeb support WebSocket Server functionality in Zephyr according to [this](http://civetweb.github.io/civetweb/Embedding.html)?
2. If 1, holds: Am I right, that WebSocket server can be implemented just same way `samples/net/sockets/civetweb` application?

jukkar, 2020-08-04:
> * Is it right then, that CivetWeb support WebSocket Server functionality in Zephyr according to [this](http://civetweb.github.io/civetweb/Embedding.html)?
> 
> * If 1, holds: Am I right, that WebSocket server application can be implemented just same way as `samples/net/sockets/civetweb` application?

I did not do the civetweb porting and I have not used the API so cannot confirm what you are asking. But the original idea of the civetweb porting was that it could be used in Zephyr as is, so I would expect it to work like you describe above.

Nukersson, 2020-08-04:
@jukkar Do you mean adding `http_server.h` API functionality from v1.13-branch to current Zephyr is better approach than CivetWeb due to it's huge memory consumption?

Nukersson, 2020-08-10:
Can be solved by [this](https://github.com/zephyrproject-rtos/civetweb/pull/2)

jukkar, 2020-08-11:
> Do you mean adding `http_server.h` API functionality from v1.13-branch to current Zephyr is better approach than CivetWeb due to it's huge memory consumption

IMHO having a native zephyr API is always a better than using external modules. That allows us to better optimize the code usage for zephyr. But if there are no resources to develop/maintain such APIs, good option is to use already existing external code.
If someone sends a PR with HTTP/Websocket server side APIs and is willing to maintain it, then that would be great. So far no one has done that.

closed 2020-09-29 by MaureenHelm