zephyrproject-rtos/zephyr #25398: UpSquared Grub build docs don't work on Ubuntu 20.04

andrewboie, 2020-05-17:
**Describe the bug**
I'm following instructions on https://docs.zephyrproject.org/latest/boards/x86/up_squared/doc/index.html

On the step where I run:

```
$ ./boards/x86/common/scripts/build_grub.sh x86_64
```

I get the following output:

```
$ ./boards/x86/common/scripts/build_grub.sh x86_64
~/projects/zephyr/zephyr/boards/x86/common/scripts/grub ~/projects/zephyr/zephyr
Cloning into 'src'...
remote: Counting objects: 94771, done.
remote: Compressing objects: 100% (21811/21811), done.
remote: Total 94771 (delta 70661), reused 94694 (delta 70616)
Receiving objects: 100% (94771/94771), 70.39 MiB | 4.48 MiB/s, done.
Resolving deltas: 100% (70661/70661), done.
~/projects/zephyr/zephyr/boards/x86/common/scripts/grub/src ~/projects/zephyr/zephyr/boards/x86/common/scripts/grub ~/projects/zephyr/zephyr
Note: switching to 'grub-2.04-rc1-17-g8e8723a6b'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 8e8723a6b f2fs: Disable gcc9 -Waddress-of-packed-member
~/projects/zephyr/zephyr/boards/x86/common/scripts/grub ~/projects/zephyr/zephyr
~/projects/zephyr/zephyr/boards/x86/common/scripts/grub/src ~/projects/zephyr/zephyr/boards/x86/common/scripts/grub ~/projects/zephyr/zephyr
./bootstrap: Bootstrapping from checked-out grub sources...
./bootstrap: consider installing git-merge-changelog from gnulib
./bootstrap: getting gnulib files...
Cloning into 'gnulib'...
remote: Counting objects: 220896, done.
remote: Compressing objects: 100% (28337/28337), done.
remote: Total 220896 (delta 192700), reused 220605 (delta 192491)
Receiving objects: 100% (220896/220896), 45.28 MiB | 3.29 MiB/s, done.
Resolving deltas: 100% (192700/192700), done.
Note: switching to 'd271f868a8df9bbec29049d01e056481b7a1a263'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at d271f868a lock: Fix link error with --enable-threads=pth.
./bootstrap: autopoint --force
Copying file ABOUT-NLS
[... 1338 lines cut ...]
```

A few spots look suspicious:

```
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:327: installing 'build-aux/compile'
configure.ac:42: installing 'build-aux/config.guess'
configure.ac:42: installing 'build-aux/config.sub'
configure.ac:48: installing 'build-aux/install-sh'
configure.ac:48: installing 'build-aux/missing'
automake: error: cannot open < Makefile.util.am: No such file or directory
autoreconf: automake failed with exit status: 1
```

```
Importing unicode...
./autogen.sh: line 20: python: command not found
```

```
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'
```

focanmess, 2020-05-18:
I encountered this a few weeks ago on Ubuntu 18.04 and resolved it by installing Python 2 (specifically "sudo apt install python").

emob-nordic, 2020-05-20:
Sorry, that was a mistake.

PR #26716 (2020-07-08): boards: x86: update build_grub.sh for Python 3
Since Python 2 is already EOL, Linux distributions are starting
to drop support for Python 2. For example, on Ubuntu 20.04,
"python" is no longer available. This breaks the bootstrap
and autogen.sh as they are both default to executing "python".
Fix this by setting the PYTHON environment variable before
building so both scripts will use python3 instead.

Also update git checkout to grab the official 2.04 release.

Fixes #25398
commit cbb252ad (2020-07-08): boards: x86: update build_grub.sh for Python 3

Since Python 2 is already EOL, Linux distributions are starting
to drop support for Python 2. For example, on Ubuntu 20.04,
"python" is no longer available. This breaks the bootstrap
and autogen.sh as they are both default to executing "python".
Fix this by setting the PYTHON environment variable before
building so both scripts will use python3 instead.

Also update git checkout to grab the official 2.04 release.

Fixes #25398
closed 2020-07-08 by nashif