zephyrproject-rtos/zephyr #25407: No tests/samples covering socket read()/write() calls

pfalcon, 2020-05-18:
While support for POSIX read()/write() calls was implemented for some time (enabled by CONFIG_POSIX_API option), there're no tests or samples using this feature. This can lead to regressions and must be addressed.

pfalcon, 2020-05-20:
Reopening, as there're more work to do (https://github.com/zephyrproject-rtos/zephyr/pull/25443).

cfriedt, 2022-11-22:
I'll add a sample demonstrating read / write functionality. Thanks for the suggestion!

cfriedt, 2023-01-24:
This ticket actually goes far beyond simple examples (and IIRC, we do have examples of `read()` / `write()` now).

It raises the larger question.

**Do POSIX calls need to be system calls or is POSIX just a library?**

There is no kernel object associated with the global file descriptor table, so right now it's use is "free".

Additionally, there are no system-call level checks (`Z_OOPS()`) for reading or writing outside of the permitted memory region for the current thread (`CONFIG_USERSPACE`).

That means, for systems *with* an MMU / MPU we'll just get a fault, but then for systems without an MMU / MPU, it will result in random corruption.

However, that reflects the same behaviour for libraries today.

I'm tempted to close this `-EWONTFIX`, but I know that may be a contentious topic, so I'll mark it for review in the Architecture meeting.

cfriedt, 2023-02-05:
The Arch meeting was cancelled on the day this was supposed to be reviewed, so I'll just jot down some notes:

I've converted the socketpair example to use read / write as well. Will close this issue.

However, I think there are some valid concerns with Zephyr's `fdtable.c`, but at the same time, it's not clear if Zephyr is really the place to focus on this sort of thing.
* the global collection of file descriptors is, well.. global
* there is no separation between kernel, userspace, or even other threads in terms of file descriptor use
* indeed, there is no file descriptor kernel object (such that permissions would be granted, etc)

This relates back to how file descriptor tables operate on POSIX systems; each process has its own file descriptor table (they are just integers) that refer to kernel file descriptor objects, a specific set of file operations.

That's a pretty heavyweight design though.

Some might argue that Zephyr's view of running in a single address space / process space is fine for its intended purposes.

Going to close this as done for now, but we can always reopen the topic in an architecture meeting.

PR #25356 (2020-05-20): samples: net: echo_async_select: Use read()/write() if possible
If build with full POSIX API, use read()/write() instead of
recv()/send() calls for sockets.

We have read()/write() support for a while, but no samples/tests
actually performed at least a build test for it (so it will be
done now).

Fixes: #25407
commit 8af71872 (2020-05-20): samples: net: echo_async_select: Use read()/write() if possible

If build with full POSIX API, use read()/write() instead of
recv()/send() calls for sockets.

We have read()/write() support for a while, but no samples/tests
actually performed at least a build test for it (so it will be
done now).

Fixes: #25407
closed 2023-02-05 by cfriedt