zephyrproject-rtos/zephyr #25409: Inconsistent naming of Kconfig options related to stack sizes of various Zephyr components

pfalcon, 2020-05-18:
Current situation:

~~~
$ grep STACKSIZE .config 
CONFIG_TEST_EXTRA_STACKSIZE=0
CONFIG_ZTEST_STACKSIZE=3048
$ grep STACK_SIZE .config 
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_IDLE_STACK_SIZE=256
CONFIG_ISR_STACK_SIZE=2048
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=1024
CONFIG_PRIVILEGED_STACK_SIZE=1024
CONFIG_UART_MUX_RX_STACK_SIZE=512
CONFIG_NET_TX_STACK_SIZE=1200
CONFIG_NET_RX_STACK_SIZE=1500
CONFIG_NET_MGMT_EVENT_STACK_SIZE=768
~~~

Desired situation: a single grep to pick up all stack sizes.

(Z)TEST is the only offender of a common style, and it's pretty much an internal option, so should be safe to rename without big compatibility measures.

pfalcon, 2020-05-18:
cc: @stephanosio, @ioannisg, @andrewboie, @carlescufi, all the stakeholders who sounded their (strong) opinion in https://github.com/zephyrproject-rtos/zephyr/pull/24913 . It would be nice to bikeshed this issue during next month, and then (hopefully) just implement when the 2.4 window opens, without surprise for anyone.

carlescufi, 2020-05-18:
I absolutely think we need to address this and come up with a solid naming convention for stack sizes, thanks for opening this to track it @pfalcon.

andrewboie, 2020-05-18:
> cc: @stephanosio, @ioannisg, @andrewboie, @carlescufi, all the stakeholders who sounded their (strong) opinion in #24913 

The only strong opinion I have is that we shouldn't be breaking our end users unnecessarily by making their project-level Kconfigs outside of the Zephyr tree stop working

pfalcon, 2020-05-19:
@andrewboie: I hope nobody would disagree, but:

1. There're price to pay for lack of clarity/consistency too - developer comfort, and just outright mistakes (because not everyone, especially novices, will know that "you just need to do extra legwork there", some will just miss it out confused by inconsistency). So, as long as there's a clear gradient of "better" direction, and the change is simple enough in nature, it's better to do it sooner than later, exactly to not break more users than needed (number of which will grow over time).
2. We do break things in Zephyr all the time. The timeout refactor alone (3rd one?) is a good example.

In this regard, we have pretty good infrastructure for Kconfig option changes. Use of undefined (no longer defined) option in user's prj.conf will lead to fail-fast error during build (and not e.g. silently incorrect behavior). And while on a first case a user facing it will likely need to consult release note for migration path, it should be easy to learn that we apply small consistency renames to the options, so just grepping .config for substring of the old option name should give a hit re: how it should be renamed.

So, again, proposition made here is that  CONFIG_ZTEST_STACKSIZE/CONFIG_TEST_EXTRA_STACKSIZE are largely in-tree options, unlikely to be used widely out-of-tree (because hopefully people would want to contribute tests upstream, as they're boring to maintain; or if those are truly in-house tests, that chances are high they're based on another testing framework than Ztest).

So, it's proposed to just rename these options without Kconfig migration path (which is possible, but cumbersome enough to reserve it to truly "publicly/widely used" options). If there's disagreement with this proposal, and requirement for deprecation/migration path, then it would be nice to here it sooner rather than later, to plan for effort required to implement this ticket accordingly.

Thanks.

carlescufi, 2020-05-27:
@pfalcon would you mind sending a patch?

pfalcon, 2020-05-27:
> @pfalcon would you mind sending a patch?

Yep, I plan to work on that when 2.4 window opens.

nashif, 2020-08-14:
reminder, 2.4.0 is going to close soon....

pfalcon, 2021-02-04:
Did not find time to go thru this, closing.

closed 2021-02-04 by pfalcon