zephyrproject-rtos/zephyr #25413: soc: ti_simplelink: kconfig: ble: placeholder for hal_ti

cfriedt, 2020-05-18:
**Is your enhancement proposal related to a problem? Please describe.**
The hal/ti module requires some RF-protocol-specific sources to be included based on the RF protocol being used (e.g. BLE, 802.15.4). They HAL would like to conditionally compile sources.

See https://github.com/zephyrproject-rtos/hal_ti/pull/13#discussion_r426676618

**Describe the solution you'd like**
I would like to create a placeholder Kconfig for BLE on the cc13xx_cc26xx platform until my BLE port is complete, in order to be consistent in the TI hal.

**Describe alternatives you've considered**
Originally, I thought it would be fine to unconditionally compile the rf patch sources. 

**Additional context**
This is just pulling a change from my existing PR #21631

carlescufi, 2021-02-04:
@cfriedt if you are not actively working on porting the Controller to TI, could you please close this?

closed 2021-02-04 by cfriedt

cfriedt, 2021-02-04:
Of course. The port is on hold currently as the client / vendor were not able to fund it in a timely manner and I have other business obligations currently. Hope to resume at some point if another contractor does not pick up the slack.