zephyrproject-rtos/zephyr #25433: Add vendor specific class custom usb device sample

jamesrosssharp, 2020-05-19:
It would be nice to have a sample which shows how to create a custom vendor specific (class 0xff) device with custom endpoints including an interrupt endpoint. Users could then use this sample to implement custom usb device protocol for e.g. sending sensor data etc.

jfischer-phytec-iot, 2020-05-19:
Actually any class implementation can be used as example, [subsys/usb/class/loopback.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/loopback.c) is the simple one and is also used as example in the [documentation](https://docs.zephyrproject.org/latest/reference/usb/index.html#implementing-non-standard-usb-class). There is also more complex [webusb sample](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/usb/webusb/src).

carlescufi, 2020-05-20:
@jamesrosssharp as per the explanation by @jfischer-phytec-iot it seems your request is addressed. I will close this issue but feel free to reopen if you need more info.

closed 2020-05-20 by carlescufi