zephyrproject-rtos/zephyr #25434: nRF5340 Bluetooth peripheral_hr sample high power consumption

cvinayak, 2020-05-19:
**Describe the bug**
Average current consumption of peripheral_hr sample in application CPU and hci_rpmsg application in network CPU, is ~2.5 mA. It appears the nRF5340pdk starts with idle current of ~14uA, but after sometime (in the image below ~24 seconds, but this is random), the base idle current increased to ~2.5 mA.

[image]

It appears the kernel does not sleep after one of the 1 second wake up as seen in the below image:
[image]

**To Reproduce**
Steps to reproduce the behavior:

in the prj.conf file, have the below:

CONFIG_BT_DEBUG_LOG=n
CONFIG_LOG=n
CONFIG_CONSOLE=n
CONFIG_SERIAL=n

cpunet application
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf5340pdk_nrf5340_cpunet  ../samples/bluetooth/hci_rpmsg
3. ninja
4. ninja flash

cpuapp application
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf5340pdk_nrf5340_cpuapp  ../samples/bluetooth/peripheral_hr
3. ninja
4. ninja flash

On the nrf5340pdk cut SB40, connect the nRF PPK daughter board and using nRF Connect for desktop v.3.3.3 Power profiler software measure the power consumption.

**Expected behavior**
idle currents should remain low

**Impact**
showstopper

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA or Version used: d2c74eb

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-12-07:
@ioannisg and @cvinayak is this still applicable?

cvinayak, 2020-12-14:
This is no longer observed. Closing the issue.

closed 2020-12-14 by cvinayak