zephyrproject-rtos/zephyr #25442: Does Zephyr support USB host mode ?

icephyr, 2020-05-19:
Hi Everyone,

Now I am developing the NXP MIMXRT 1050 EVK board with zephyr system and everything goes well till now. But I notice that in the development guide page of  MIMXRT EVK, it describes that " Only USB device function is supported in Zephyr at the moment."， But the USB on the EVK should work in host mode, according to our design and requirements. So I wonder if the USB Host mode has been supported in the lastest version of Zephyr ? Or I have to use other embedded system instead of Zephyr ?
By the way, the zephyr version I am using is V1.14.0

jfischer-phytec-iot, 2020-05-19:
> So I wonder if the USB Host mode has been supported in the lastest version of Zephyr ?

Still not supported.

>Or I have to use other embedded system instead of Zephyr ?

certainly not, except Linux based OS.

closing in favour of #12386

closed 2020-05-19 by jfischer-phytec-iot

icephyr, 2020-05-19:
> > So I wonder if the USB Host mode has been supported in the lastest version of Zephyr ?
> 
> Still not supported.
> 
> > Or I have to use other embedded system instead of Zephyr ?
> 
> certainly not, except Linux based OS.
> 
> closing in favour of #12386

Thanks for your reply. So only Linux based OS supports USB host mode ?
What about freertos or rt-thread, etc.

icephyr, 2020-05-20:
Hi Guys, I'd like to add some information about my project for your futher analysis.

Now I am working to develope a board with MIMXRT 1050, it collects data from sensor and send the data to the server with LTE dongle module. The MIMXRT is connected with LTE module by USB. So the MIMXRT should be in host mode while the LTE in device mode, in my understanding. Since  " Only USB device function is supported in Zephyr at the moment.", I think MIMXRT  running zephyr cannot meet my requirements, so do I have to change another embedded system like freeRTOS， or some ways can help to implement the requirements with zephyr ?
Since I'm new to USB developement, hope I didn't misunderstand the documents. Thanks.

carlescufi, 2020-05-20:
> Now I am working to develope a board with MIMXRT 1050, it collects data from sensor and send the data to the server with LTE dongle module. The MIMXRT is connected with LTE module by USB. So the [...]

Your assumption is correct. Zephyr is currently not capable of acting as a USB Host and therefore you will not be able to achieve what you want at the present time with Zephyr. However, there will be more than just having USB Host support to achieve what you want.
cc @MaureenHelm

jfischer-phytec-iot, 2020-05-20:
The host stack alone will probably not fulfill the requirements for your project. AFAIK Zephyr supports at least one LTE modem connected over UART.

icephyr, 2020-05-21:
Thank you guys， well noted.

kaushalmp, 2022-09-14:
Hi Any updates on the availability of USB xHCI host driver? Please point me to any other host stack that supports xHCI and more number of classes.